Koenkk/zigbee2mqtt #1719: Retrieving data from a sensor attached to ESP8266 with CC25230

sophanox, 2019-07-10:
Hi there,

Apologies in advance if this is a basic question, but I can't find any information on how exactly I can retrieve data from an ESP8266 (via the CC25230). I have followed the guide and everything seems to be configured properly. Running: 

`socat -d -d pty,raw,b115200,echo=0,link=/opt/zigbee2mqtt/vusb/zigbee_cc2530 tcp:192.168.0.66:1775` 

Results in a successful connection:

`socat[19065] N successfully connected from local address AF=2 192.168.0.64:42356
`

 I have then set up the following rule in ESPEasy:

`on System#Boot do
  timerSet,1,1
endon

On Rules#Timer=1 do
    Publish temp,"Air-temp-humid":"[Air-temp-humid#Temperature]",
  timerSet,1,10
endon
`

And then I can see the message being published in the log:

`
1883618: ACT : Publish temp,'Air-temp-humid':'24.95',
`

However, if I use mosquitto to sub to this topic or all topics:

` mosquitto_sub -v -d -t '#'`

I do not receive the data. The only relevant info I receive is:

`zigbee2mqtt/bridge/config {"version":"1.4.0","commit":"927c4db","coordinator":20190608,"log_level":"info","permit_join":true}`

Could you please let me know what I'm doing wrong?

Thanks!

JLFN, 2019-07-11:
How to connect a CC2530 coordinator via an Esp8266. 
As topic said its a cordinator only. 
Coordinator
A coordinator is a special router. In addition to all of the router capabilities, the coordinator is responsible for forming the network.

sophanox, 2019-07-11:
Ahh thank you for your comment. I realise now I have misinterpreted the purpose of the ESP8266. I thought it would allow me to read the data of a sensor (SHT-31 in my case) over ZigBee. But if it had to use wifi in any case that really negates the point of my use of ZigBee.

I don't suppose you know if such a module exists? Something that would take the data from a non-zigbee sensor and publish it? Whilst being low power of course as well. 

Thanks

Koenkk, 2019-07-11:
I will close this as it's not an issue, feel free to continue discussion though.

closed 2019-07-11 by Koenkk

JLFN, 2019-07-11:
@sophanox see https://github.com/Koenkk/zigbee2mqtt/issues/1606