Koenkk/zigbee2mqtt #1720: it should be possible to supress retransmission of sensor data on restart/reconnection to HA

davidmonro, 2019-07-11:
I've got some xiaomi temperature sensors. They seem to report at least every hour even if nothing changes, so I've set

    homeassistant:
      expire_after: 3600

which works fine; if the device falls off the network the values go to Unknown in HA.

However if I restart either zigbee2mqtt or HA, then all the sensor data gets republished, which means that HA resets the expire timer, and for the next hour I've got stale data showing up in HA (and in the graphs).

I could potentially deal with this by creating template sensors in HA which compare the last_seen attribute with the current time and drop the message if it is old, but that is rather messy. I think there should probably be an option (possibly per device) of disabling the republishing of data on restart/reconnect; I'd rather have no data than wrong data.

(currently running zigbee2mqtt version 1.4.0 (commit #927c4db) on a USB Coordinator firmware version: '20181224' but I don't think that is relevant).

Cheers

Koenkk, 2019-07-11:
I think you just shouldn't set `birth_message` in the home assistant `configuration.yaml` OR define in zigbee2mqtt `configuration.yaml`

```yaml
advanced:
  homeassistant_status_topic: 'dummy'
```

davidmonro, 2019-07-15:
Hmm. Hadn't thought of that. I guess the issue there is that you can't have some devices republish and others not, but I don't have lots of different types of sensors so that isn't an issue for me. I'll give it a go.

closed 2019-09-20 by stale[bot]