Koenkk/zigbee2mqtt #1721: How to debug issues adding CC2530 router

SittingDuc, 2019-07-11:
Hello,

Throwaway question: can a xbee s2c be used as a router without a micro attached? i.e. flash the xbee, provide it with 3v3 and ground and nothing else. I am expecting the answer is "no", because the thing seems to be more of a modem and less of a radio+micro.

Real question: I am having trouble verifying my CC2530 router is working / attached to the zigbee network, and wondering if anyone has any suggestions for what I have done wrong, or what I should do next?

###  Summary

zigbee2mqtt Hass.io addion v1.4.0 (commit #uknown), x86-64 vm
coordinator: CC2531, firmware 20190608
router: CC2530, firmware router-cc2530-diag_201902.hex

The two were originally paired in 2018; but I've always been suspicious if they actually work together.
In June I moved from a RaPi3 to an x86 vm. The coordinator works and finds 14 endpoints just fine. But I don't think it can see the router. I'm not getting a lot of feedback so I'm not sure how to debug nor what to do next.

### Details
> 7/6/2019, 7:32:13 PM - info: Starting zigbee2mqtt version 1.4.0 (commit #unknown)
> 7/6/2019, 7:32:13 PM - info: Starting zigbee-shepherd
> 7/6/2019, 7:32:15 PM - info: zigbee-shepherd started
> 7/6/2019, 7:32:15 PM - info: Coordinator firmware version: '20190608'
> 7/6/2019, 7:32:15 PM - info: Currently 14 devices are joined:
> 7/6/2019, 7:32:15 PM - info: router1_cc2530 (0x00124b001b7b1d9f): CC2530.ROUTER - Custom devices (DiY) [CC2530 router](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/) (Router)
> 7/6/2019, 7:32:15 PM - info: sense_thp_roomhobby (0x00158d000234d3d3): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
> [...]

So clearly a count of Xiaomi devices are attached to my coordinator. But the edges of the network are a little low on signal, I would like to use a router to increase my range.

So I have a CC2530, and I recently flashed it with router-cc2530-diag_201902.hex

It was originally paired to the coordinator back in 2018 when both units were running older firmware. When I enable joining with mqtt yesterday, no new nodes are reported. And zigbee2mqtt's configuration.yaml has the device listed

>  '0x00124b001b7b1d9f':
>    friendly_name: router1_cc2530
>    retain: false

But I am not seeing a whole lot of output in the zigbee2mqtt log; nor in the mqtt topics.

### Steps I have tried

**USB serial cable**
Hanging a usbtty off the cc2530 (txd=p0_2, rxd=p0_3, p2_0 to ground or floating), I get output like

> ??H?TexasIntrumens   Sample Light
> ???H?IEEE:00124B001B7B1D9F
> ???H?TexasInstruments SW5: Help

With more or less corruption each time; suggesting that the flow control is a little wrong. The IEEE number matches so I know I've got the right router.

**graphvis**
Looking at zigbee2mqtt/bridge/networkmap/raw after 24hr of uptime, I see (abbreviated):
> [{"ieeeAddr":"0x0013a200410ac8b4","nwkAddr":41832,"lqi":0,"parent":"0x00124b0018ed3830","status":"online"},
{"ieeeAddr":"0x00124b001b7b18dd","nwkAddr":13067,"lqi":0,"parent":"0x00124b0018ed3830","status":"offline"},
{"ieeeAddr":"0x00124b001b7b1d9f","nwkAddr":61311,"lqi":0,"parent":"0x00124b0018ed3830","status":"offline"}]

Everybody is parent to 0012..3830 which must be the CC2531 USB stick coordinator. Nobody is parent to any of the routers. I see 0013..c8b4 which must be the xbee I was using for debugging. It is shown as "online" but it's powered off. I see my two CC2530, 0012..1d9f which is showing with an LQI of 0 and Offline.

Moving the CC2530 into the same room as the coordinator, it still has an LQI of 0 and is offline.

**xbee network graph**
Using old threads like https://github.com/Koenkk/zigbee2mqtt/issues/252 and my xbee to debug the network state, the Xbee graph can see the coordinator, and can see the routers, but the routers are always pink.
[image]
(xbee top left in green, this CC2530 bottom right in pink)

**next?**
I have ordered some more pieces from AliExpress; they should be here in August; which should let me rule out faulty HW, and let me test an entirely new IEEE address for pairing.

After that, I am out of ideas, so it is time to open an issue ticket and see if the great people of the Internet can help me.

ccorderod, 2019-07-11:
@SittingDuc Yes, you can. I am using xbee s2c pro (with external antenae) in my setup as router; cc2531 usb stick with router firmware were not working for me. With xbee it works like a charm!. You can monitor your zb network in real time if you plug the xbee to a pc and use xbee tool. To setup xbees, just make sure you write the following parameters to your xbee with xbee tool: panid, channel mask, encryption enabled and the encryption key you use to use in your current zb network (if you did not, your network is using the default encryption key documented by @Koenkk on zigbee2mqtt documentation). Hope this will help. Cheers.

ccorderod, 2019-07-11:
@SittingDuc One more thing: upgrade your coordinator firmware!; firmware 20190608 is not good AT ALL!.

SittingDuc, 2019-07-11:
@ccorderod the latest firmware on https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/default is 20190608, from "16 days ago". Or are you suggesting I move to the 3.0 stack? Because those are marked WIP :)

I'll look at my xbee later; since my main target are Xiaomi endpoints, I have probably left the default key which I can grab off the internet. Thanks for the details!

Okay, I left the router powered sitting about 1m from the coordinator overnight. the word "router" doesn't appear in the zigbee2mqtt logs at all; it is mostly "MQTT publish" for the endpoints as they chime in.

patthoyts, 2019-07-13:
@SittingDuc I recently made a ZigBee router using the CC2530 and initially it did not connect to zigbee2mqtt but on [the blog](http://ptvo.info/cc2531-based-router-firmware-136/) written by the person that created the router firmware there is a hint to power the router on/off 3 times to get it to re-pair. This persuaded mine to join the network and I then saw router messages in the zigbee2mqtt log and mqtt topics.

SittingDuc, 2019-07-13:
### Coordinator Versions
I found another thread from earlier this year wherein he mentioned the 20190425 firmware was not kind to him, so he went back to 20190315 "max stability" firmware. So I've tried that; overstriking the coordinator 2531 dongle with 20190315. And now, 8 hours later the networkmap/raw is kinda sparse. In MQTT log, Endpoints are chiming in about as many and about as often as before. But the map looks empty
`zigbee2mqtt/bridge/networkmap/raw [{"ieeeAddr":"0x0013a200410ac8b4","nwkAddr":41832,"lqi":32,"parent":"0x00124b0018ed3830","status":"offline"},{"ieeeAddr":"0x00124b0018ed3830","nwkAddr":0,"lqi":250,"parent":"0x0013a200410ac8b4","status":"online"}]`
So that appears to have not helped.

### xbee x2c as router
I tweaked my xbee (firmware 4060, ID 0, channel mask 0xffff, Coordinator role 0, Encryption enable, Encryption key something pasted from the internet). In interactive mode, it has joined the network (AI/AO/whatever went from 0xff to 0x0 indicating it has found a peer) and moving it off the computer and onto a USB charger, it shows up in the raw map as 0x0013..c8b4 as noted above.
So I have moved the xbee out to another room (it's LQI dropped from 133 to 33) and bought all the CC2530 boxes back into my hobby room, home of the PC and the Coordinator.

For a brief moment after the move, the network map was longer:
`zigbee2mqtt/bridge/networkmap/raw [{"ieeeAddr":"0x0013a200410ac8b4","nwkAddr":41832,"lqi":34,"parent":"0x00124b0018ed3830","status":"offline"},{"ieeeAddr":"0x00124b0018ed3830","nwkAddr":0,"lqi":253,"parent":"0x0013a200410ac8b4","status":"online"},{"ieeeAddr":"0x00124b001b7b1d9f","nwkAddr":11109,"lqi":255,"parent":"0x0013a200410ac8b4","status":"offline"}]`
a third device has appeared! 0x0012..3830, so it's one of the two CC2530 routers.
But by this morning it has gone again.
And the xbee reports as "offline" too.

Overnight (say, 8hr?) devices which are closer to the xbee router have not jumped in lqi/rssi, so I speculate none of them have moved to the router. Devices that are further away have also not changed values, which is reassuring.

### Summary
I might have added the xbee as a router, I have moved it farther from the coordinator. I have moved all the CC2530 closer to the coordinator. Reflashing the coordinator to 20190315 does not appear to have helped.

Next, I will try first sending the enable-join MQTT and then mechanically power-cycling the 2530's on-and-off-2s, since they have FW newer than Jan 2019, that should work and maybe they will repair.

SittingDuc, 2019-07-13:
Enabling Join was pretty dramatic and immediate
(hmm, no spoiler tag in markdown. sorry)
`
7/14/2019, 9:58:33 AM - info: Zigbee: allowing new devices to join.
7/14/2019, 9:58:33 AM - info: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.4.0","commit":"unknown","coordinator":20190315,"log_level":"info","permit_join":true}'
7/14/2019, 9:58:44 AM - warn: No converter available for 'CC2530.ROUTER' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"manufacturerName":"LUMI","modelId":"lumi.router","powerSource":1}}'
7/14/2019, 9:58:44 AM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
7/14/2019, 9:58:44 AM - warn: No converter available for 'CC2530.ROUTER' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"hwVersion":1,"dateCode":"2018-09-26"}}'
7/14/2019, 9:58:44 AM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
7/14/2019, 9:58:45 AM - warn: No converter available for 'CC2530.ROUTER' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"locationDesc":"","physicalEnv":0,"deviceEnabled":1}}'
7/14/2019, 9:58:45 AM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
7/14/2019, 9:58:45 AM - info: MQTT publish: topic 'zigbee2mqtt/router1_cc2530', payload '{"state":false,"linkquality":107}'
7/14/2019, 9:58:45 AM - info: Device incoming...
7/14/2019, 9:58:45 AM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
7/14/2019, 9:58:49 AM - info: MQTT publish: topic 'zigbee2mqtt/router2_cc2530', payload '{"state":false,"linkquality":110}'
7/14/2019, 9:58:49 AM - info: Device incoming...
7/14/2019, 9:58:49 AM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
7/14/2019, 9:59:44 AM - info: MQTT publish: topic 'zigbee2mqtt/router1_cc2530', payload '{"state":false,"linkquality":110}'
7/14/2019, 9:59:44 AM - info: MQTT publish: topic 'zigbee2mqtt/router1_cc2530', payload '{"state":false,"linkquality":107,"description":"0/0x00124B0018ED3830","type":"COORD","rssi":101}'
7/14/2019, 9:59:44 AM - info: MQTT publish: topic 'zigbee2mqtt/router1_cc2530', payload '{"state":false,"linkquality":110,"description":"25308/0x00124B001B7B18DD","type":"NEIGHBOR","rssi":165}'
7/14/2019, 9:59:44 AM - info: MQTT publish: topic 'zigbee2mqtt/router1_cc2530', payload '{"state":false,"linkquality":110,"description":"41832/0x0013A200410AC8B4","type":"NEIGHBOR","rssi":38}'
7/14/2019, 9:59:48 AM - info: MQTT publish: topic 'zigbee2mqtt/router2_cc2530', payload '{"state":false,"linkquality":110}'
7/14/2019, 9:59:48 AM - info: MQTT publish: topic 'zigbee2mqtt/router2_cc2530', payload '{"state":false,"linkquality":110,"description":"0/0x00124B0018ED3830","type":"COORD","rssi":105}'
7/14/2019, 9:59:48 AM - info: MQTT publish: topic 'zigbee2mqtt/router2_cc2530', payload '{"state":false,"linkquality":110,"description":"59061/0x00124B001B7B1D9F","type":"NEIGHBOR","rssi":167}'
7/14/2019, 9:59:48 AM - info: MQTT publish: topic 'zigbee2mqtt/router2_cc2530', payload '{"state":false,"linkquality":110,"description":"41832/0x0013A200410AC8B4","type":"NEIGHBOR","rssi":51}'
7/14/2019, 9:59:53 AM - info: MQTT publish: topic 'zigbee2mqtt/sense_motion_backdoor', payload '{"illuminance":46,"linkquality":92,"occupancy":true,"battery":100,"voltage":3005}'
7/14/2019, 9:59:53 AM - info: MQTT publish: topic 'zigbee2mqtt/sense_motion_backdoor', payload '{"illuminance":46,"linkquality":94,"occupancy":true,"battery":100,"voltage":3005}'
7/14/2019, 10:00:44 AM - info: MQTT publish: topic 'zigbee2mqtt/router1_cc2530', payload '{"state":false,"linkquality":110,"description":"41832/0x0013A200410AC8B4","type":"NEIGHBOR","rssi":38}'
7/14/2019, 10:00:44 AM - info: MQTT publish: topic 'zigbee2mqtt/router1_cc2530', payload '{"state":false,"linkquality":110,"description":"0/0x00124B0018ED3830","type":"COORD","rssi":101}'
7/14/2019, 10:00:44 AM - info: MQTT publish: topic 'zigbee2mqtt/router1_cc2530', payload '{"state":false,"linkquality":110,"description":"25308/0x00124B001B7B18DD","type":"NEIGHBOR","rssi":165}'
7/14/2019, 10:00:44 AM - info: MQTT publish: topic 'zigbee2mqtt/router1_cc2530', payload '{"state":false,"linkquality":110,"description":"41832/0x0013A200410AC8B4","type":"NEIGHBOR","rssi":35}'
7/14/2019, 10:00:48 AM - info: MQTT publish: topic 'zigbee2mqtt/router2_cc2530', payload '{"state":false,"linkquality":110,"description":"41832/0x0013A200410AC8B4","type":"NEIGHBOR","rssi":51}'
7/14/2019, 10:00:48 AM - info: MQTT publish: topic 'zigbee2mqtt/router2_cc2530', payload '{"state":false,"linkquality":110,"description":"0/0x00124B0018ED3830","type":"COORD","rssi":105}'
7/14/2019, 10:00:48 AM - info: MQTT publish: topic 'zigbee2mqtt/router2_cc2530', payload '{"state":false,"linkquality":110,"description":"59061/0x00124B001B7B1D9F","type":"NEIGHBOR","rssi":166}'
7/14/2019, 10:00:48 AM - info: MQTT publish: topic 'zigbee2mqtt/router2_cc2530', payload '{"state":false,"linkquality":110,"description":"41832/0x0013A200410AC8B4","type":"NEIGHBOR","rssi":49}'
7/14/2019, 10:00:54 AM - info: MQTT publish: topic 'zigbee2mqtt/sense_motion_backdoor', payload '{"illuminance":38,"linkquality":94,"occupancy":true,"battery":100,"voltage":3005}'
`

Yeah, wall of text.
1. new device incoming
[... cut ...] me":"LUMI","modelId":"lumi.router","powerSource":1}}'
3. both routers now seem to be chirping with their own linkquality, and with neighbour messages including link quality

If this keeps up, and it works after I move the routers out of the hobby room, that might just be a solution to debugging my zigbee network.

#### Going forward
one open question: is there a "best" coordinator version?
one action item for me: move routers around and see if they still pair and chirp with neighbourly messages.

SittingDuc, 2019-07-19:
So I have been on Coordinator 2019-03-15 for nearly a week. The two CC2530 running diag 2019-02 firmware certainly seem "chattier", periodically reporting RSSI for nodes they have seen. I tried an experiment with re-pairing some endpoints in the rooms they live in. It didn't work well. Devices have to be paired within line-of-sight of the coordinator, alas.

Graphviz shows me the routers are in a mesh; but the endpoints either show with no link, or linked to the coordinator.

Actually, Graphviz also shows me both routers have RSSI of 0-3 to the coordinator, so they must be talking via the xbee! Which I guess means this issue could really be 'closed'. I got CC2530 to work as routers by flashing them with 2019-02 diag firmware, flashing the coordinator with CC2531 2019-03-15 "most stable" firmware, and by putting an xbee s2c with firmware 4060 programmed as a router geographically between the routers and the coordinator :)

I think my zigbee network still has 'gremlins', I am not sure how to exorcise them. Maybe modifying the CC2531 with an RP-SMA aerial and buying some more CC2530 nodes for even more routers around the house will help?

furiaceka87, 2019-07-27:
One question, how to be sure that one EndPoint is connected to router instead of the far coordinator? I have 2019-02 router firmware and the last coordinator Z-Stack 1.2 firmware

I think answer could be using networkmap,  when I try to run networkmap command:

mosquitto_sub -h 192.168.1.xx -p 1883 -u xx -P yy -C 1 -t zigbee2mqtt/bridge/networkmap/raw
mosquitto_sub -h 192.168.1.xx -p 1883 -u xx -P yy -C 1 -t zigbee2mqtt/bridge/networkmap/graphviz

I always receive nothing as answer....
I need to ctrl+c to terminate the command. I wait one hour after reboot but still no lucky.

closed 2019-10-02 by stale[bot]

SittingDuc, 2019-10-28:
Yesterday two of my Xiaomi WSDCGQ11LM stopped working.

So I sent the mqtt message "permit_join" = "true" and reset each sensor in place. (push reset button for ~10s until three led blinks. release button, one more blink, wait ~5s and 3 blinks = bound and 1 blink = adrift).
One rebound immediately. The other is still adrift. I've tried restarting the coordinator, power-cycling the routers, removing the endpoint battery, doing it all again and again (the definition of insanity).

How do I debug this?

1. if I update the coordinator FW past '20190315' will that help?
2. if I update zigbee2mqtt from 1.4.0 to 1.6.1 will that help? (I tried 1.5.0 briefly and everything broke, I've been loath to try again; but if it helps, I can burn a day, pick a new network key and repair all the things)
3. is there something I can build or buy that will let me 'sniff' the radio packets, so I can drag it into the room with the sensor and try to SEE the handshake and see where it is going wrong?

SittingDuc, 2019-10-28:
stop zigbee2mqtt, power cycle the coordinator, edit database.db, state.json, and configuration.yaml to remove the sensor, set permit_join=true in the hasio frontend, restart the coordinator, check the log says it is running and it is set to join and less than 60s have elapsed, power cycle the WSDCGQ11LM (remove battery) and physically move it to within line-of-sight of the coordinator, about 2m away.
push and hold reset button ~10s until three blinks, release for one blink, wait ~5s and.. one blink. Still unbound / adrift.

What am I doing wrong?