Koenkk/zigbee2mqtt #1722: Help on experimenting with changed reporting configuration

clockbrain, 2019-07-11:
I have some Z809A power sockets that work well but report excessively. Their reporting is configured as:
```
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.report('haElectricalMeasurement', 'rmsVoltage', 10, 1000, 1, cb),
                (cb) => device.report('haElectricalMeasurement', 'rmsCurrent', 10, 1000, 1, cb),
                (cb) => device.report('haElectricalMeasurement', 'activePower', 10, 1000, 1, cb),
                (cb) => device.report('haElectricalMeasurement', 'powerFactor', 10, 1000, 1, cb),
            ];
```
which results in 4 ZCL reporting configuration messages sent to each device:
[image]

It seems to me more than a coincidence that the Z809A then starts reporting in groups of 4 messages with essentially the same 4 measurements:
[image]

Which of course result in 4 near identical mqtt messages within 1 or 2 seconds. I know there are debounce efforts such as #1309 but I'm wondering if there might be a different fix. Perhaps each reporting configuration message initiates a separate reporting thread in the Z809A device and they all pick up the same reportable measures.

Given the zigbee standard seems to allow sending multiple reporting configuration records in a single message I'd like to experiment with sending a single aggregate ZCL reporting configuration frame containing all 4 measure configurations and seeing if the device then behaves more reasonably and only sends single reports instead of quadruple reports. Refer 2.4.7.1 in http://www.zigbee.org/wp-content/uploads/2014/11/docs-07-5123-04-zigbee-cluster-library-specification.pdf
[image]

This is where I am stuck. Can anyone suggest how I could even hard-code such a bundled reporting configuration into the startup so I could see if it changes the Z809A reporting behaviour and is therefore worth pursuing.

Koenkk, 2019-07-11:
Please try with:

```js
{
    zigbeeModel: ['Z809AE3R'],
    model: 'Z809A',
    vendor: 'Netvox',
    description: 'Power socket with power consumption monitoring',
    supports: 'on/off, power measurement',
    fromZigbee: [fz.state, fz.ignore_onoff_change, fz.ignore_electrical_change, fz.Z809A_power],
    toZigbee: [tz.on_off],
    configure: (ieeeAddr, shepherd, coordinator, callback) => {
        const cfg = [
            {direction: 0, attrId: 1285, dataType: 33, minRepIntval: 10, maxRepIntval: 1000, repChange: 1},
            {direction: 0, attrId: 1288, dataType: 33, minRepIntval: 10, maxRepIntval: 1000, repChange: 1},
            {direction: 0, attrId: 1291, dataType: 33, minRepIntval: 10, maxRepIntval: 1000, repChange: 1},
            {direction: 0, attrId: 1296, dataType: 40, minRepIntval: 10, maxRepIntval: 1000, repChange: 1},
        ];

        const actions = [
            (cb) => device.foundation('haElectricalMeasurement', 'configReport', cfg, foundationCfg, cb),
        ];
        execute(device, actions, callback);
    },
},
```

clockbrain, 2019-07-12:
@Koenkk thanks for this. I'll do some testing and let you know how it goes.

clockbrain, 2019-07-13:
With a bit of adjusting the code (fixing one of the dataTypes and borrowing some other bits from Innr SP 120) I got my Z809A to accept the all-at-once report configuration.
```
    // Netvox
    {
        zigbeeModel: ['Z809AE3R'],
        model: 'Z809A',
        vendor: 'Netvox',
        description: 'Power socket with power consumption monitoring',
        supports: 'on/off, power measurement',
        fromZigbee: [fz.state, fz.ignore_onoff_change, fz.ignore_electrical_change, fz.Z809A_power],
        toZigbee: [tz.on_off],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const onOff = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 1000, repChange: 0};

            const readRmsVoltage = {
                direction: 0, attrId: 1285,
            };

            const readRmsCurrent = {
                direction: 0, attrId: 1288,
            };

            const readActivePower = {
                direction: 0, attrId: 1291,
            };

            const readPowerFactor = {
                direction: 0, attrId: 1296,
            };

            const readCfg = [readRmsVoltage, readRmsCurrent, readActivePower, readPowerFactor];

            const rmsVoltage = {
                direction: 0, attrId: 1285, dataType: 33, minRepIntval: 60, maxRepIntval: 300, repChange: 1,
            };

            const rmsCurrent = {
                direction: 0, attrId: 1288, dataType: 33, minRepIntval: 60, maxRepIntval: 300, repChange: 1,
            };

            const activePower = {
                direction: 0, attrId: 1291, dataType: 41, minRepIntval: 60, maxRepIntval: 300, repChange: 1,
            };

            const powerFactor = {
                direction: 0, attrId: 1296, dataType: 40, minRepIntval: 60, maxRepIntval: 300, repChange: 1,
            };

            const electricalCfg = [rmsVoltage, rmsCurrent, activePower, powerFactor];
            const actions = [
                (cb) => device.foundation('haElectricalMeasurement', 'configReport', electricalCfg, foundationCfg, cb),
                (cb) => device.bind('haElectricalMeasurement', coordinator, cb),
                (cb) => device.foundation('haElectricalMeasurement', 'readReportConfig', readCfg, foundationCfg, cb),
            ];

            execute(device, actions, callback);
        },
    },

```

However, this doesn't help. It still reports excessively. Digging deeper I see that although I can set reporting with min=60 and max=300 seconds for each of the 4 measures, e.g (fragment):
[image]
and this configure reporting is successful
[image]

When I do an immediate read back of configured reporting I see it has ignored the min and only set the max for each of the 4 measures.
[image]

Is this something you have seen before? Any other ideas on what might be causing this?

Koenkk, 2019-07-13:
I've not seen this before, after updating the configure, did you also factory reset the device? Because otherwise the old configuration is kept.

clockbrain, 2019-07-19:
@Koenkk I've tried a factory reset but that didn't help.

However, after much reading of zigbee specs and testing a few things I've developed a new theory. I'm not able to progress it myself though.

I think the Z809A smart plug is sending repeated attribute reports because it is expecting a zcl default response and never gets one. 2.5.12 of the ZCL reference describes the default response command.

The attribute report frame  control field has 'Disable Default Response' set to false.

[image]

I'm thinking in deviceReceive.js it needs to send a defaultRsp back when it gets an attribute report (but not for a readRsp). Note that the repeated attribute reports all have the same sequence number which suggests the device is sending again as it doesn't think it got through.

Are you able to suggest a patch for deviceReceive.js that I could test.

This problem could well be wider than just with the Z809A.

clockbrain, 2019-07-21:
@Koenkk don't worry about adding default response to deviceReceive.js. I managed to add some code there to send back defaultRsp messages in response to attReport messages. Didn't help in reducing duplicate reports (even though I bypassed the queue to make sure the response was timely).

Looking even deeper I see that the ZCL spec requires default response messages to have the same sequence number as the message it is responding to and I can't see how that could be done in deviceReceive.js or even in zigbee.js. I think it would have to be done at a lower level in zigbee-herdsman but I don't know where to start with that.

I am becoming convinced that sending defaultRsp is the real solution to #1309 and #649.

Also see some general discussion regarding defaultRSP I found: https://www.digi.com/support/forum/2086/default-response-for-zcl

Koenkk, 2019-07-21:
Isn't this available as `data.transid` on the message?

clockbrain, 2019-07-21:
Thanks.  I didn't know about that. But I'll also have to force the outgoing defaultrsp message  to have the same seq number and can't see any parameters or options to control that.

Koenkk, 2019-07-21:
I see that this parameter is not exposed yet, I will add it in the zigbee-herdsman refactoring (currently ongoing)

clockbrain, 2019-07-22:
I was wrong - turns out that sequence number parameter is available, see https://github.com/Koenkk/zigbee-herdsman/blob/4485b8f8de23df81173b35e17b1d5c0fc9ccaf6a/src/lib/components/af.ts#L265

I'm still experimenting with getting the default response message to behave properly. Will post here if I get it working.

closed 2019-07-24 by clockbrain