Koenkk/zigbee2mqtt #1723: New tradfri motion sensor

Antonides, 2019-07-11:
# Bug Report
## What happened
Tried to pair the new motion from ikea (https://www.ikea.com/nl/nl/catalog/products/70429913/)

## What did you expect to happen
Show up as new device but doesn't show up as new device.

## How to reproduce it (minimal and precise)
Reset the motion sensor(press 4x the link button. and hold it for 10 sec.

## Debug Info
zigbee2mqtt version: newest
CC253X firmware version: newest

JLFN, 2019-07-11:
Any logs?

Koenkk, 2019-07-12:
Did you follow http://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html ?

tcoenraad, 2019-07-12:
I tried the reset procedure myself with the current release of both firmware and zigbee2mqtt, but I didn't get any useful feedback from the log.

Antonides, 2019-07-15:
After some tries and restarts it works. Thanks

closed 2019-07-15 by Antonides

loglux, 2019-08-05:
But how have you managed to link the sensor to stick - using RESET (4x press) or LINK (10 sec press) function?

alwashe, 2019-09-03:
I stuggled hours, you have to press 4x the link button, then the E1745 is connecting to the bridge.
pls add this to the manual

nickyromeijn, 2020-02-09:
Even though this is closed i'm having issues with linking the sensor to a CC2531 stick. It doesn't seem to be spotted at all. The logs don't show any mention of it when putting the sensor in pairing mode (pressing the link button 4 times). I'm using CC2531_DEFAULT_20190608 on the stick and running zigbee2mqtt 1.9. Wondering how you guys got it to work @alwashe @Antonides

canilsen, 2020-02-29:
> Even though this is closed i'm having issues with linking the sensor to a CC2531 stick. It doesn't seem to be spotted at all. The logs don't show any mention of it when putting the sensor in pairing [...]

Having the same problem. It doesn't notice it at all. Have tried the 4 button press mode and the hold 10 sec mode. This was a bummer since there is no way to adjust the timer when using it the intended way from Ikea. I don't want to wave my hands like crazy every 3 minutes when taking a shower...

sashsc, 2020-05-29:
Hi @all.

I bought the device from IKEA. Tried to get it connecetd.
I have no information about a movement.
The only thing I get is the status over the radio connection and the battery.

I tried several times to reconnect the device.

Does anyone have precise instructions on how to integrate and / or reset the device?

Here ist something from the log

`
info  2020-05-29 16:32:51: Device 'BWM_Ikea' joined
info  2020-05-29 16:32:51: Starting interview of 'BWM_Ikea'
info  2020-05-29 16:32:51: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"BWM_Ikea"}}'
info  2020-05-29 16:32:51: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"BWM_Ikea"}}'
info  2020-05-29 16:32:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"BWM_Ikea"}}'
info  2020-05-29 16:34:19: Successfully interviewed 'BWM_Ikea', device has successfully been paired
info  2020-05-29 16:34:19: Device 'BWM_Ikea' is supported, identified as: IKEA TRADFRI motion sensor (E1525/E1745)
info  2020-05-29 16:34:19: Configuring 'BWM_Ikea'
info  2020-05-29 16:34:19: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"BWM_Ikea","model":"E1525/E1745","vendor":"IKEA","description":"TRADFRI motion sensor","supported":true}}'
error 2020-05-29 16:34:29: Failed to configure 'BWM_Ikea', attempt 2 (Error: Bind 0xec1bbdfffea8e4cc/1 genPowerCfg from '0x00124b0012023268/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
    at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:244:23)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
info  2020-05-29 16:35:39: Configuring 'BWM_Ikea'
info  2020-05-29 16:35:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"BWM_Ikea"}}'
info  2020-05-29 16:35:40: Successfully configured 'BWM_Ikea'
warn  2020-05-29 16:37:48: Device '0xec1bbdfffea8e4cc' left the network
info  2020-05-29 16:37:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0xec1bbdfffea8e4cc"}}'
info  2020-05-29 16:38:00: Device 'BWM_Ikea' joined
info  2020-05-29 16:38:00: Starting interview of 'BWM_Ikea'
info  2020-05-29 16:38:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"BWM_Ikea"}}'
info  2020-05-29 16:38:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"BWM_Ikea"}}'
info  2020-05-29 16:38:02: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"BWM_Ikea"}}'
warn  2020-05-29 16:38:56: Device '0xec1bbdfffea8e4cc' left the network
info  2020-05-29 16:38:56: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0xec1bbdfffea8e4cc"}}'
error 2020-05-29 16:39:15: Failed to interview 'BWM_Ikea', device has not successfully been paired
info  2020-05-29 16:39:15: Configuring 'BWM_Ikea'
info  2020-05-29 16:39:15: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"BWM_Ikea"}}'
error 2020-05-29 16:39:25: Failed to configure 'BWM_Ikea', attempt 3 (Error: Bind 0xec1bbdfffea8e4cc/1 genPowerCfg from '0x00124b0012023268/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
    at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:244:23)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
info  2020-05-29 16:47:00: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
info  2020-05-29 16:47:01: Disconnecting from MQTT server
info  2020-05-29 16:47:01: zigbee-herdsman stopped

`

thanks for your help