Koenkk/zigbee2mqtt #1726: Error: Cannot find module 'zigbee-herdsman/dist/zcl-id' at Docker Container

cbirkenbeul, 2019-07-12:
Hello,

using Docker Container with latest Tag, so using version 1.4.0. Deleted the container image and pulled it again. If i bind mount the zigbee-shepherd git repo i get an error message:

`homeassistant-zigbee    | Using '/app/data' as data directory
homeassistant-zigbee    |
homeassistant-zigbee    | > zigbee2mqtt@1.4.0 start /app
homeassistant-zigbee    | > node index.js
homeassistant-zigbee    |
homeassistant-zigbee    |   zigbee2mqtt:info 7/12/2019, 6:57:45 PM Logging to directory: '/app/data/log/2019-07-12.18-57-45'
homeassistant-zigbee    | internal/modules/cjs/loader.js:583
homeassistant-zigbee    |     throw err;
homeassistant-zigbee    |     ^
homeassistant-zigbee    |
homeassistant-zigbee    | Error: Cannot find module 'zigbee-herdsman/dist/zcl-id'
homeassistant-zigbee    |     at Function.Module._resolveFilename (internal/modules/cjs/loader.js:581:15)
homeassistant-zigbee    |     at Function.Module._load (internal/modules/cjs/loader.js:507:25)
homeassistant-zigbee    |     at Module.require (internal/modules/cjs/loader.js:637:17)
homeassistant-zigbee    |     at require (internal/modules/cjs/helpers.js:22:18)
homeassistant-zigbee    |     at Object.<anonymous> (/app/node_modules/zigbee-shepherd-converters/converters/toZigbee.js:5:15)
homeassistant-zigbee    |     at Module._compile (internal/modules/cjs/loader.js:689:30)
homeassistant-zigbee    |     at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
homeassistant-zigbee    |     at Module.load (internal/modules/cjs/loader.js:599:32)
homeassistant-zigbee    |     at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
homeassistant-zigbee    |     at Function.Module._load (internal/modules/cjs/loader.js:530:3)
homeassistant-zigbee    | npm ERR! code ELIFECYCLE
homeassistant-zigbee    | npm ERR! errno 1
homeassistant-zigbee    | npm ERR! zigbee2mqtt@1.4.0 start: `node index.js`
homeassistant-zigbee    | npm ERR! Exit status 1
homeassistant-zigbee    | npm ERR!
homeassistant-zigbee    | npm ERR! Failed at the zigbee2mqtt@1.4.0 start script.
homeassistant-zigbee    | npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
homeassistant-zigbee    |
homeassistant-zigbee    | npm ERR! A complete log of this run can be found in:
homeassistant-zigbee    | npm ERR!     /root/.npm/_logs/2019-07-12T18_57_46_002Z-debug.log`

The mentioned logfile will not be created. Maybe some missconfigurations with your new project?

Thanks!

Koenkk, 2019-07-13:
Inside the zigbee-shepherd-converters you need to do `rm -rf node_modules && npm install`

cbirkenbeul, 2019-07-13:
Even thought using a docker container? Should i do this command inside the container, or on my docker host? Because then i need to install npm on that host. I would like to avoid that.

Koenkk, 2019-07-13:
In the zigbee-shepherd-converters repo that you bound.

jnavarrodev, 2019-07-14:
Same problem here when trying to upgrade from 1.4.0 to 1.5.0.  Is there any way to solve this without installing npm in my system?

> Even thought using a docker container? Should i do this command inside the container, or on my docker host? Because then i need to install npm on that host. I would like to avoid that.

Did you found any other solution?

millie75, 2019-07-17:
I tried the rm -rf node_modules && npm install
which crashed half way through well I think it was halfway as it had already taken 30mins

and got the below - 2 days so far trying to install this on a docker and they said dockers made live easier!

rm -rf node_modules && npm install
npm WARN tarball tarball data for typescript@3.5.2 (sha512-7KxJovlYhTX5RaRbUdkAXN1KUZ8PwWlTzQdHV6xNqvuFOs7+WBo10TQUqT19Q/Jz2hk5v9TQDIhyLhhJY4p5AA==) seems to be corrupted. Trying one more time.

> @serialport/bindings@2.0.8 install /root/data/zigbee-shepherd-converters/node_modules/zigbee-herdsman/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=11.15.0 runtime=node arch=arm libc= platform=linux)
gyp ERR! configure error
gyp ERR! stack Error: Can't find Python executable "python", you can set the PYTHON env variable.
gyp ERR! stack     at PythonFinder.failNoPython (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:484:19)
gyp ERR! stack     at PythonFinder.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:406:16)
gyp ERR! stack     at F (/usr/local/lib/node_modules/npm/node_modules/which/which.js:68:16)
gyp ERR! stack     at E (/usr/local/lib/node_modules/npm/node_modules/which/which.js:80:29)
gyp ERR! stack     at /usr/local/lib/node_modules/npm/node_modules/which/which.js:89:16
gyp ERR! stack     at /usr/local/lib/node_modules/npm/node_modules/isexe/index.js:42:5
gyp ERR! stack     at /usr/local/lib/node_modules/npm/node_modules/isexe/mode.js:8:5
gyp ERR! stack     at FSReqCallback.oncomplete (fs.js:158:21)
gyp ERR! System Linux 4.19.57+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /root/data/zigbee-shepherd-converters/node_modules/zigbee-herdsman/node_modules/@serialport/bindings
gyp ERR! node -v v11.15.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/zigbee-herdsman/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@2.0.8 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the @serialport/bindings@2.0.8 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-07-17T15_04_15_803Z-debug.log

millie75, 2019-07-17:
this is running on a complete clean build on Raspberry Pi Zero W

Koenkk, 2019-07-17:
Does it work when not mapping zigbee-shepherd-converters volume?

millie75, 2019-07-17:
[image]

   docker run \
   -it \
   -v $(pwd)/data:/app/data \
   -v $(pwd)/data/zigbee-shepherd-converters:/app/node_modules/zigbee-shepherd-converters \
   -v $(pwd)/data/lib/extension/homeassistant.js:/app/lib/extension/homeassistant.js \
   --device=/dev/ttyACM0 \
   koenkk/zigbee2mqtt:1.4.0-arm32v6

I have it as a volume as I copied your exact docker file and have this issue, unless there is something else I am doing wrong

Koenkk, 2019-07-18:
Mouting zigbee-shepherd-converters is only needed when wanting to support new devices, please use the configurations from http://www.zigbee2mqtt.io/information/docker.html

millie75, 2019-07-18:
not sure how I can show you the log when the log it says it created is not even in the log location

npm ERR!     **/root/.npm/_logs/2019-07-18T07_36_34_670Z-debug.log**
root@DietPi:~/data# cd /root/.npm
root@DietPi:~/.npm# cd _logs
root@DietPi:~/.npm/_logs# ls
2019-07-17T15_04_15_803Z-debug.log  2019-07-17T15_23_23_674Z-debug.log

Koenkk, 2019-07-19:
Can you run with `docker run -it -v $(pwd)/data:/app/data -e DEBUG=* --device=/dev/ttyACM0 ` and provide the logs?

millie75, 2019-07-19:
Thanks here you go

root@DietPi:~# docker run -it -v $(pwd)/data:/app/data -e DEBUG=* --device=/dev/ttyACM0 koenkk/zigbee2mqtt:arm32v6
Using '/app/data' as data directory

> zigbee2mqtt@1.5.1 start /app
> node index.js

  serialport/bindings loading LinuxBinding +0ms
  zigbee2mqtt:info 7/19/2019, 4:40:26 PM Logging to directory: '/app/data/log/2019-07-19.16-40-17'
  zigbee2mqtt:debug 7/19/2019, 4:40:27 PM Loaded state from file /app/data/state.json
  zigbee2mqtt:debug 7/19/2019, 4:40:27 PM Saving state to file /app/data/state.json
  zigbee2mqtt:info 7/19/2019, 4:40:27 PM Starting zigbee2mqtt version 1.5.1 (commit #unknown)
  zigbee2mqtt:info 7/19/2019, 4:40:27 PM Starting zigbee-shepherd
  zigbee2mqtt:debug 7/19/2019, 4:40:27 PM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/app/data/database.db","coordBackupPath":"/app/data/coordinator_backup.json","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee-shepherd:init zigbee-shepherd booting... +0ms
  cc-znp Initialize, opening serial port +0ms
  serialport/stream opening path: /dev/ttyACM0 +0ms
  serialport/binding-abstract open +0ms
  serialport/stream _read queueing _read for after open +91ms
  serialport/bindings/poller Creating poller +0ms
  serialport/stream opened path: /dev/ttyACM0 +134ms
  serialport/stream _write 1 bytes of data +14ms
  serialport/binding-abstract write 1 bytes +227ms
  serialport/stream _read reading +29ms
  serialport/binding-abstract read +21ms
  serialport/bindings/unixWrite Starting write 1 bytes offset 0 bytesToWrite 1 +0ms
  serialport/bindings/unixRead Starting read +0ms
  serialport/bindings/unixWrite write returned null 1 +81ms
  serialport/bindings/unixWrite wrote 1 bytes +7ms
  serialport/bindings/unixWrite Finished writing 1 bytes +3ms
  serialport/stream binding.write write finished +124ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +99ms
  serialport/bindings/poller Polling for "readable" +220ms
  cc-znp The serialport /dev/ttyACM0 is opened. +1s
  zigbee-shepherd:request REQ --> SYS:version +0ms
  cc-znp:SREQ --> SYS:version, {} +0ms
  unpi:writer --> frame [254,0,33,2,35] +0ms
  serialport/stream _write 5 bytes of data +1s
  serialport/binding-abstract write 5 bytes +1s
  serialport/bindings/unixWrite Starting write 5 bytes offset 0 bytesToWrite 5 +1s
  serialport/bindings/unixWrite write returned null 5 +14ms
  serialport/bindings/unixWrite wrote 5 bytes +5ms
  serialport/bindings/unixWrite Finished writing 5 bytes +5ms
  serialport/stream binding.write write finished +53ms
  serialport/bindings/poller received "readable" +1s
  serialport/binding-abstract read +65ms
  serialport/bindings/unixRead Starting read +1s
  serialport/bindings/unixRead Finished read 10 bytes +6ms
  serialport/stream binding.read finished +39ms
  unpi:parser <-- [254,5,97,2,2,0,2,6,3,99] +0ms
  unpi:parser --- parseNext [254,5,97,2,2,0,2,6,3,99] +12ms
  unpi:parser --> parsed 5 - 3 - 1 - 2 - [2,0,2,6,3] - 99 +12ms
  unpi:parser --- parseNext [] +152ms
  serialport/stream _read reading +232ms
  serialport/binding-abstract read +259ms
  serialport/bindings/unixRead Starting read +251ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +5ms
  serialport/bindings/poller Polling for "readable" +283ms
  cc-znp:SRSP <-- SYS:version, __timeout__ +0ms
  zigbee-shepherd:request RSP <-- SYS:version +3s
  zigbee-shepherd:request REQ --> SYS:osalNvRead +48ms
  cc-znp:SREQ --> SYS:osalNvRead, { id: 3840, offset: 0, len: 1, value: [ 85 ] } +3s
  unpi:writer --> frame [254,3,33,8,0,15,0,37] +3s
  serialport/stream _write 8 bytes of data +3s
  serialport/binding-abstract write 8 bytes +3s
  serialport/bindings/unixWrite Starting write 8 bytes offset 0 bytesToWrite 8 +3s
  serialport/bindings/unixWrite write returned null 8 +8ms
  serialport/bindings/unixWrite wrote 8 bytes +5ms
  serialport/bindings/unixWrite Finished writing 8 bytes +3ms
[... 356 lines cut ...]
    at Timeout.<anonymous> (/app/node_modules/zigbee-herdsman/dist/cc-znp/index.js:192:18)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10) +1m
  zigbee2mqtt:error 7/19/2019, 4:41:43 PM Error while starting zigbee-shepherd! (Error: request timeout)
  zigbee2mqtt:error 7/19/2019, 4:41:43 PM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 7/19/2019, 4:41:43 PM Failed to start
        {"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/cc-znp/index.js:208:22)\n    at Object.onceWrapper (events.js:273:13)\n    at CcZnp.emit (events.js:182:13)\n    at Timeout.<anonymous> (/app/node_modules/zigbee-herdsman/dist/cc-znp/index.js:192:18)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}
  zigbee2mqtt:error 7/19/2019, 4:41:43 PM Exiting...
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.5.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.5.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-07-19T16_41_43_755Z-debug.log
root@DietPi:~# ^C
root@DietPi:~# cd /root/.npm/_logs/
root@DietPi:~/.npm/_logs# ls
2019-07-17T15_04_15_803Z-debug.log  2019-07-17T15_23_23_674Z-debug.log
root@DietPi:~/.npm/_logs#

Koenkk, 2019-07-20:
Which cc2531 firmware are you running?

millie75, 2019-07-20:
The router as I already have a coordinator

Koenkk, 2019-07-20:
zigbee2mqtt can only communicate with a coordinator, not a router.

millie75, 2019-07-20:
How do you pair a router, not a lot of documentation on how to add router cc2531. Do you need any software is it just plug and play?

Koenkk, 2019-07-20:
Yes, http://www.zigbee2mqtt.io/how_tos/how_to_create_a_cc2530_router.html#4-done

millie75, 2019-07-20:
That’s for the cc2530 not the smaller cc2531. How would I power that? Do I need to buy a cc2530 and all the add on to get a router working on my network?

Koenkk, 2019-07-20:
For the cc2531, just plug it into a usb phone charger.

millie75, 2019-07-20:
Thanks all working after a deflating of the cc2531. Also big kudos for all the work you do on this

closed 2019-07-21 by Koenkk

frankhommers, 2019-08-14:
I have this problem with 1.5.1 but not with 1.4.0

Bhavana-S, 2020-08-13:
How to change the cc2531 router pan-id