Koenkk/zigbee2mqtt #1728: PM Failed network rtg scan, PM Failed network lqi scan on EDGE

dreimer1986, 2019-07-13:
When using edge version recent on 13.07.19 I am bombed with
  zigbee2mqtt:warn 7/13/2019, 7:42:15 PM Failed network rtg scan for device: '0x8418260000003e28' with error: 'Error: Timed out after 10000 ms'
  zigbee2mqtt:warn 7/13/2019, 7:42:17 PM Failed network lqi scan for device: '0x84182600000b88e1' with error: 'Error: Timed out after 10000 ms'

This happens for any device registered and after a few minutes all devices are set to unavailable. This does not happen with release version with same config and same db. There all devices are running fine

Config:
{
  "data_path": "/share/zigbee2mqtt",
  "homeassistant": true,
  "permit_join": false,
  "mqtt_base_topic": "zigbee2mqtt",
  "mqtt_server": "mqtt://homeassistant",
  "serial_port": "/dev/ttyUSB0",
  "devices": [],
  "network_key": [],
  "include_device_information": true,
  "availability_timeout": 60,
  "overwrite": true,
  "report": true,
  "rtscts": false
}

Full log: https://www.pastiebin.com/5d2a1ac864437

Koenkk, 2019-07-14:
This is a new feature, do you regular execute network scans?

dreimer1986, 2019-07-14:
I don't but: https://github.com/rgruebel/ha_zigbee2mqtt_networkmap does... So, yes there are scans on a regular basis.

Koenkk, 2019-07-14:
I wouldnt recommend executing network scans regularly as they put heavy stress on the network. They should be used as a tool for debugging.

Anyway @clockbrain could we perhaps move make it possible to separate these request? Perhaps it would be possible to append this to the mqtt topic to enable this feature.

charraeus, 2019-07-14:
After update from version 1.4.0 to 1.5.0 I have the same problem: 

>7/14/2019, 11:07:06 AM - warn: Network scan timeout, skipping outstanding lqi scans for ''
7/14/2019, 11:07:06 AM - warn: Network scan timeout, skipping outstanding rtg scans for '0x7cb03eaa00a7a4f5 0x7cb03eaa00ad05fb 0x7cb03eaa00ad293e 0x7cb03eaa00ad326d'
7/14/2019, 11:07:12 AM - warn: Failed network rtg scan for device: '0x7cb03eaa00a7a4f5' with error: 'Error: Timed out after 10000 ms'
7/14/2019, 11:07:12 AM - warn: Failed network rtg scan for device: '0x7cb03eaa00ad05fb' with error: 'Error: Timed out after 10000 ms'
7/14/2019, 11:07:13 AM - warn: Failed network rtg scan for device: '0x7cb03eaa00ad293e' with error: 'Error: Timed out after 10000 ms'
7/14/2019, 11:07:13 AM - warn: Failed network rtg scan for device: '0x7cb03eaa00ad326d' with error: 'Error: Timed out after 10000 ms'

I use the latest docker image. A network scan occurs every few seconds.

**My _configuration.yaml_:**

>homeassistant: false
permit_join: false

>mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://mqtt'

>serial:
  port: /dev/ttyACM0
  disable_led: false

>advanced:
  channel: 25
  network_key: [0xaf, 0xf6, 0x1c, 0x69, 0x76, 0x94, 0x2c, 0x8d, 0x89, 0x3f, 0x22, 0x94, 0x5b, 0x58, 0x6e, 0x57]
  log_level: warn

>devices: devices.yaml
>groups: groups.yaml

**My _devices.yaml_:**

>'0x7cb03eaa00a7a4f5':
  friendly_name: 'Fernsehlicht'
  retain: false
'0x7cb03eaa00ad05fb':
  friendly_name: 'Sofalicht'
  retain: false
'0x7cb03eaa00ad293e':
  friendly_name: 'Steckdose-3'
  retain: false
'0x7cb03eaa00ad326d':
  friendly_name: 'Steckdose-2'
  retain: false
'0x00158d0002c773b6':
  friendly_name: 'WSDCGQ01LM'
  retain: false`

charraeus, 2019-07-14:
This only occurs when the XIAOMI WSDCGQ01LM is paired. So I think it is another problem. I will open a separate issue.

ciB89, 2019-07-15:
Having the same problems

Koenkk, 2019-07-15:
Should be fixed in the latest dev branch, can anybody confirm? (for hassio please wait 2 hours till new edge version is published).

dreimer1986, 2019-07-15:
https://www.pastiebin.com/5d2ce404ae59a Confirmed. Running newest EDGE on Hass.IO 0.96B2. Still shows up once (Line 208-211), but the devices stay alive. Some other funny stuff is happening with some OSRAM device, but thats not for this report. (Line 484, 486, 619, 621)

Koenkk, 2019-07-16:
Thanks for confirming.

I will release 1.5.1 with this fix asap.

closed 2019-07-16 by Koenkk