Koenkk/zigbee2mqtt #1730: Not connected to MQTT server

pzystorm, 2019-07-13:
**Sorry guys, of course I have seen the other topic here regarding the same issue "Not connected to MQTT server"... but this seems different.**

## What happened
The connection to the mqtt cannot be established

## What did you expect to happen
The connection to the mqtt can be established

## How to reproduce it (minimal and precise)
apt-get install mosquitto
Mosquitto default config has not been changed.

Ensure the service is running:
```
# netstat -pltn | grep mosq
tcp        0      0 0.0.0.0:1883            0.0.0.0:*               LISTEN      18272/mosquitto
```

Test message arrives:
`mosquitto_pub -h localhost -t test -m "hello world"`

```
$ mosquitto_sub -h localhost -t test
hello world
```

zigbee2mqtt config:
```
# cat data/configuration.yaml
homeassistant: false
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost:1883'
serial:
  port: /dev/ttyACM0
devices:
  '0x00158d*****':
    friendly_name: '0x00158d*****'
    retain: false
advanced:
  log_level: debug
```

Output of zigbee2mqtt:
```
  zigbee2mqtt:info 2019-7-13 22:23:06 Currently 1 devices are joined:
  zigbee2mqtt:info 2019-7-13 22:23:06 0x00158d***** (0x00158d*****): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
  zigbee2mqtt:info 2019-7-13 22:23:06 Zigbee: disabling joining new devices.
  zigbee2mqtt:info 2019-7-13 22:23:06 Connecting to MQTT server at mqtt://localhost:1883
  zigbee2mqtt:info 2019-7-13 22:23:06 zigbee-shepherd ready
  zigbee2mqtt:error 2019-7-13 22:23:16 Not connected to MQTT server!
  zigbee2mqtt:error 2019-7-13 22:23:26 Not connected to MQTT server!
  zigbee2mqtt:error 2019-7-13 22:23:36 Not connected to MQTT server!
  zigbee2mqtt:error 2019-7-13 22:23:38 Not connected to MQTT server!
  zigbee2mqtt:error 2019-7-13 22:23:38 Cannot send message: topic: 'zigbee2mqtt/0x00158d*****', payload: '{"temperature":26.29,"linkquality":39,"humidity":65.46,"pressure":954}
  zigbee2mqtt:error 2019-7-13 22:23:38 Not connected to MQTT server!
```

## Debug Info
zigbee2mqtt version: latest (git clone on 13th July, 10pm)
CC253X firmware version: latest (June 2019)
mosquitto: https://www.ubuntuupdates.org/package/core/trusty/universe/updates/mosquitto
npm version: 6.10.1
node version: v10.14.0
OS: Ubuntu 14.04.5 LTS (Trusty)

andreasbrett, 2019-07-14:
Are you sure debug log level is on? Your log does not contain a single debug line. Typically turning debugging in results in a lot of debug lines in the log.

pzystorm, 2019-07-14:
Here you are:
```
  zigbee2mqtt:info 2019-7-14 18:59:34 Currently 1 devices are joined:
  zigbee2mqtt:info 2019-7-14 18:59:34 0x00158d***** (0x00158d*****): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
  zigbee2mqtt:info 2019-7-14 18:59:34 Zigbee: disabling joining new devices.
  zigbee2mqtt:info 2019-7-14 18:59:34 Connecting to MQTT server at mqtt://127.0.0.1:1883
  zigbee2mqtt:info 2019-7-14 18:59:34 zigbee-shepherd ready
  zigbee2mqtt:debug 2019-7-14 18:59:34 Mounted the cieApp (epId 11)
  zigbee2mqtt:debug 2019-7-14 18:59:39 Received zigbee message of type 'attReport' with data '{"cid":"msTemperatureMeasurement","data":{"measuredValue":2510}}' of device 'lumi.weather' (0x00158d*****) of endpoint 1
  zigbee2mqtt:error 2019-7-14 18:59:39 Not connected to MQTT server!
  zigbee2mqtt:error 2019-7-14 18:59:39 Cannot send message: topic: 'zigbee2mqtt/0x00158d*****', payload: '{"temperature":25.1,"linkquality":39,"humidity":62.45,"pressure":954}
  zigbee2mqtt:debug 2019-7-14 18:59:39 Received zigbee message of type 'attReport' with data '{"cid":"msRelativeHumidity","data":{"measuredValue":5788}}' of device 'lumi.weather' (0x00158d*****) of endpoint 1
  zigbee2mqtt:error 2019-7-14 18:59:39 Not connected to MQTT server!
  zigbee2mqtt:error 2019-7-14 18:59:39 Cannot send message: topic: 'zigbee2mqtt/0x00158d*****', payload: '{"temperature":25.1,"linkquality":36,"humidity":57.88,"pressure":954}
```

Publishing to zigbee2mqtt works:

`mosquitto_pub -h localhost -t zigbee2mqtt -m "hello world"`
gives:
```
$ mosquitto_sub -h localhost -t zigbee2mqtt
hello world
```

andreasbrett, 2019-07-14:
Thanks. Anything in the mosquitto logs then?

pzystorm, 2019-07-14:
Exactly the same like I have posted. Nothing more.
I think, I have to debug on my own... 
I have already looked at  node_modules/mqtt/lib/connect/index.js and tried to figure out how the connection is opened... and where I can insert some debug statements.

Koenkk, 2019-07-14:
Zigbee2mqtt uses the mqtt library, perhaps you can try to make an example based on https://www.npmjs.com/package/mqtt#example.

pzystorm, 2019-07-14:
I think, this library is broken... or incompatible with my installed version...

In the folder /opt/zigbee2mqtt/node_modules/mqtt I have executed
`./mqtt.js pub -t 'test' -h '127.0.0.1' -p 1883 -C mqtt -m 'test1234'`

But this command is not succesful. There is an infinit loop... no errors... no debug messages... 
I need to press CTRL+C to exit this command.

Will continue debugging tomorrow...

pzystorm, 2019-07-14:
On this page https://www.npmjs.com/package/mqtt I have read:

> If you are connecting to a broker that supports only MQTT 3.1 (not 3.1.1 compliant), you should pass these additional options:
```
{
  protocolId: 'MQIsdp',
  protocolVersion: 3
}
```
My mosquitto broker is indeed only 3.1:
```
# mosquitto -h
mosquitto is an MQTT v3.1 broker.

Usage: mosquitto [-c config_file] [-d] [-h] [-p port]
```

Can you help me, where is best place to insert this options in zigbee2mqtt? Just as a quick&dirty solution for me to test if it works...

Koenkk, 2019-07-15:
@pzystorm what mosquitto version? I'm running 1.6.3 and this one seems to support 3.1.1:

``` 
/ # mosquitto -h
mosquitto version 1.6.3

mosquitto is an MQTT v3.1.1 broker
```

pzystorm, 2019-07-15:
I think I have version 1.5 from September 2018

```
# apt-cache showpkg mosquitto | head
Package: mosquitto
Versions: 
0.15-2+deb7u3ubuntu0.1 (/var/lib/apt/lists/de.archive.ubuntu.com_ubuntu_dists_trusty-updates_universe_binary-amd64_Packages) (/var/lib/apt/lists/security.ubuntu.com_ubuntu_dists_trusty-security_universe_binary-amd64_Packages) (/var/lib/dpkg/status)
```

```
# mosquitto -h
mosquitto is an MQTT v3.1 broker.

# strings /usr/sbin/mosquitto  | grep -E "2018"
2018-09-05 15:14:14+0000
```

Koenkk, 2019-07-15:
https://mosquitto.org/ChangeLog.txt shows that 3.1.1 support has been completed in version 1.4 (20150218). Did perhaps explicitly disable it? Are you sure you are running 1.5?

pzystorm, 2019-07-15:
My config is very small:
```
# cat /etc/mosquitto/mosquitto.conf | grep "^[^#]"
bind_address 0.0.0.0
persistence true
persistence_location /var/lib/mosquitto/
log_dest none
```
Thats the default config which comes with the apt-get install

Really dont know which version this is.
Then lets assume it is 0.15 - really old
If I look in your change log I see an entry from 20120205

Koenkk, 2019-07-15:
It's indeed very old, you can grab the latest version (1.6.3) for trusty from the PPA (https://launchpad.net/~mosquitto-dev/+archive/ubuntu/mosquitto-ppa).

closed 2019-09-13 by Koenkk