Koenkk/zigbee2mqtt #1731: Wrong attribute reporting on Gledopto GL-C-008 and bad CCT to RGB switching

doino-gretchenliev, 2019-07-14:
# Bug Report
## What happened
Gledopto GL-C-008 is always reporting all attributes rather than the active ones.
 
## What did you expect to happen and how to reproduce it
### Issue 1
When you send `ON` command(without attributes) to the device, the device turns on with the latest set attribute, either `color_temp` or `color_xy`, so the light goes either white(with some temperature) or color. So the device knows which attribute was last set and which attribute should be active. But at that point if you ask for status it will report all attributes including both `color_temp` or `color_xy`, so from that message you can't know which of them is actually active and what is the actual color or color temperature of the light. The same behaviour happens when the device is in `ON` state and you change from `color` to `color_temperature` mode, the light goes from color(e.g. blue) to only white. But the state is still reporting the last set `color_xy` as well as the `color_temperature`.
The only way of knowing, from the MQTT status, which color is active at the moment is to track them historically, but that will fail also if restart your app and device changes state during the restart. You have no way to sync the states at start, because of the lack of correct status reporting.

### Issue 2
When you change from `color_temperature`(e.g. cold white) to `color_xy`(e.g. blue), that activates the white leds(white channels accordingly) and deactivates the color ones. But when you change from `color_xy`(e.g. blue) to `color_temperature`(e.g. cold white), that activates the color leds(accordingly), but remains the white leds active. At that point the only way to disable the white leds and leave only color is to restart the device. On boot it will activate only the color leds.

## Debug Info
zigbee2mqtt version: 1.5.0
CC253X firmware version: 20190608

directman66, 2019-07-14:
Work begins if he sends a management team, for example
in the topic zigbee2mqtt / 0x00124b001ba6e3c0 / set must be sent
{"state": "ON", "brightness": 255, "color": {"x": 0.123, "y": 0.123}}
or
{"state": "ON", "brightness": 255, "color": {"hex": "# FF0000"}}.

doino-gretchenliev, 2019-07-16:
@directman66 This issue describes a problem with CCT to RGB switching and device reporting.

doino-gretchenliev, 2019-07-16:
I'm going to setup a ZigBee sniffer to check what is actually reported by the device and confirm whether the issue is `zigbee2mqtt` related or not.

directman66, 2019-07-16:
> @ directman66 Эта проблема описывает проблему с коммутацией CCT на RGB и отчетами об устройствах.
After sending these commands, the device is working normally, the color and temperature change

doino-gretchenliev, 2019-07-16:
@directman66 My instance is working as described above. From the manufacture date I can assume it's the latest model/firmware version. As described above I'm already using the payloads you mentioned.

Koenkk, 2019-07-18:
Is this the same as #1740?

doino-gretchenliev, 2019-07-19:
@Koenkk , kinda sound like my issue, but not completely. The main difference is that report says, that the bulb only reports last state, which is red color coordinates and *not* the white color temperature. My device is always reporting both parameter no matter what is the color or white temperature. I'm waiting for my `CC2531` to setup a sniffer and I'm definitely going to write a comment here about the outcome.

Koenkk, 2019-07-20:
Ok, so let's first try to fix it for the color temperature. I think we need to do the following.

- Find a function on how to convert color temperature to RGB.
- Also return new rgb values here: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/toZigbee.js#L338

If that works, we basically need to do the same for the color convert (converting rgb -> color_temp)

shodge12, 2019-07-31:
Hi,

Reporting against ver 1.4.0 (I did not see anything in the notes which would have changed this behavior in 1.51)

I too have a GL-C-008 and have to agree with @doino-gretchenliev, the device is not behaving as expected.

My expectations is that the RGB component and the W component would act as 2 independent controlled items.  This is not what I am observing.

Through a test bed set up in Node-red, I can add the following observations:

1.	This is likely more of a quirk with the unit but do not use brightness values of 0.  When in the context of white (with adding color_temp to the json) it flashes and returns to the previous level.  With the RGB context, it acks as a no-op.

2.	If starting from all off, separately, both the RGB and the W by itself come on and change brightness as expected.  Then also respond to state on / off as expected.

3.	If I command the W on then off, followed by a command to set RGB to a color and brightness (in a single json), the result is a combination of the RGB at its commanded color / brightness and the W coming on at its previous level.  A subsequent state off and they both go off; and issuing the RGB / brightness command again works correctly on the second time.

4.	There does not seem to be a difference in the state command sent to /.../device/set, /.../device/rgb/set, or /.../device/white/set.  All turn off ALL RGBW leds.

Lastly from my expectations above, I would like to turn on the RGB and the white on and off independently.  As I am not set up to look at the zigbee data I do not have an insight if this is possible or not.

I would be happy to run any additional tests for data as well.

Thanks for your time.

-Stan

Koenkk, 2019-08-01:
@shodge12 I think https://github.com/Koenkk/zigbee2mqtt/issues/1711#issuecomment-512185930 is interesting for you.

shodge12, 2019-08-02:
Yes #1711 gets to the heart of the issue.  The white and RGB components are sharing their “state” variables.  There should be as noted a separate white_state control...

-Stan

Koenkk, 2019-08-03:
Just to be sure, with state you mean the on/off variables?

shodge12, 2019-08-03:
Sure...

Since we are treating these as 2 different lights - a RGB and a white light; The json "state": variable in the payload of the MQTT packet should control the context of the data sent with it.

For example, if I send a json "brightness" variable with json RGB data the brightness command controls the RGB light.  If I send the "brightness" variable with no RGB data and send "color_temp", the brightness controls the white light.

However the json "state" controls BOTH.  Adding context of "color_temp" causes the packet to be ignored.

Status which is returned from the device shows: (Node_RED debug dump of the received MQTT packet)

object
topic: "zigbee2mqtt/LED_1"
payload: object
brightness: 64
state_white: "OFF"
color: object
x: 0.6142
y: 0.2599
color_temp: 650
state: "OFF"
qos: 0
retain: false
_msgid: "259e3f9f.de412"

This dump implies a "state_white" json element BUT I cannot get this to be accepted.  Controlling "state_white" and "state_rgb" (these occur in status packets returned after other RGB color commands) may provide the full control needed.

-Stan

Koenkk, 2019-08-04:
Reading a bit more about how Home Assistant does this (https://www.home-assistant.io/components/light.mqtt/#brightness-and-rgbw-support) I think we should treat it as one light instead of two. 

As I don't have this device (or any other device which has a white channel) it's hard for me to implement this. Therefore I ordered the device so I can fix this integration (from AliExpress so will take some time before it arrives).

shodge12, 2019-08-04:
@Koenkk - thanks for your support.  Treating it as one device is ok but again my desire is to have both the RGB section and the W section active at the same time.  Using the schema from HA, if the "white_value": supports 0 to turn it off this would work; although you would not have control to turn off the RGB independently.  (present implementation has "state": "OFF" turning off both RGB and W.

I'll ping this thread again in a month....  Keep us posted on what you find....

Thanks again for your support...
-Stan

BudBundi, 2019-09-16:
> Ok, so let's first try to fix it for the color temperature. I think we need to do the following.
>
>     * Find a function on how to convert color temperature to RGB.
>
>     * Also return new rgb values [...]

https://github.com/home-assistant/home-assistant/pull/23495 is maybe interesting for that purpose

shodge12, 2019-09-19:
The problem is not color temperature.  The problem is the 4th channel W.  Again these are RGBW lights.  When you control color temperature it MUST involve the RGB channels.  The W channel itself is a single non-changing color temperature.  If you turn it on (W) in color temperature control mode - it will make it brighter and pull the color temperature to whatever the W channel LEDs are.  IMHO color temperature is not related to the issue.  Specifically we need to be able to control the W channel directly and with the present interface - we cannot - at least not consistently.

-Stan

Koenkk, 2019-09-20:
I've improved the behaviour of the RGB + W controller.

- In home assistant also a white value slider will be discovered now
- When set to RGB, the controller will switch to the full RGB channel and disable the white channel
- When set to white, the controller will switch to the full white channel and disable the RGB channel

Koenkk, 2019-10-03:
Both issues should be fixed now in the latest dev branch. Assuming this can be closed now.

closed 2019-10-03 by Koenkk

Xan8, 2019-10-19:
> I've improved the behaviour of the RGB + W controller.
>
> * In home assistant also a white value slider will be discovered now
> * When set to RGB, the controller will switch to the full RGB channel and [...]

I don't think is working for me. It only disscovered the rgb channel, I don't find the white value slider.
Do you know what could happend or is it that I don't see the slider?

Koenkk, 2019-10-19:
Are you on the dev branch?

Xan8, 2019-10-21:
I'm at 1.6.0. Is that it?
By the way thanks for answering so fast.

Koenkk, 2019-10-22:
See https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html