Koenkk/zigbee2mqtt #1733: After pairing a Xiaomi WSDCGQ01LM

charraeus, 2019-07-14:
I have the following error messages, when a WSDCGQ01LM is paired: 

>7/14/2019, 11:07:06 AM - warn: Network scan timeout, skipping outstanding lqi scans for ''
7/14/2019, 11:07:06 AM - warn: Network scan timeout, skipping outstanding rtg scans for '0x7cb03eaa00a7a4f5 0x7cb03eaa00ad05fb 0x7cb03eaa00ad293e 0x7cb03eaa00ad326d'
7/14/2019, 11:07:12 AM - warn: Failed network rtg scan for device: '0x7cb03eaa00a7a4f5' with error: 'Error: Timed out after 10000 ms'
7/14/2019, 11:07:12 AM - warn: Failed network rtg scan for device: '0x7cb03eaa00ad05fb' with error: 'Error: Timed out after 10000 ms'
7/14/2019, 11:07:13 AM - warn: Failed network rtg scan for device: '0x7cb03eaa00ad293e' with error: 'Error: Timed out after 10000 ms'
7/14/2019, 11:07:13 AM - warn: Failed network rtg scan for device: '0x7cb03eaa00ad326d' with error: 'Error: Timed out after 10000 ms'

I use the latest docker image. A network scan occurs every few seconds.
My Firmware: CC2531ZNP-Prod.hex  (20190608)

**My _configuration.yaml_:**

>homeassistant: false
permit_join: false

>mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://mqtt'

>serial:
  port: /dev/ttyACM0
  disable_led: false

>advanced:
  channel: 25
  network_key: [0xaf, 0xf6, 0x1c, 0x69, 0x76, 0x94, 0x2c, 0x8d, 0x89, 0x3f, 0x22, 0x94, 0x5b, 0x58, 0x6e, 0x57]
  log_level: warn

>devices: devices.yaml
>groups: groups.yaml

**My _devices.yaml_:**

>'0x7cb03eaa00a7a4f5':
  friendly_name: 'Fernsehlicht'
  retain: false
'0x7cb03eaa00ad05fb':
  friendly_name: 'Sofalicht'
  retain: false
'0x7cb03eaa00ad293e':
  friendly_name: 'Steckdose-3'
  retain: false
'0x7cb03eaa00ad326d':
  friendly_name: 'Steckdose-2'
  retain: false
'0x00158d0002c773b6':
  friendly_name: 'WSDCGQ01LM-Cockpit'
  retain: false

_Originally posted by @charraeus in https://github.com/Koenkk/zigbee2mqtt/issues/1728#issuecomment-511195434_

charraeus, 2019-07-14:
And here is some additional logging info (beginning from start):

>7/14/2019, 4:31:04 PM - info: Logging to directory: '/app/data/log/2019-07-14.16-31-02'
7/14/2019, 4:31:04 PM - info: Starting zigbee2mqtt version 1.5.0 (commit #unknown)
7/14/2019, 4:31:04 PM - info: Starting zigbee-shepherd
7/14/2019, 4:31:07 PM - info: zigbee-shepherd started
7/14/2019, 4:31:07 PM - info: Coordinator firmware version: '20190608'
7/14/2019, 4:31:07 PM - info: Currently 5 devices are joined:
7/14/2019, 4:31:07 PM - info: Fernsehlicht (0x7cb03eaa00a7a4f5): AB32840 - OSRAM LIGHTIFY LED Classic B40 tunable white (Router)
7/14/2019, 4:31:07 PM - info: Sofalicht (0x7cb03eaa00ad05fb): AB3257001NJ - OSRAM Smart+ plug (Router)
7/14/2019, 4:31:07 PM - info: Steckdose-3 (0x7cb03eaa00ad293e): AB3257001NJ - OSRAM Smart+ plug (Router)
7/14/2019, 4:31:07 PM - info: Steckdose-2 (0x7cb03eaa00ad326d): AB3257001NJ - OSRAM Smart+ plug (Router)
7/14/2019, 4:31:07 PM - info: WSDCGQ01LM-Cockpit (0x00158d0002c773b6): unkown - undefined unknown (EndDevice)
7/14/2019, 4:31:07 PM - info: Zigbee: disabling joining new devices.
7/14/2019, 4:31:07 PM - info: Connecting to MQTT server at mqtt://mqtt
7/14/2019, 4:31:08 PM - info: zigbee-shepherd ready
7/14/2019, 4:31:08 PM - info: Connected to MQTT server
7/14/2019, 4:31:08 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
7/14/2019, 4:31:08 PM - info: MQTT publish: topic 'zigbee2mqtt/Fernsehlicht', payload '{"state":"OFF","linkquality":28,"brightness":128,"color":{"x":0.381,"y":0.377},"color_temp":370,"color_mode":2}'
7/14/2019, 4:31:08 PM - info: MQTT publish: topic 'zigbee2mqtt/Sofalicht', payload '{"state":"OFF","linkquality":60}'
7/14/2019, 4:31:08 PM - info: MQTT publish: topic 'zigbee2mqtt/Steckdose-3', payload '{"state":"ON","linkquality":18}'
7/14/2019, 4:31:08 PM - info: MQTT publish: topic 'zigbee2mqtt/Steckdose-2', payload '{"state":"ON","linkquality":23}'
7/14/2019, 4:31:08 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.5.0","commit":"unknown","coordinator":20190608,"log_level":"info","permit_join":false}'
7/14/2019, 4:31:27 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"devices","message":[{"ieeeAddr":"0x00124b0018e1a452","type":"Coordinator"},{"ieeeAddr":"0x7cb03eaa00a7a4f5","type":"Router","model":"AB32840","friendly_name":"Fernsehlicht","nwkAddr":61096,"manufId":48042,"manufName":"OSRAM","powerSource":"Mains (single phase)","modelId":"Classic B40 TW - LIGHTIFY","hwVersion":1,"swBuildId":"V1.04.12","dateCode":"20140331CNEF****"},{"ieeeAddr":"0x7cb03eaa00ad05fb","type":"Router","model":"AB3257001NJ","friendly_name":"Sofalicht","nwkAddr":11982,"manufId":48042,"manufName":"OSRAM","powerSource":"Mains (single phase)","modelId":"Plug 01"},{"ieeeAddr":"0x7cb03eaa00ad293e","type":"Router","model":"AB3257001NJ","friendly_name":"Steckdose-3","nwkAddr":5506,"manufId":48042,"manufName":"OSRAM","powerSource":"Mains (single phase)","modelId":"Plug 01"},{"ieeeAddr":"0x7cb03eaa00ad326d","type":"Router","model":"AB3257001NJ","friendly_name":"Steckdose-2","nwkAddr":46472,"manufId":48042,"manufName":"OSRAM","powerSource":"Mains (single phase)","modelId":"Plug 01"},{"ieeeAddr":"0x00158d0002c773b6","type":"EndDevice","friendly_name":"WSDCGQ01LM-Cockpit","nwkAddr":25697,"manufId":4151}]}'
7/14/2019, 4:31:29 PM - info: Successfully configured Sofalicht (0x7cb03eaa00ad05fb)
7/14/2019, 4:31:30 PM - info: Successfully configured Steckdose-3 (0x7cb03eaa00ad293e)
7/14/2019, 4:31:31 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"devices","message":[{"ieeeAddr":"0x00124b0018e1a452","type":"Coordinator"},{"ieeeAddr":"0x7cb03eaa00a7a4f5","type":"Router","model":"AB32840","fri
endly_name":"Fernsehlicht","nwkAddr":61096,"manufId":48042,"manufName":"OSRAM","powerSource":"Mains (single phase)","modelId":"Classic B40 TW - LIGHTIFY","hwVersion":1,"swBuildId":"V1.04.12","dateCode":"20140331CNEF****"},{"ieeeAddr":"0x7cb03eaa00ad05fb","type":"Router","model":"AB3257001NJ","friendly_name":"Sofalicht","nwkAddr":11982,"manufId":48042,"manufName":"OSRAM","powerSource":"Mains (single phase)","modelId":"Plug 01"},{"ieeeAddr":"0x7cb03eaa00ad293e","type":"Router","model":"AB3257001NJ","friendly_name":"Steckdose-3","nwkAddr":5506,"manufId":48042,"manufName":"OSRAM","powerSource":"Mains (single phase)","modelId":"Plug 01"},{"ieeeAddr":"0x7cb03eaa00ad326d","type":"Router","model":"AB3257001NJ","friendly_name":"Steckdose-2","nwkAddr":46472,"manufId":48042,"manufName":"OSRAM","powerSource":"Mains (single phase)","modelId":"Plug 01"},{"ieeeAddr":"0x00158d0002c773b6","type":"EndDevice","friendly_name":"WSDCGQ01LM-Cockpit","nwkAddr":25697,"manufId":4151}]}'
7/14/2019, 4:31:31 PM - info: MQTT publish: topic 'zigbee2mqtt/Steckdose-3', payload '{"state":"ON","linkquality":23}'
7/14/2019, 4:31:32 PM - info: MQTT publish: topic 'zigbee2mqtt/Sofalicht', payload '{"state":"OFF","linkquality":60}'
7/14/2019, 4:31:35 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"devices","message":[{"ieeeAddr":"0x00124b0018e1a452","type":"Coordinator"},{"ieeeAddr":"0x7cb03eaa00a7a4f5","type":"Router","model":"AB32840","friendly_name":"Fernsehlicht","nwkAddr":61096,"manufId":48042,"manufName":"OSRAM","powerSource":"Mains (single phase)","modelId":"Classic B40 TW - LIGHTIFY","hwVersion":1,"swBuildId":"V1.04.12","dateCode":"20140331CNEF****"},{"ieeeAddr":"0x7cb03eaa00ad05fb","type":"Router","model":"AB3257001NJ","frie [... cut ...]
7/14/2019, 4:32:05 PM - warn: Failed network rtg scan for device: '0x7cb03eaa00ad326d' with error: 'Error: Timed out after 10000 ms'
7/14/2019, 4:32:06 PM - warn: Failed network rtg scan for device: '0x7cb03eaa00ad293e' with error: 'Error: Timed out after 10000 ms'

(...)

>7/14/2019, 4:32:28 PM - warn: Network scan timeout, skipping outstanding lqi scans for '0x7cb03eaa00ad326d'
7/14/2019, 4:32:28 PM - warn: Network scan timeout, skipping outstanding rtg scans for '0x7cb03eaa00a7a4f5 0x7cb03eaa00ad05fb 0x7cb03eaa00ad293e 0x7cb03eaa00ad326d'

Koenkk, 2019-07-15:
Closing this as it is the same issue as #1728

closed 2019-07-15 by Koenkk