Koenkk/zigbee2mqtt #1734: 1.5.0 on Raspbian Stretch does not work with WeBee CC2530+CC2591 any more

TLBM, 2019-07-15:
# Bug Report
## What happened
Updated from v1.4.0 to 1.5.0 via update.sh
`"error: Error while starting zigbee-shepherd! (Error: request timeout)"`

[console output](https://pastebin.com/fz3gedqs)
[full log](https://pastebin.com/zQDBWD7K)
[NPM log](https://pastebin.com/N8jndXgM)

## What did you expect to happen
Normal operation like before update.
[normal log](https://pastebin.com/RWvaqGns)

## How to reproduce it (minimal and precise)
Update from 1.4.0 to 1.5.0 via update.sh or install 1.5.0
cd /opt/zigbee2mqtt && npm start

## Debug Info
zigbee2mqtt version: 1.5.0
CC253x firmware version: CC2530_CC2591_DEFAULT_20190608
```
serial:
  port: /dev/ttyS0
advanced:
  rtscts: true
```

Hello!

Currently I'm evaluating ZigBee with some Aqara sensors and wall switches on a Raspberry Pi 3B+ (Raspbian Stretch) with the UART connected to a WeBee CC2530+CC2591.
After the update it does not work anymore. Quick switch to an CC2531 USB stick and changing the port setting worked. Reverting to v1.4.0 also restores functionality.
Flashed the WeBee board again, installed zigbee2mqtt from scratch, no change.

holgerebert, 2019-07-15:
Hello,
I have the same issue with a CC2530. I have tested the following firmware versions without success:
- CC2530_DEFAULT_20190608
- CC2530_MAX_STABILITY_20190315
- CC2530_SOURCE_ROUTING_20190619
Reverting solves the problem immediately

Koenkk, 2019-07-15:
Can you provide the logs when running with `DEBUG=* npm start` on v1.4.0 AND v1.5.0?

dreimer1986, 2019-07-15:
Same problem here as mentioned in #1728 . I could try to provide logs today at home, if you would please explain where to set DEBUG= at HassIO. Maybe @TLBM is faster though

holgerebert, 2019-07-15:
log from 1.4.0

[](https://pastebin.com/hvZDwF0p)

log from 1.5.0

[](https://pastebin.com/b03XYRey)

holgerebert, 2019-07-15:
[log from 1.4.0](https://pastebin.com/hvZDwF0p)

[log from 1.5.0](https://pastebin.com/b03XYRey)

Koenkk, 2019-07-15:
@holgerebert I don't see debug logging, can you try running with `sudo DEBUG=* npm start` instead of `DEBUG=* sudo npm start`

holgerebert, 2019-07-15:
@Koenkk sorry, here are the full logs. I have only seen the some of the log entries above and they are identical to me

[log from 1.4.0](https://pastebin.com/diAkPf8a)

[log from 1.5.0](https://pastebin.com/gZHxxpVL)

TLBM, 2019-07-15:
[DEBUG npm 1.4.0 as pi - works](https://pastebin.com/VKndQq06)

[DEBUG npm 1.5.0 as root - Error while starting zigbee-shepherd! (Error: request timeout)](https://pastebin.com/CzZPxPGQ)
[DEBUG npm 1.5.0 as pi - Error: EACCES: permission denied, open '/opt/zigbee2mqtt/data/database.db'](https://pastebin.com/94kWPbFV)
sudo chown -R pi:pi /opt/zigbee2mqtt
[DEBUG npm 1.5.0 as pi - works](https://pastebin.com/gvJSdygs)
[DEBUG npm 1.5.0 as pi - works 2nd time](https://pastebin.com/gWWvfsBM)
[npm 1.5.0 as pi - Error while starting zigbee-shepherd! (Error: request timeout)](https://pastebin.com/tm9iXXGy)
[DEBUG npm 1.5.0 as pi - Error while starting zigbee-shepherd! (Error: request timeout) 206 lines](https://pastebin.com/s7e4usQM)
[DEBUG npm 1.5.0 as pi - Error while starting zigbee-shepherd! (Error: request timeout) 705 lines](https://pastebin.com/VHx9Uz6y)
[DEBUG npm 1.5.0 sudo mc - works](https://pastebin.com/VxgPcC28)
[DEBUG npm 1.5.0 sudo bash - Error while starting zigbee-shepherd! (Error: request timeout) 261 lines](https://pastebin.com/3VD0a5pZ)

[DEBUG npm 1.5.0 as pi - works](https://pastebin.com/wh30D4V4)
[DEBUG npm 1.5.0 as pi - Error while starting zigbee-shepherd! (Error: request timeout)](https://pastebin.com/SDbGzEb6)
[DEBUG npm 1.5.0 as pi - Error while starting zigbee-shepherd! (Error: request timeout)](https://pastebin.com/k0sFkJs0)

@Koenkk Sometimes it starts, but I can't see a pattern.

Koenkk, 2019-07-15:
I've probably found the issue, will try to publish a fix tonight.

TLBM, 2019-07-15:
Nice - Thank you!

Koenkk, 2019-07-15:
Should be fix in latest dev branch (for hassio please wait 2 hours until new image is published).

LouisMT, 2019-07-15:
I can confirm the fix is working using the `arm32v6-dev` Docker image.

TLBM, 2019-07-15:
Just tried it and it worked four times in a row and also after a reboot.
Thank you for this great and fast service and this whole project!

Is there a forum for generic questions which are not bug related?

Have a nice evening!

dreimer1986, 2019-07-15:
Works fine here. CC2530 still works after restart of zigbee2mqtt. Looks fine. Hass.IO 0.96B2

Koenkk, 2019-07-16:
@tlbm http://zigbee2mqtt.discourse.group/

I will release 1.5.1 with this fix asap.

closed 2019-07-16 by Koenkk

holgerebert, 2019-07-16:
works for me too. Thank you!

Koenkk, 2019-07-16:
1.5.1 released, @danielwelch can you update the hassio image?

HA-TB303, 2019-07-20:
There is no (hassio) update yet. Pretty anoying when your entire zigbee network is down.

Koenkk, 2019-07-20:
You can use edge for the time being.

HA-TB303, 2019-07-20:
Thanks. I'm afraid that didn't fix it. I'm on Hassio 0.96.2 in docker on Ubuntu 18.04.
The problem started right after the update to 0.96.2. I did not make any configuration changes.

Logs:

```
[Info] Configuration file found: /share/zigbee2mqtt/configuration.yaml
[Info] Configuration written to /share/zigbee2mqtt/configuration.yaml
2019-07-20T23:26:28: PM2 log: Launching in no daemon mode
2019-07-20T23:26:29: PM2 log: App [npm:0] starting in -fork mode-
2019-07-20T23:26:29: PM2 log: App [npm:0] online
> zigbee2mqtt@1.5.1 start /app
> node index.js
  zigbee2mqtt:info 7/20/2019, 11:26:32 PM Logging to directory: '/share/zigbee2mqtt/log/2019-07-20.23-26-31'
  zigbee2mqtt:info 7/20/2019, 11:26:32 PM Starting zigbee2mqtt version 1.5.1 (commit #unknown)
  zigbee2mqtt:info 7/20/2019, 11:26:32 PM Starting zigbee-shepherd
  zigbee2mqtt:info 7/20/2019, 11:26:39 PM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: request timeout)
  zigbee2mqtt:info 7/20/2019, 11:27:39 PM Starting zigbee-shepherd
  zigbee2mqtt:error 7/20/2019, 11:27:46 PM Error while starting zigbee-shepherd! (Error: request timeout)
  zigbee2mqtt:error 7/20/2019, 11:27:46 PM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 7/20/2019, 11:27:46 PM Failed to start
	{"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/cc-znp/index.js:208:22)\n    at Object.onceWrapper (events.js:286:20)\n    at CcZnp.emit (events.js:198:13)\n    at Timeout.<anonymous> (/app/node_modules/zigbee-herdsman/dist/cc-znp/index.js:192:18)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}
  zigbee2mqtt:error 7/20/2019, 11:27:46 PM Exiting...
npm
 ERR! code
 ELIFECYCLE
npm 
ERR! errno 1
npm ERR! zigbee2mqtt@1.5.1 start: `node index.js`
npm
 ERR! Exit status 1
npm ERR! 
npm ERR!
 Failed at the zigbee2mqtt@1.5.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-07-20T21_27_46_483Z-debug.log
2019-07-20T23:27:49: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2019-07-20T23:27:49: PM2 log: App [npm:0] starting in -fork mode-
2019-07-20T23:27:49: PM2 log: App [npm:0] online
``` 

Is this the same issue, or do I need to file a new bug?

Nosstrom, 2019-08-02:
Does anyone know why in the config when i put "experimental":  {"livolo" :true} when i i save it ,removes the lines entirely?