Koenkk/zigbee2mqtt #1737: Installation fails

monsterflo, 2019-07-15:
# Bug Report
following the tutorial ( except for sudo chown -R openhabian:openhabian /opt/zigbee2mqtt )
the installation fails with the following report

## What happened
[15:07:14] root@openHABianPi:/opt/zigbee2mqtt# npm install

> @serialport/bindings@2.0.8 install /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install EACCES: permission denied, access '/root/.npm'
gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/10.16.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/.node-gyp"
gyp WARN install got an error, rolling back install
gyp WARN install got an error, rolling back install
gyp ERR! configure error
gyp ERR! stack Error: EACCES: permission denied, mkdir '/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/.node-gyp'
gyp ERR! System Linux 4.9.35-v7+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings
gyp ERR! node -v v10.16.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/zigbee-herdsman/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@2.0.8 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the @serialport/bindings@2.0.8 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-07-15T13_09_45_523Z-debug.log

## What did you expect to happen
node-pre-gyp info ok
added 383 packages in 111.613s

## How to reproduce it (minimal and precise)
tried to install it several times. always with the same outcome

## Debug Info
zigbee2mqtt version: 1.5.0
CC253X firmware version: culfw Version v1.67

[error.txt](https://github.com/Koenkk/zigbee2mqtt/files/3392515/error.txt)

Koenkk, 2019-07-15:
It's a permission error: `prebuild-install WARN install EACCES: permission denied, access '/root/.npm'`, either fix the permission are execute `sudo npm install`.

monsterflo, 2019-07-15:
Hello Koenkk,
thanks for the answer. I ran the script as root and with sudo npm install. Same result.
Root should have sufficient privileges:
drwx------   8 root    root        4096 Jul 15 15:07 root

Regards
Florian

Koenkk, 2019-07-16:
Your `configuration.yaml` is not valid, see `base_topic` in http://www.zigbee2mqtt.io/configuration/configuration.html

battosai30, 2019-07-17:
I had the same problem (Banapi M2+) and I saw you release a 1.5.1 and know it seems ok, I don't know if it's linked ....

monsterflo, 2019-07-17:
Follow up:
After the 5th reboot and start it somehow managed to get started:
[15:56:00] root@openHABianPi:/opt/zigbee2mqtt# npm start

> zigbee2mqtt@1.5.0 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 7/17/2019, 3:56:13 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-07-17.15-56-12'
  zigbee2mqtt:info 7/17/2019, 3:56:13 PM Starting zigbee2mqtt version 1.5.0 (commit #31b8352)
  zigbee2mqtt:info 7/17/2019, 3:56:13 PM Starting zigbee-shepherd
  zigbee2mqtt:info 7/17/2019, 3:56:14 PM zigbee-shepherd started
  zigbee2mqtt:info 7/17/2019, 3:56:14 PM Coordinator firmware version: '20181024 '
  zigbee2mqtt:info 7/17/2019, 3:56:14 PM Currently 0 devices are joined:
  zigbee2mqtt:warn 7/17/2019, 3:56:14 PM `permit_join` set to `true` in configuration.yaml.
  zigbee2mqtt:warn 7/17/2019, 3:56:14 PM Allowing new devices to join.
  zigbee2mqtt:warn 7/17/2019, 3:56:14 PM Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 7/17/2019, 3:56:14 PM Zigbee: allowing new devices to join.
  zigbee2mqtt:info 7/17/2019, 3:56:14 PM Connecting to MQTT server at mqtt://localhost:1883
  zigbee2mqtt:info 7/17/2019, 3:56:15 PM zigbee-shepherd ready
  zigbee2mqtt:info 7/17/2019, 3:56:15 PM Connected to MQTT server
  zigbee2mqtt:info 7/17/2019, 3:56:15 PM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 7/17/2019, 3:56:15 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.5.0","commit":"31b8352","coordinator":20181024, "log_level":"info","permit_join":true}'
  zigbee2mqtt:info 7/17/2019, 3:58:54 PM Successfully reenabled joining
  zigbee2mqtt:info 7/17/2019, 4:01:34 PM Successfully reenabled joining

Unfortunately I can't pair any devices. The Xiaomi Bridge is offline (unplugged) and I even removed the batteries from the devices (Xiaomi window sensor, PIR and wireless switch). I followed the procedure on the site for supported devices.
Logs are empty except for the above.

I appreciate your help

Thanks
Florian

closed 2019-09-22 by stale[bot]