Koenkk/zigbee2mqtt #1740: Device Reporting Wrong Colour

boojew, 2019-07-16:
# Bug Report
When one of the Ecosmart bulbs reports in that it has been turned on, it always seems to report in as red ("x":0.749,"y":0.249) rather than it's actual color - warm white. Manually changing the colour of the bulb in HASS works and it shows the correct colour - additionally if I then set the colour back to red - the bulb does turn red. My guess is when I added support I messed up for warmth? Not sure

zigbee2mqtt version: 1.5 
CC253X firmware version: 20190608

Koenkk, 2019-07-18:
The color you see is probably a cached value, does it happen when e.g. change color -> change color temp -> color state is incorrect?

boojew, 2019-07-18:
Changing the colour always works. But going from on to off always shows
this behaviour. Even in private browsing and with a different browser.

Koenkk, 2019-07-18:
So when e.g:

- bulb is on and red
- turn bulb off
- turn bulb on
- bulb is not red anymore but still reports red -> is this the problem?

boojew, 2019-07-19:
1. Bulb is off -> actual bulb is off, hass shows off
2. Bulb is turned on -> actual bulb is white, hass shows red
3. Bulb changes colour -> hass and bulb show correct colour
4. Bulb is turned off -> see 2

I don’t know how to translate the XY colour into rgb, so not sure if
zigbee2mqtt is sending the red or hass is going something weird.

Thanks

Koenkk, 2019-07-20:
Can you provide me a log showing the problems and annotating the lines where wrong states get published? (make sure to set `log_level: debug`). http://www.zigbee2mqtt.io/configuration/configuration.html

closed 2019-09-18 by Koenkk