Koenkk/zigbee2mqtt #1742: How to add support for unsupported devices when running hassio?

QuocTuanNguyen1980, 2019-07-16:
Hi,
I'm running this in Hassio, and could not find the path for node_modules/zigbee-shepherd-converters/devices.js. Can you guide me how to install the zigbee-shepherd-converters? Thank you.

kiwinol, 2019-07-16:
You need to enable developer access for HASSIO.  This link will give you the basic process
[http://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices_on_hassio.html](url)

kvvoff, 2019-07-16:
Tonight also struggled with this problem. I found how to do it. It turns out it's very simple

QuocTuanNguyen1980, 2019-07-17:
> Tonight also struggled with this problem. I found how to do it. It turns out it's very simple

Can you guide me through, cause I'm still struggling :)

kvvoff, 2019-07-20:
@QuocTuanNguyen1980 read this in my blog (on russian): 
https://kvvhost.ru/2019/07/20/zigbee2mqtt-add-new-device-hassio/

My problem was that I only added 1 device to devices.js, and I need devices.js to be completely filled, like this:
https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js
+ you new device

closed 2019-09-25 by stale[bot]