Koenkk/zigbee2mqtt #1746: LLKZMK11LM "binding" to Hue dimmer

resitan, 2019-07-17:
# Bug Report
## What happened
I have a Philips Hue Dimmer Switch and a LLKZMK11LM Xiaomi relay (among many other stuff) on the zigbee network, all working fine. The relay acts as a router to the switch, I can see it from zigbee sniffing. After a short power outage to the relay, the following happened:
- the ON switch on the dimmer now directly turns on relay 1 on the relay 
- no other dimmer buttons work (OFF doesn't turn the relay off)
- dimmer switch fails to produce any message in zigbee2mqtt, debug logs show no incoming zigbee message
- the dimmer "thinks" its messages are going through: it blinks green after every button press
- the relay otherwise works normally

Sniffing zigbee traffic showed that the messages from the switch are propery forwarded to the coordinator, which seems to throw them away for some reason. 

(Off: imagine my wife face when our bedroom light switch refused to turn on the lights, but opened the garage door instead...)

## How to reproduce it (minimal and precise)
This is the second time it happens, first it was the power outage and I could resolve it only by resetting the Xiaomi relay. It now happened again out of the blue. 

## Debug Info
zigbee2mqtt version: Hassio Add-on 1.4.0
CC253X firmware version: CC2530_CC2591ZNP-Prod_20190223
Logs: [https://pastebin.com/GGtDz3yL](url) not very informative, I pressed the switch right before the relay reported "ON", but no message shown for the switch.
Pcap file: [https://wsi.li/EXIpb1fAtBPWEr/1561793](url)

Koenkk, 2019-07-18:
I would indeed expect that message `No. 5` would result in a MQTT message. Can you post the log when starting with `DEBUG=* npm start` of this?

closed 2019-09-16 by Koenkk

leonardpitzu, 2023-04-28:
I have exactly the same setup & problem