Koenkk/zigbee2mqtt #1748: Factory resetting Hue Bulbs with only app

ministryofsillywalks, 2019-07-17:
This is not really a bug report only the information that I didn't have to factory reset my hue bulbs to get them paired with my cc2531.
I had them running via the hue brigde but wanted to migrate them for a better mesh quality.
All I did was delete the lamps from the hue app on my android and after that they paired up with my cc2531 without any other input form me.
The hue device page reads, that you need to factory reset the lights using a remote (which I don't have)
https://www.zigbee2mqtt.io/devices/9290012573A
Maybe this info can be changed?

CypherMK, 2019-07-18:
Would be cool if the user would be able to reset through zigbee2mqtt.

Koenkk, 2019-07-18:
@ministryofsillywalks I've added this to the documentation, thanks!

@CypherMK for this we would need to sniff the traffic, note that zigbee2mqtt can of course not reset bulbs that are still connected to the hue bridge, so the use case may somewhat be limited.

closed 2019-07-18 by Koenkk