Koenkk/zigbee2mqtt #1750: "Device with modelID 'undefined' is not supported." for known device.

Jim-Noble, 2019-07-17:
I bought a batch of 4 Aqara brand door sensors. Three of them paired just fine, but the last one did not:

    7/17/2019, 8:57:07 PM - info: New device 'undefined' with address 0x00158d0003f13950 connected!
    7/17/2019, 8:57:07 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x00158d0003f13950","meta":{}}'
    7/17/2019, 8:57:07 PM - warn: Device with modelID 'undefined' is not supported.

All four units are externally identical, right down to the date code on the box (2019.05).

I've tried repeating the pairing process, but get the same result. Ditto after removing the device completely, and then re-adding. 

The device does not function as expected either (where as the others do). Separating device and magnet either isn't detected or isn't reported as far as I can tell (I can hear the reed switch operate).

The database entry for the failed device:

{"id":24,"type":"EndDevice","ieeeAddr":"0x00158d0003f13950","nwkAddr":22108,"manufId":4151,"epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":24321,"inClusterList":[0,3,6,65535],"outClusterList":[0,4,65535],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"manuSpecificCluster":{"dir":{"value":3},"attrs":{}}}}},"_id":"171CFYd2qDwjCSQo"}

For comparison, the database entry for one of the successful additions from this batch:

{"id":23,"type":"EndDevice","ieeeAddr":"0x00158d0003f13989","nwkAddr":44028,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_magnet.aq2","epList":[1],"zclVersion":1,"appVersion":3,"stackVersion":2,"hwVersion":2,"swBuildId":"3000-0001","status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":24321,"inClusterList":[0,3,6,65535],"outClusterList":[0,4,65535],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"manuSpecificCluster":{"dir":{"value":3},"attrs":{}}}}},"dateCode":"20161128","_id":"xv47X3zyxPmQuQXJ"}

zigbee2mqtt version: 1.5
CC253X firmware version: 20190619 (source routing)

Defective device perhaps? 

Thanks,

James

Ps. As as aside, I have been having some significant issues with network stability, and device operation. This has recently changed though, and network stability at the moment is rock solid. The improvement is probably down to a combination of upgrading the coordinator firmware, (using the source routing version), the recent zigbee2mqtt release, and adding a USB extension to the sniffer to move it away from a wifi basestation. 
I've been keeping an eye on the network map, and the topology hasn't changed at all in the few days prior to adding these new devices, which is something I've not experienced up until now. The new devices haven't affected the stability so far either.  
However I am (still) finding significant issues getting most of the door/window sensor devices to actually send any status updates. Running with full debug there's no relevant output when a door is repeatedly opened and closed. The 3 working devices just added are currently working just fine though. 
If/when I get to the point of having something quantifiable to report, I'll file a separate issue - at the moment I figure "It doesn't work" isn't particularly useful.

Jim-Noble, 2019-07-17:
Ok, I should have read all the related issues. I tried the method reported here:

https://github.com/Koenkk/zigbee2mqtt/issues/934

ie. 
1. Remove the device with a zigbee2mqtt/bridge/config/remove message.
2. Stop the zigbee2mqtt service.
3. Remove the entries in database.db related to the device.
4. Start zigbee2mqtt
5. Enable pairing.
6. Pair the device again

Voila! It's now paired correctly :-)

James

closed 2019-07-17 by Jim-Noble

radumpopescu, 2019-08-23:
For me it didn't work what you did, however I realised that in database.db, the entry didn't have a `modelId` so I wrote it manually, restarted and now it works.
I had problems with Xiaomi ZNCZ02LM with the `lumi.plug` modelId however, I had issues with other devices as well that didn't pair from first try.