Koenkk/zigbee2mqtt #1753: Impossible to use systemctl service

mitch55500, 2019-07-19:
Hi everyone, 

I've installed Jeedom with docker (I'm on a Synology). 
Everything works with zigbee2mqtt.

But I can't access to systemctl service. 
These lines are added
`[Unit]
Description=zigbee2mqtt
After=network.target

[Service]
ExecStart=/usr/bin/npm start
WorkingDirectory=/opt/zigbee2mqtt
StandardOutput=inherit
StandardError=inherit
Restart=always
User=root

[Install]
WantedBy=multi-user.target`

When I verify with these commands "sudo systemctl start zigbee2mqtt" or "systemctl status zigbee2mqtt.service" I have this message : " Failed to connect to bus: No such file or directory".

What's my problem?

JLFN, 2019-07-20:
Install portainer and you will manage. those commands are not for docker installation.

mitch55500, 2019-07-23:
Thanks for your help. 
It's done. Now I can see Jeedom installed with docker.

Could you tell me more? How to manage?

JLFN, 2019-07-23:
Ps basic usage you need to read docker manual. No step by step help here at this level.

closed 2019-09-28 by stale[bot]