Koenkk/zigbee2mqtt #1756: Please help me

skxxx, 2019-07-21:
zigbee2mqtt:info 7/21/2019, 9:41:06 AM Logging to directory: '/share/zigbee2mqtt/log/2019-07-21.09-41-05'
  zigbee2mqtt:info 7/21/2019, 9:41:06 AM Starting zigbee2mqtt version 1.5.1 (commit #unknown)
  zigbee2mqtt:info 7/21/2019, 9:41:06 AM Starting zigbee-shepherd
/zigbee2mqtt-1.5.1/node_modules/zigbee-herdsman/dist/lib/components/controller.js:429
                    throw new TypeError('net.precfgkey should be an array with 16 uint8 integers.');
                    ^
TypeError: net.precfgkey should be an array with 16 uint8 integers.
    at /zigbee2mqtt-1.5.1/node_modules/zigbee-herdsman/dist/lib/components/controller.js:429:27
    at Object._.forOwn (/zigbee2mqtt-1.5.1/node_modules/zigbee-herdsman/node_modules/busyman/index.js:242:27)
    at Object._.forEach (/zigbee2mqtt-1.5.1/node_modules/zigbee-herdsman/node_modules/busyman/index.js:415:18)
    at Controller.setNvParams (/zigbee2mqtt-1.5.1/node_modules/zigbee-herdsman/dist/lib/components/controller.js:414:7)
    at new ZShepherd (/zigbee2mqtt-1.5.1/node_modules/zigbee-herdsman/dist/lib/shepherd.js:28:21)
    at Zigbee.start (/zigbee2mqtt-1.5.1/lib/zigbee.js:57:25)
    at startupLogVersion (/zigbee2mqtt-1.5.1/lib/controller.js:167:25)
    at utils.getZigbee2mqttVersion (/zigbee2mqtt-1.5.1/lib/controller.js:193:13)
    at git.getLastCommit (/zigbee2mqtt-1.5.1/lib/util/utils.js:102:9)
    at /zigbee2mqtt-1.5.1/node_modules/git-last-commit/source/index.js:44:5
npm ERR! code ELIFECYCLE
npm ERR! errno
 1
npm ERR! zigbee2mqtt@1.5.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.5.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-07-21T07_41_06_845Z-debug.log
2019-07-21T09:41:07: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2019-07-21T09:41:07: PM2 log: App [npm:0] starting in -fork mode-
2019-07-21T09:41:07: PM2 log: App [npm:0] online
> zigbee2mqtt@1.5.1 start /zigbee2mqtt-1.5.1
> node index.js

cretzu8523, 2019-07-21:
i have the same problem.....

bmangodt, 2019-07-21:
Same issue here... reinstall did not work either

Koenkk, 2019-07-21:
Can you post your configuration.yaml?

skxxx, 2019-07-21:
after update zigbee2mqtt 1.5.1...all stopped
But now im using old version 1.4.0 and all working.

bmangodt, 2019-07-21:
Maybe just a FYI but I'm using the hassio add-on version... yesterday was working fine and today it stopped. I have no idea what version I was running before as the "auto update" feature was on...

Koenkk, 2019-07-21:
@skkx are you also using hassio?

DerMattes, 2019-07-21:
Same error here, using hass.io. This is my configuration.yaml:

`{
  "devices": "[]",
  "groups": "groups.yaml",
  "homeassistant": true,
  "permit_join": false,
  "mqtt": {
    "base_topic": "zigbee2mqtt",
    "server": "mqtt://homeassistant",
    "user": "user",
    "password": "password"
  },
  "serial": {
    "port": "/dev/ttyACM0"
  },
  "advanced": {
    "pan_id": 1234,
    "channel": 11,
    "network_key": []
  }
}`

I just censored user, pwd and pan_id ;-)

HA-TB303, 2019-07-21:
I have the same issue, but even after restoring to a snapshot form before this update, containing zigbee2mqtt 1.4, zigbee-shepherd won't start. :(

RikVerstijnen, 2019-07-21:
Same issue here

gonzasmts, 2019-07-21:
same issue here

NullEnt1ty, 2019-07-21:
**Version 1.5.1 introduced a new config format! You need to adjust your configuration.**

Example:

Before:
```json
{
  "data_path": "/share/zigbee2mqtt",
  "devices": [],
  "groups": "groups.yaml",
  "homeassistant": true,
  "permit_join": false,
  "serial": {
    "port": "/dev/ttyACM0"
  },
  "advanced": {
    "pan_id": 6754,
    "channel": 11
  },
  "mqtt_base_topic": "zigbee2mqtt",
  "mqtt_server": "mqtt://localhost:1883",
  "serial_port": "/dev/ttyACM0",
  "disable_led": true,
  "network_key": [1, 2, 3, 4, 5, 6, 7, 8]
}
```

After:
```json
{
  "data_path": "/share/zigbee2mqtt",
  "devices": [],
  "groups": "groups.yaml",
  "homeassistant": true,
  "permit_join": false,
  "mqtt": {
    "base_topic": "zigbee2mqtt",
    "server": "mqtt://localhost:1883"
  },
  "serial": {
    "port": "/dev/ttyACM0",
    "disable_led": true
  },
  "advanced": {
    "pan_id": 6754,
    "channel": 11,
    "network_key": [1, 2, 3, 4, 5, 6, 7, 8]
  }
}
```

Notice that some keys are now nested. For example `network_key` is now located in the `advanced` section.

bmangodt, 2019-07-21:
> **Version 1.5.1 introduced a new config format! You need to adjust your configuration.**
> 
> Example:
> 
> Before:
> 
> ```json
> {
>   "data_path": "/share/zigbee2mqtt",
>   "devices": [],
>   "groups": "groups.yaml",
>   "homeassistant": true,
>   "permit_join": false,
>   "serial": {
>     "port": "/dev/ttyACM0"
>   },
>   "advanced": {
>     "pan_id": 6754,
>     "channel": 11
>   },
>   "mqtt_base_topic": "zigbee2mqtt",
>   "mqtt_server": "mqtt://localhost:1883",
>   "serial_port": "/dev/ttyACM0",
>   "disable_led": true,
>   "network_key": [1, 2, 3, 4, 5, 6, 7, 8]
> }
> ```
> 
> After:
> 
> ```json
> {
>   "data_path": "/share/zigbee2mqtt",
>   "devices": [],
>   "groups": "groups.yaml",
>   "homeassistant": true,
>   "permit_join": false,
>   "mqtt": {
>     "base_topic": "zigbee2mqtt",
>     "server": "mqtt://localhost:1883"
>   },
>   "serial": {
>     "port": "/dev/ttyACM0",
>     "disable_led": true
>   },
>   "advanced": {
>     "pan_id": 6754,
>     "channel": 11,
>     "network_key": [1, 2, 3, 4, 5, 6, 7, 8]
>   }
> }
> ```
> 
> Notice that some keys are now nested. For example `network_key` is now located in the `advanced` section.
> 
> You can find the current config format here: http://www.zigbee2mqtt.io/configuration/configuration.html

Yes I was aware of that issue but even with the new config after a complete reinstallation and just changing the minimum required things to make it work, it still throws the same error... so I think the configuration is not the issue here...

HA-TB303, 2019-07-21:
I have also made the required changes, but the same errors.
```
{
  "data_path": "/share/zigbee2mqtt",
  "devices": "[]",
  "groups": "groups.yaml",
  "homeassistant": true,
  "permit_join": false,
  "mqtt": {
    "base_topic": "zigbee2mqtt",
    "server": "mqtt://192.168.2.101",
    "user": "zigbee2mqtt",
    "password": "1234"
  },
  "serial": {
    "port": "/dev/ttyUSB0"
  },
  "advanced": {
    "pan_id": 6754,
    "channel": 11,
    "rtscts": false,
    "network_key": [
      11,
      3,
      13,
      3,
      11,
      9,
      13,
      8,
      10,
      12,
      17,
      7,
      2,
      6,
      2,
      15
    ]
  }
}
```

Network key and password has been changed.

gonzasmts, 2019-07-21:
for me updating the network_key solved the problem, is now working.

"advanced": {
    "pan_id": 6754,
    "channel": 11,
    "network_key": [
      1,
      3,
      5,
      7,
      9,
      11,
      13,
      15,
      0,
      2,
      4,
      6,
      8,
      10,
      12,
      13
    ]
  }

bmangodt, 2019-07-21:
network_key did the trick!!! Huge thanks!

> for me updating the network_key solved the problem, is now working.
> 
> "advanced": {
> "pan_id": 6754,
> "channel": 11,
> "network_key": [1, 3, 5, 7, 9, 11, 13, 15, 0, 2, 4, 6, 8, 10, 12, 13 ]
> }

DerMattes, 2019-07-21:
I think I did not configure a network key before, so what would be a solution then?

NullEnt1ty, 2019-07-21:
> 
> 
> I think I did not configure a network key before, so what would be a solution then?

[Configuration page](http://www.zigbee2mqtt.io/configuration/configuration.html) says that `network_key` is optional. So you should be able to omit it from your configuration. Although I would recommend configuring it because it improves security. Note that changing the key requires a re-pair of all of your devices.

HA-TB303, 2019-07-21:
huh?

If I change the network key to:
```
{
  "data_path": "/share/zigbee2mqtt",
  "devices": "[]",
  "groups": "groups.yaml",
  "homeassistant": true,
  "permit_join": false,
  "mqtt": {
    "base_topic": "zigbee2mqtt",
    "server": "mqtt://192.168.2.101:1883",
    "user": "zigbee2mqtt",
    "password": "Qwe1234"
  },
  "serial": {
    "port": "/dev/ttyUSB0",
    "disable_led": true
  },
  "advanced": {
    "pan_id": 6754,
    "channel": 11,
    "rtscts": false,
    "network_key": [18, 3, 11, 3, 15, 9, 13, 8, 10, 12, 11, 7, 2, 6, 2, 13 ]
  }
}
```

and press 'save' it will reformat to:

{
  "data_path": "/share/zigbee2mqtt",
  "devices": "[]",
  "groups": "groups.yaml",
  "homeassistant": true,
  "permit_join": false,
  "mqtt": {
    "base_topic": "zigbee2mqtt",
    "server": "mqtt://192.168.2.101:1883",
    "user": "zigbee2mqtt",
    "password": "Qwe1234"
  },
  "serial": {
    "port": "/dev/ttyUSB0",
    "disable_led": true
  },
  "advanced": {
    "pan_id": 6754,
    "channel": 11,
    "rtscts": false,
    "network_key": [
      18,
      3,
      11,
      3,
      15,
      9,
      13,
      8,
      10,
      12,
      11,
      7,
      2,
      6,
      2,
      13
    ]
  }
}

NullEnt1ty, 2019-07-21:
@HA-TB303 that's perfectly fine.

```
[1, 2, 3, 4]

and

[
  1,
  2,
  3,
  4
]
```

...is semantically the same thing.

Do you get a new type of error after your config adjustment?

DerMattes, 2019-07-21:
> [Configuration page](http://www.zigbee2mqtt.io/configuration/configuration.html) says that `network_key` is optional. So you should be able to omit it from your configuration. Although I would [...]

Yes, but the configuration page seems outdated to me, as it is still im yaml.

HA-TB303, 2019-07-21:
````
./run.sh: line 13: [Info] Configuration backup found in /share/zigbee2mqtt/.configuration.yaml.bk. Skipping config backup.: No such file or directory
2019-07-21T13:25:23: PM2 log: Launching in no daemon mode
2019-07-21T13:25:23: PM2 log: App [npm:0] starting in -fork mode-
2019-07-21T13:25:23: PM2 log: App [npm:0] online
> zigbee2mqtt@1.5.1 start /zigbee2mqtt-1.5.1
> node index.js
  zigbee2mqtt:info 7/21/2019, 1:25:26 PM Logging to directory: '/share/zigbee2mqtt/log/2019-07-21.13-25-25'
  zigbee2mqtt:info 7/21/2019, 1:25:26 PM Starting zigbee2mqtt version 1.5.1 (commit #unknown)
  zigbee2mqtt:info 7/21/2019, 1:25:26 PM Starting zigbee-shepherd
  zigbee2mqtt:info 7/21/2019, 1:25:33 PM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: request timeout)
  zigbee2mqtt:info 7/21/2019, 1:26:33 PM Starting zigbee-shepherd
  zigbee2mqtt:error 7/21/2019, 1:26:40 PM Error while starting zigbee-shepherd! (Error: request timeout)
  zigbee2mqtt:error 7/21/2019, 1:26:40 PM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 7/21/2019, 1:26:40 PM Failed to start
	{"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/zigbee2mqtt-1.5.1/node_modules/zigbee-herdsman/dist/cc-znp/index.js:208:22)\n    at Object.onceWrapper (events.js:286:20)\n    at CcZnp.emit (events.js:198:13)\n    at Timeout.<anonymous> (/zigbee2mqtt-1.5.1/node_modules/zigbee-herdsman/dist/cc-znp/index.js:192:18)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}
  zigbee2mqtt:error 7/21/2019, 1:26:40 PM Exiting...
npm 
ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.5.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.5.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-07-21T11_26_40_638Z-debug.log
2019-07-21T13:26:43: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2019-07-21T13:26:43: PM2 log: App [npm:0] starting in -fork mode-
2019-07-21T13:26:43: PM2 log: App [npm:0] online
> zigbee2mqtt@1.5.1 start /zigbee2mqtt-1.5.1
> node index.js
  zigbee2mqtt:info 7/21/2019, 1:26:46 PM Logging to directory: '/share/zigbee2mqtt/log/2019-07-21.13-26-46'
  zigbee2mqtt:info 7/21/2019, 1:26:47 PM Starting zigbee2mqtt version 1.5.1 (commit #unknown)
  zigbee2mqtt:info 7/21/2019, 1:26:47 PM Starting zigbee-shepherd
  zigbee2mqtt:info 7/21/2019, 1:26:54 PM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: request timeout)
  zigbee2mqtt:info 7/21/2019, 1:27:54 PM Starting zigbee-shepherd
  zigbee2mqtt:error 7/21/2019, 1:28:01 PM Error while starting zigbee-shepherd! (Error: request timeout)
  zigbee2mqtt:error 7/21/2019, 1:28:01 PM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 7/21/2019, 1:28:01 PM Failed to start
	{"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/zigbee2mqtt-1.5.1/node_modules/zigbee-herdsman/dist/cc-znp/index.js:208:22)\n    at Object.onceWrapper (events.js:286:20)\n    at CcZnp.emit (events.js:198:13)\n    at Timeout.<anonymous> (/zigbee2mqtt-1.5.1/node_modules/zigbee-herdsman/dist/cc-znp/index.js:192:18)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}
  zigbee2mqtt:error 7/21/2019, 1:28:01 PM Exiting...
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.5.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.5.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-07-21T11_28_01_301Z-debug.log
2019-07-21T13:28:04: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2019-07-21T13:28:04: PM2 log: App [npm:0] starting in -fork mode-
2019-07-21T13:28:04: PM2 log: App [npm:0] online
> zigbee2mqtt@1.5.1 start /zigbee2mqtt-1.5.1
> node index.js
  zigbee2mqtt:info 7/21/2019, 1:28:07 PM Logging to directory: '/share/zigbee2mqtt/log/2019-07-21.13-28-06'
  zigbee2mqtt:info 7/21/2019, 1:28:07 PM Starting zigbee2mqtt version 1.5.1 (commit #unknown)
  zigbee2mqtt:info 7/21/2019, 1:28:07 PM Starting zigbee-shepherd
  zigbee2mqtt:info 7/21/2019, 1:28:14 PM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: request timeout)
```

I have a real issue now, reverting to a snapshot from before this update worn't work eigther and a large portion of my house automation is depending on zigbee :(

gonzasmts, 2019-07-21:
@HA-TB303 did you reset the stick?

HA-TB303, 2019-07-21:
I have a cc2530+cc2591 and a CP2102 rs232 to usb convertor.

I worked just fine until before the update. I can't press reset, but I have tried rebooting, re-inserting etc.

NullEnt1ty, 2019-07-21:
@DerMattes 
> Yes, but the configuration page seems outdated to me, as it is still im yaml.

I am assuming you're using Hass.io. The zigbee2mqtt Hass.io add-on defaults `network_key` to `[]` if it is omitted, but it won't pass the empty `network_key` to zigbee2mqtt.

At least that's what the changelog says:

> Add `network_key` to options, defaulting to empty list. If left empty, the option will not be passed to zigbee2mqtt.

If you're observing a different behavior, I would suggest, that you open an issue [here](https://github.com/danielwelch/hassio-zigbee2mqtt).

NullEnt1ty, 2019-07-21:
> 
> 
> I have a cc2530+cc2591 and a CP2102 rs232 to usb convertor.
> 
> I worked just fine until before the update. I can't press reset, but I have tried rebooting, re-inserting etc.

I had the same error after the update. But a system reboot fixed this issue for me. Hmm... strange.

Are you using the latest firmware for your ZigBee-sniffer?

HA-TB303, 2019-07-21:
I thing the version from May this year. I'll trie re-flashing it, but last time I had to re-pair all my devices and that was nu fun ;)

NullEnt1ty, 2019-07-21:
> 
> 
> I thing the version from May this year. I'll trie re-flashing it, but last time I had to re-pair all my devices and that was no fun ;)

The firmware from May should be fine (at least the changelog does not say that a newer firmware is required).

I found this in your log:
> zigbee2mqtt:info 7/21/2019, 1:28:07 PM Logging to directory: '/share/zigbee2mqtt/log/2019-07-21.13-28-06'

Can you check if there are additional log entries in `/share/zigbee2mqtt/log/2019-07-21.13-28-06`?

HA-TB303, 2019-07-21:
Thanks for helping me out. Before you wrote that I had already re-flashed the coordinator to firmware version 20190608. When removing the stick zigbee2mqtt restarted repeatedly and therefore rotated the logs.

After inserting it and restarting zigbee2mqtt it worked. I just don't understand how the coordinator could have corrupted itself in the zigbee2mqtt update process. I am 100% sure this problem started right after the update.

```
./run.sh: line 13: [Info] Configuration backup found in /share/zigbee2mqtt/.configuration.yaml.bk. Skipping config backup.: No such file or directory
2019-07-21T13:56:34: PM2 log: Launching in no daemon mode
2019-07-21T13:56:34: PM2 log: App [npm:0] starting in -fork mode-
2019-07-21T13:56:34: PM2 log: App [npm:0] online
> zigbee2mqtt@1.5.1 start /zigbee2mqtt-1.5.1
> node index.js
  zigbee2mqtt:info 7/21/2019, 1:56:37 PM Logging to directory: '/share/zigbee2mqtt/log/2019-07-21.13-56-37'
  zigbee2mqtt:info 7/21/2019, 1:56:37 PM Starting zigbee2mqtt version 1.5.1 (commit #unknown)
  zigbee2mqtt:info 7/21/2019, 1:56:37 PM Starting zigbee-shepherd
  zigbee2mqtt:info 7/21/2019, 1:56:45 PM zigbee-shepherd started
  zigbee2mqtt:info 7/21/2019, 1:56:45 PM Coordinator firmware version: '20190608'
```
Anyway, it works and thank you all for helping!

NullEnt1ty, 2019-07-21:
Great to hear, you're welcome!

HA-TB303, 2019-07-21:
Now the horror of re-pairing all my devices except strangely enough the (Ikea routers e.g. the 2 bulbs and 1 repeater). Those do work.

All xiaomi devices need re-pairing. 👎

douglasjam, 2019-07-22:
adding the values to the network key worked for me
https://github.com/Koenkk/zigbee2mqtt/issues/1756#issuecomment-513544632

bmtKIA6, 2019-07-23:
@NullEnt1ty are you stating that we require a re-flash after the latest upgrade?
Is there a way to validate that the stick requires a reflash?

it seems that after the hass.io upgrade to 0.96.2 and zigbee2mqtt addon to 1.5.1.1. I can no longer pair the devices.

I tried resetting the devices and try to re-pair, w/o any luck.

NullEnt1ty, 2019-07-24:
@bmtKIA6 No, I'm not stating that. I was using [this firmware](https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_Home_1.2/bin/default/CC2531_DEFAULT_20190608.zip) before the update and everything is still working as far as I can tell. I did **not** need to reflash the firmware.

bmtKIA6, 2019-07-25:
@NullEnt1ty Great. I am on the same firmware, but it took way longer to pair, than before. I think the logging is now also slower. I suspect that the devices do pair up, and the logging is lagging.

dirstel, 2019-08-08:
I had some trouble when switching from 1.4 to 1.5.1 too. The reason has been a messed up group configuration, which I had been playing with. So make sure to have a look at the groups configuration when migrating.

closed 2019-10-14 by stale[bot]

freire-nuno, 2020-01-05:
Upgrading from version 1.4 to 1.8 only worked with the steps below.

Old config:
```
{
  "data_path": "/share/zigbee2mqtt",
  "homeassistant": true,
  "permit_join": false,
  "mqtt_base_topic": "zigbee2mqtt",
  "mqtt_server": "mqtt://homeassistant",
  "serial_port": "/dev/ttyACM0",
  "devices": [],
  "network_key": [],
  "include_device_information": true,
  "mqtt_user": "mqtt",
  "mqtt_pass": "helloworld"
}
```

New Config:
```
{
  "data_path": "/share/zigbee2mqtt",
  "devices": "devices.yaml",
  "groups": "groups.yaml",
  "homeassistant": true,
  "permit_join": false,
  "mqtt": {
    "base_topic": "zigbee2mqtt",
    "server": "mqtt://homeassistant",
    "user": "mqtt",
    "password": "helloworld"
  },
  "serial": {
    "port": "/dev/ttyACM0"
  },
  "advanced": {
    "pan_id": 6754,
    "channel": 11,
    "network_key": [
      1,
      3,
      5,
      7,
      9,
      11,
      13,
      15,
      0,
      2,
      4,
      6,
      8,
      10,
      12,
      13
    ],
    "availability_blacklist": []
  },
  "ban": [],
  "whitelist": [],
  "queue": {},
  "socat": {
    "enabled": false,
    "master": "pty,raw,echo=0,link=/dev/ttyZ2M,mode=777",
    "slave": "tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5",
    "restartdelay": 1,
    "initialdelay": 1,
    "options": "-d -d",
    "log": false
  }
}
```
Steps:

1. add `devices.yaml` with content from `devices:` in `configuration.yaml`
```
  '0x00158d0002b546db':
    friendly_name: TempHumPressSensorRoom2
    retain: false
  '0x00158d0002c13bda':
    friendly_name: TempHumPressSensorRoom1
    retain: false
```
2. define network_key with the default value:
`network_key : [1, 3, 5, 7, 9, 11, 13, 15, 0, 2, 4, 6, 8, 10, 12, 13]`

3. guarantee the same channel as it was before