Koenkk/zigbee2mqtt #1758: Allow manually restart of configuration of EndDevice

tb-killa, 2019-07-21:
To be able to test during debugging, but also on devices that sometimes cause problems during configuration (e.g. DeepSleep), it would be appropriate to reset the status of the configuration.

The extension of existing mqtt-topics such as ``bridge`` would be ideal.

tb-killa, 2019-09-21:
hi @Koenkk 
i see the new zigbee-herdsman-converter use some sort of ``meta`` field like: 
``meta: {configureKey: 1},`` 
btw: we need some type of development docu for the converters to know what this type of data would mean ;).

Do you think we could add this part as soon on new herdsman :) ?

Koenkk, 2019-09-22:
hi @tb-killa, yes it should be added to https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html (will update that before 1.7 release).

This configureKey indicates if zigbee2mqtt has to configure the device again (e.g. due to change a change in the `configure` section).

closed 2019-11-28 by stale[bot]