Koenkk/zigbee2mqtt #1759: Error while starting zigbee-shepherd! (Error: request timeout)

shane151, 2019-07-21:
Just set this up 4 days ago running zigbee2mqtt in docker, and was working great with Xiaomi buttons (WXKG12LM, WXKG01LM), and plugs ZNCZ02LM. I noticed today that the buttons were not responding, and checked the logs to see what was happening.

I have tried the reset button on the stick, restart/recreating the container, rebooting the machine and I am unable to get it to start now.

zigbee2mqtt 1.5.1
CC2531 Firmware: 20190608

configuration.yml
```yml
permit_join: false

mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://mqtt'

serial:
  port: /dev/ttyACM0

advanced:
  log_level: debug
```

log.txt
```txt
7/21/2019, 1:16:35 PM - info: Logging to directory: '/app/data/log/2019-07-21.13-16-35'
7/21/2019, 1:16:35 PM - debug: Loaded state from file /app/data/state.json
7/21/2019, 1:16:35 PM - debug: Saving state to file /app/data/state.json
7/21/2019, 1:16:35 PM - info: Starting zigbee2mqtt version 1.5.1 (commit #unknown)
7/21/2019, 1:16:35 PM - info: Starting zigbee-shepherd
7/21/2019, 1:16:35 PM - debug: Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/app/data/database.db","coordBackupPath":"/app/data/coordinator_backup.json","sp":{"baudRate":115200,"rtscts":true}}'
7/21/2019, 1:16:42 PM - info: Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: request timeout)
7/21/2019, 1:17:43 PM - info: Starting zigbee-shepherd
7/21/2019, 1:17:50 PM - error: Error while starting zigbee-shepherd! (Error: request timeout)
```

Koenkk, 2019-07-21:
How many devices do you have?

shane151, 2019-07-21:
5 switches, 2 plugs

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Koenkk, 2019-07-22:
Can you post the output when running with `DEBUG=* npm start` and provide the output of `ls /dev` (on pastebin.com and link it here)

shane151, 2019-07-22:
Forgot to mention this is running on a 2009 MacBook Pro.

Log output, <https://pastebin.com/wanJd6VH>
ls /dev (on host machine, /dev/ttyACM0 exists, and is mounted into the container)<https://pastebin.com/Gzjb2uS5>

docker-compose config for reference <https://pastebin.com/9H2fKR5n>

Koenkk, 2019-07-22:
Can you try reflashing the stick? (to make sure that's not the problem)

shane151, 2019-07-23:
I purchased my stick pre-flashed. The strange this is it was working fine for about 4 days, and I made no changes.

I have a downloader cable in the mail, it should arrive in 1-2 weeks and I'll be able to reflash it. I will keep you updated.

Thanks for your help so far Koen.

jameshclrk, 2019-07-28:
I have the same issue when trying to use networks in my docker-compose.yml.
It works with `network_mode: host`. Is this required?

shane151, 2019-07-31:
I have reflashed the stick and it is working again now.

closed 2019-08-01 by shane151