Koenkk/zigbee2mqtt #1760: Errors after update

stanislavmerts, 2019-07-21:
Hello! I use Hass.io addon. Today i update addon from 1.4.0 to 1.5.1 and got a lot of problems.
Firstly, when i try to start addon I saw in the message log as in issue # 1756. I addon in config line with "network_key": [1, 3, 5, 7, 9, 11, 13, 15, 0, 2, 4, 6, 8, 10, 12, 13]. And it succesfully started.
Then I encountered the following problem. I see what my Aqara sensors and Lamp is connected, but I don't get information from them. I tried reconnect my sensors and lamp but no result.
As for the lamp, I see some additional errors:

7/21/2019, 9:35:38 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.","meta":{"entity":{"ID":"0x00158d0002d39c50","type":"device","friendlyName":"0x00158d0002d39c50"},"message":"{\"state\": \"ON\"}"}}'

 zigbee2mqtt:error 7/21/2019, 10:23:31 PM Failed to ping 0x00158d0002d39c50

I tried downgrade to version 1.4.0 from my backup but the result is exactly the same.
Please help me.

Koenkk, 2019-07-21:
Did you change your network key? Because that requires repairing all devices

stanislavmerts, 2019-07-21:
What i need to do? I reapairing all my sensors and lamp (it's Xiaomi Aqara). I click button on sensor for 5 second, then blue LED flashing tripple. For a lamp I turn power off and on 5 times and the bulb blink twice. In log i see what finding new device. 
Now my config in version 1.4.0 is:
{
  "data_path": "/share/zigbee2mqtt",
  "homeassistant": true,
  "permit_join": false,
  "mqtt_base_topic": "zigbee2mqtt",
  "mqtt_server": "mqtt://192.168.0.4:1883",
  "serial_port": "/dev/ttyACM0",
  "devices": [],
  "network_key": [],
  "mqtt_user": "user",
  "mqtt_pass": "password"
}
Now the lamp in front of me on the table, disconnected from the power supply. But in hass.io my lamp is online.

anrada73, 2019-07-21:
I have the same problems

stanislavmerts, 2019-07-22:
Ok, I solved my problem.
I change config on version 1.4.0
Before: "network_key": [],
After: "network_key": [1, 3, 5, 7, 9, 11, 13, 15, 0, 2, 4, 6, 8, 10, 12, 13]
Next I repair all the sensors. After this i remove aqara lamp. When i try pairing lamp, in log i see errors:
In the process, I got a lot of errors:
zigbee2mqtt:warn 7/22/2019, 1:51:56 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 7/22/2019, 1:51:56 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:error 7/22/2019, 1:52:16 PM Cannot get the Node Descriptor of the Device: 0x00158d0002d39c50 (Error: Timed out after 10000 ms)
First and second pairing unsuccessful, but in third time pairing is successful.

flatsiedatsie, 2019-08-01:
Perhaps related, I'm also getting `zigbee2mqtt:error 2019-8-1 08:27:26 Cannot get the Node Descriptor of the Device: 0x00158dxxxxxxx (Error: Timed out after 10000 ms)` on an Aqara cube. At the same time an Aqara magnetic contact sensor (V2) works just time.

closed 2019-10-07 by stale[bot]