Koenkk/zigbee2mqtt #1768: Retain group status

snoopyCode, 2019-07-23:
zigbee2mqtt version 1.2.1 (commit #e5ca977)
 Coordinator firmware version: '20190223'

I want to retain all states in my system in mqtt so that other services get the latest state when they connect. This accounts also to group states. I tried naively to configure the group as device but that makes zigbee2mqtt to through an error that 'Lights' could not be found anymore.

```
groups:
  '1':
    friendly_name: Lights
devices:
  'Lights':
    friendly_name: 'Lights'
    retain: true
```

snoopyCode, 2019-07-23:
Ok. I find it out.

```
groups:
  '1':
      friendly_name: Lights
      retain: true
```

Nice feature. You should document it.

closed 2019-07-23 by snoopyCode

Koenkk, 2019-07-24:
Done, thanks!