Koenkk/zigbee2mqtt #1769: unable to pairing anything with 1.5.1 update

Rihan9, 2019-07-23:
# Bug Report
## What happened
after update to 1.5.1 the zigbee2mqtt temporarily broke because the network_key was missing on config file.

I fix it, and all my devices have stopped working. 
I try to repair them, without success.
I try to rollback zigbee2mqtt's update, I update the cc2531 firmware with 2 different version, shut down all device and try to repair only one. Nothing changes.

## Debug Info
zigbee2mqtt version: 
1.5.1

CC253X firmware version: 
https://github.com/Koenkk/Z-Stack-firmware/blob/dev/coordinator/Z-Stack_3.0.x/bin/CC2531_20190425.zip

https://github.com/Koenkk/Z-Stack-firmware/blob/dev/coordinator/Z-Stack_Home_1.2/bin/default/CC2531_DEFAULT_20190608.zip

config:
{
  "data_path": "/share/zigbee2mqtt",
  "devices": "[]",
  "groups": "groups.yaml",
  "homeassistant": true,
  "permit_join": true,
  "mqtt": {
    "base_topic": "zigbee2mqtt",
    "server": "mqtt://172.30.32.1:1883",
    "user": "zigbee2mqtt",
    "password": "zigbee2mqtt"
  },
  "serial": {
    "port": "/dev/ttyACM0"
  },
  "advanced": {
    "channel": 15,
    "log_level": "debug",
    "network_key": [ 1,3, 5,7,9,11,13,15,0,2, 4,6, 8,10,12,13]
  }
}

log:
> node index.js
  zigbee2mqtt:info 7/23/2019, 9:14:38 PM Logging to directory: '/share/zigbee2mqtt/log/2019-07-23.21-14-37'
  zigbee2mqtt:debug 7/23/2019, 9:14:38 PM Removing old log directory '/share/zigbee2mqtt/log/2019-07-22.23-32-09'
  zigbee2mqtt:debug 7/23/2019, 9:14:38 PM Loaded state from file /share/zigbee2mqtt/state.json
  zigbee2mqtt:debug 7/23/2019, 9:14:38 PM Saving state to file /share/zigbee2mqtt/state.json
  zigbee2mqtt:info 7/23/2019, 9:14:38 PM Starting zigbee2mqtt version 1.5.1 (commit #unknown)
  zigbee2mqtt:info 7/23/2019, 9:14:38 PM Starting zigbee-shepherd
  zigbee2mqtt:debug 7/23/2019, 9:14:38 PM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[15],"precfgkey":"HIDDEN"},"dbPath":"/share/zigbee2mqtt/database.db","coordBackupPath":"/share/zigbee2mqtt/coordinator_backup.json","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:info 7/23/2019, 9:18:53 PM zigbee-shepherd started
  zigbee2mqtt:info 7/23/2019, 9:18:53 PM Coordinator firmware version: '20190425'
  zigbee2mqtt:debug 7/23/2019, 9:18:53 PM zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":"15","panId":65535,"extPanId":[0,0,18,25,54,75,189,239],"ieeeAddr":"0x00124b001936bdef","nwkAddr":0},"firmware":{"transportrev":2,"product":2,"version":"2.7.2","revision":20190425},"startTime":1563909533,"joinTimeLeft":0}
  zigbee2mqtt:info 7/23/2019, 9:18:53 PM Currently 0 devices are joined:
  zigbee2mqtt:warn 7/23/2019, 9:18:53 PM `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 7/23/2019, 9:18:53 PM Allowing new devices to join.
  zigbee2mqtt:warn 7/23/2019, 9:18:53 PM Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 7/23/2019, 9:18:53 PM Zigbee: allowing new devices to join.
  zigbee2mqtt:info 7/23/2019, 9:18:53 PM Connecting to MQTT server at mqtt://172.30.32.1:1883
  zigbee2mqtt:info 7/23/2019, 9:18:53 PM zigbee-shepherd ready
  zigbee2mqtt:debug 7/23/2019, 9:18:53 PM Mounted the cieApp (epId 11)
  zigbee2mqtt:info 7/23/2019, 9:18:54 PM Connected to MQTT server
  zigbee2mqtt:info 7/23/2019, 9:18:54 PM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 7/23/2019, 9:18:54 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.5.1","commit":"unknown","coordinator":20190425,"log_level":"debug","permit_join":true}'
  zigbee2mqtt:debug 7/23/2019, 9:19:33 PM Received MQTT message on 'zigbee2mqtt/bridge/config/permit_join' with data 'true'
  zigbee2mqtt:info 7/23/2019, 9:19:33 PM Zigbee: allowing new devices to join.
  zigbee2mqtt:info 7/23/2019, 9:19:33 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.5.1","commit":"unknown","coordinator":20190425,"log_level":"debug","permit_join":true}'
  zigbee2mqtt:debug 7/23/2019, 9:19:34 PM Received MQTT message on 'zigbee2mqtt/bridge/config/permit_join' with data 'false'
  zigbee2mqtt:info 7/23/2019, 9:19:34 PM Zigbee: disabling joining new devices.
  zigbee2mqtt:info 7/23/2019, 9:19:35 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.5.1","commit":"unknown","coordinator":20190425,"log_level":"debug","permit_join":false}'
  zigbee2mqtt:debug 7/23/2019, 9:19:38 PM Saving state to file /share/zigbee2mqtt/state.json
  zigbee2mqtt:debug 7/23/2019, 9:19:51 PM Received MQTT message on 'zigbee2mqtt/bridge/config/permit_join' with data 'true'
  zigbee2mqtt:info 7/23/2019, 9:19:51 PM Zigbee: allowing new devices to join.
  zigbee2mqtt:info 7/23/2019, 9:19:51 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.5.1","commit":"unknown","coordinator":20190425,"log_level":"debug","permit_join":true}'
  zigbee2mqtt:info 7/23/2019, 9:22:31 PM Successfully reenabled joining

JLFN, 2019-07-23:
looks like same issue https://github.com/danielwelch/hassio-zigbee2mqtt/issues/200

Rihan9, 2019-07-23:
But i don't have any error message on log. I'm using hassio docker image on raspberry pi 4,  with danielwelch add-on.

jerrychong25, 2019-07-24:
Faced similar issue like @Rihan9, but solved if after some many round of trials and errors.

**Error Screenshot:**
[image]

**Error Debug Log:**
```
"0x00158d0002a23402" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{0x00158d0002a23402|0x8f31 |Xiaomi Aqara vibration sensor (DJT11LM)|online (unknown)}"];
  "0x00158d0002a23402" -> "0x00124b001b7caacd" [penwidth=1, weight=0, color="#994444", label="106"]
  "0x00158d00023496f7" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{0x00158d00023496f7|0x1aef |Xiaomi Aqara water leak sensor (SJCGQ11LM)|online (unknown)}"];
  "0x00158d00023496f7" -> "0x00124b001b7caacd" [penwidth=1, weight=0, color="#994444", label="95"]
  "0x00158d000245aeb2" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{0x00158d000245aeb2|0x773d |Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|online (unknown)}"];
  "0x00158d000245aeb2" -> "0x00124b001b7caacd" [penwidth=1, weight=0, color="#994444", label="100"]
  "0x00158d000237a808" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{0x00158d000237a808|0xeac2 |Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|online (unknown)}"];
  "0x00158d000237a808" -> "0x00124b001b7caacd" [penwidth=1, weight=0, color="#994444", label="170"]
  "0x00158d000309ede7" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{0x00158d000309ede7|0x9daa |Xiaomi Aqara door & window contact sensor (MCCGQ11LM)|online (unknown)}"];
  "0x00158d000309ede7" -> "0x00124b001b7caacd" [penwidth=1, weight=0, color="#994444", label="170"]
}'
  zigbee2mqtt:warn 7/24/2019, 9:39:10 AM Failed network lqi scan for device: '0x00124b001bc74813' with error: 'Error: Timed out after 10000 ms'
  zigbee2mqtt:warn 7/24/2019, 9:39:10 AM Failed network lqi scan for device: '0x00124b001bc74838' with error: 'Error: Timed out after 10000 ms'
  zigbee2mqtt:info 7/24/2019, 9:40:00 AM Starting network scan includeRoutes='0'...
  zigbee2mqtt:info 7/24/2019, 9:40:01 AM MQTT publish: topic 'zigbee2mqtt/0x00158d000239c9c3', payload '{"battery":100,"voltage":3015,"linkquality":109,"contact":true,"device":{"ieeeAddr":"0x00158d000239c9c3","friendlyName":"0x00158d000239c9c3","type":"EndDevice","nwkAddr":47510,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_magnet","hwVersion":"unknown","swBuildId":"unknown","dateCode":"unknown","status":"online"}}'
  zigbee2mqtt:warn 7/24/2019, 9:40:03 AM Network scan timeout, skipping outstanding lqi scans for '0x00124b001bc74813 0x00124b001bc74838'
  zigbee2mqtt:warn 7/24/2019, 9:40:03 AM Network scan timeout, skipping outstanding rtg scans for ''
  zigbee2mqtt:info 7/24/2019, 9:40:03 AM MQTT publish: topic 'zigbee2mqtt/bridge/networkmap/graphviz', payload 'digraph G {
node[shape=record];
  "0x00124b001b7caacd" [style="bold, filled", fillcolor="#e04e5d", fontcolor="#ffffff", label="{0x00124b001b7caacd|0x0 |No model information available|online (2019-07-24T09:40:03+08:00)}"];
  "0x00158d000239c9c3" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{0x00158d000239c9c3|0xb996 |Xiaomi MiJia door & window contact sensor (MCCGQ01LM)|online (00:00:01s ago)}"];
  "0x00158d000239c9c3" -> "0x00124b001b7caacd" [penwidth=1, weight=0, color="#994444", label="126"]
  "0x00158d00023dc1fb" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{0x00158d00023dc1fb|0xeb0 |Xiaomi Aqara wireless switch (WXKG11LM)|online (unknown)}"];
  "0x00158d00023dc1fb" -> "0x00124b001b7caacd" [penwidth=1, weight=0, color="#994444", label="153"]
  "0x00124b001bc74813" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{0x00124b001bc74813|0xfac4  no lqi|Custom devices (DiY) [CC2530 router](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/) (CC2530.ROUTER)|offline (unknown)}"];
  "0x00124b001bc74813" -> "0x00124b001b7caacd" [penwidth=0.5, weight=0, color="#994444", label="0"]
  "0x00124b001bc74838" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{0x00124b001bc74838|0xe70e  no lqi|Custom devices (DiY) [CC2530 router](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/) (CC2530.ROUTER)|offline (unknown)}"];
  "0x00158d0002a23402" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{0x00158d0002a23402|0x8f31 |Xiaomi Aqara vibration sensor (DJT11LM)|online (unknown)}"];
  "0x00158d0002a23402" -> "0x00124b001b7caacd" [penwidth=1, weight=0, color="#994444", label="106"]
  "0x00158d00023496f7" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{0x00158d00023496f7|0x1aef |Xiaomi Aqara water leak sensor (SJCGQ11LM)|online (unknown)}"];
  "0x00158d00023496f7" -> "0x00124b001b7caacd" [penwidth=1, weight=0, color="#994444", label="95"]
  "0x00158d000245aeb2" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{0x00158d000245aeb2|0x773d |Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)|online (unknown)}"];
  "0x00158d000245aeb2" -> "0x00124b001b7caacd" [penwidth=1, weight=0, color="#994444", label="100"]
  "0x00158d000237a808" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{0x00158d000237a808|0xeac2 |Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|online (unknown)}"];
[... cut ...]
2019-07-24T09:41:00: PM2 log: App [npm:0] starting in -fork mode-
2019-07-24T09:41:00: PM2 log: App [npm:0] online
> zigbee2mqtt@1.5.1 start /zigbee2mqtt-1.5.1
> node index.js
```

**My way to solve this error:**
1. Double check Zigbee2mqtt config settings after migration to new code from **configuration.yaml.bk** to **configuration.yaml**.
2. Still not working after restart Zigbee2mqtt add on several times, even though all settings are same.
3. Restart whole system.
4. It works again after restart.

Hope this helps!

Rihan9, 2019-07-24:
Hi,
follow my configuration.yaml.bk:
devices:
  '0x00158d0001e7f618':
    friendly_name: '0x00158d0001e7f618'
    retain: false
  '0x00158d0002c44540':
    friendly_name: '0x00158d0002c44540'
    retain: false
  '0x7cb03eaa00af3647':
    friendly_name: '0x7cb03eaa00af3647'
    retain: false
  '0x7cb03eaa00af42ad':
    friendly_name: '0x7cb03eaa00af42ad'
    retain: false
  '0x7cb03eaa0a00eb1e':
    friendly_name: '0x7cb03eaa0a00eb1e'
    retain: false
homeassistant: true
mqtt:
  base_topic: zigbee2mqtt
  password: zigbee2mqtt
  server: mqtt://172.30.32.1:1883
  user: zigbee2mqtt
permit_join: false
serial:
  port: /dev/ttyACM0

I restart my whole system several times.

Rihan9, 2019-07-31:
Hi,
I bought a new coordinator and the issues seemed to be solved. Yesterday my lights stopped responding to commands, and after resetting them, no device is more able to join the network.

Today, restarting integration, in the logs I see that it does not go further:
zigbee2mqtt:info 7/31/2019, 2:47:41 PM Starting zigbee-shepherd

I'm tired of fighting with it.

Later i try to uninstall integration, reflash the firmware and retry.

closed 2019-10-07 by stale[bot]