Koenkk/zigbee2mqtt #1770: Support for Ikea LED1737 and LED1739

wizzor, 2019-07-23:
# Bug Report
Ikea is replacing the LED1537 adjustable color temperature lights with the LED1737 and LED1739. 
## What happened
The bulbs do not appear to pair with the controller at all, no information is visible in the log when they are powered on

## What did you expect to happen
Operation similar to LED1537

## How to reproduce it (minimal and precise)
Allow pairing in ZB2MQTT
Purchase a LED1737 bulb from Ikea
Power up the bulb close to the ZB network controller

## Debug Info
zigbee2mqtt version:
CC253X firmware version:

Koenkk, 2019-07-24:
Even though it's not supported you should already be able to pair it, are you sure you didn't hit the device limit and that the device is in range of the coordinator?

alucao, 2019-07-29:
I can confirm that LED1737 pairs and works correctly.
It was difficult to pair though. My suggestion is to get the led as close as possible to the sniffer and spamming attempts to factory reset the leds by turning it on and off six times until the brightness of the led finally changes, signalling a successful pairing.

closed 2019-10-05 by stale[bot]