Koenkk/zigbee2mqtt #1771: the coordinator doesn't respond to "Match Description Request"

msadeghz, 2019-07-24:
Hi @Koenkk , 
I have a new zigbee device. it sends Match Description Request to the coordinator during join procedure. but the coordinator doesn't respond to it.
I use the latest version of CC2531 firmware and zigbee2mqtt. you can see the issue in below image:
[image]
the normal condition should be like below:
[image]

msadeghz, 2019-07-24:
I have found this guide in TI forum:
"You have to add OTA cluster to inCluster on Simple Descriptor of your coordinator"
how can I do that?

Koenkk, 2019-07-24:
Moving this to zigbee-herdsman, will look into this once refactor is completed.