Koenkk/zigbee2mqtt #1773: No converter available for 'QBKG03LM' with cid 'genDeviceTempCfg', type 'devChange'

VivantSenior, 2019-07-24:
Hi,

I noticed that QBKG03LM (s) measure temperature.

`No converter available for 'QBKG03LM' with cid 'genDeviceTempCfg', type 'devChange' and data '{"cid":"genDeviceTempCfg","data":{"currentTemperature":29}}'`

Is it possible to expose it to the user?

Best,
Viv

Koenkk, 2019-07-24:
Done, will be in the dev branch within an hour.

closed 2019-07-24 by Koenkk

antst, 2019-09-03:
Actually, looks like not all QBKG03LM report temperature. Mine don't. And because of that, now I have in HA bunch of temperature sensors which are empty. And, as result, my logs are full of 
```
Sep 04 01:05:00 hass hass[32744]: 2019-09-04 01:05:00 WARNING (MainThread) [homeassistant.components.alexa.capabilities] Invalid temp value  for sensor.entrance_switch_temperature
Sep 04 01:05:00 hass hass[32744]: 2019-09-04 01:05:00 WARNING (MainThread) [homeassistant.components.alexa.capabilities] Invalid temp value  for sensor.entrance_switch_temperature
Sep 04 01:05:00 hass hass[32744]: 2019-09-04 01:05:00 WARNING (MainThread) [homeassistant.components.alexa.capabilities] Invalid temp value  for sensor.entrance_switch_temperature
Sep 04 01:08:42 hass hass[32744]: 2019-09-04 01:08:42 WARNING (MainThread) [homeassistant.components.alexa.capabilities] Invalid temp value  for sensor.hallway_switch2_temperature
Sep 04 01:08:44 hass hass[32744]: 2019-09-04 01:08:44 WARNING (MainThread) [homeassistant.components.alexa.capabilities] Invalid temp value  for sensor.hallway_switch2_temperature
Sep 04 01:08:50 hass hass[32744]: 2019-09-04 01:08:50 WARNING (MainThread) [homeassistant.components.alexa.capabilities] Invalid temp value  for sensor.hallway_switch2_temperature
Sep 04 01:08:50 hass hass[32744]: 2019-09-04 01:08:50 WARNING (MainThread) [homeassistant.components.alexa.capabilities] Invalid temp value  for sensor.hallway_switch2_temperature
Sep 04 01:09:14 hass hass[32744]: 2019-09-04 01:09:14 WARNING (MainThread) [homeassistant.components.alexa.capabilities] Invalid temp value  for sensor.hallway_switch2_temperature
```
Can we make it configurable?

miroslavpetrov, 2019-12-05:
I have the same issue as @antst. Maybe there is a way to enable this temperature sensor?