Koenkk/zigbee2mqtt #1774: Router pairing issue - "TypeError: Cannot read property"

szporwolik, 2019-07-24:
# Bug Report
Pairing a router Ikea Trådfri outlet is a reason that zigbee2mqtt cannot start.

## What happened
After pairing with the router, I cannot start zigbee2mqtt anymore.

## What did you expect to happen
zigbee2mqtt working started and working

## How to reproduce it (minimal and precise)
I have zigbee2mqtt runing inside docker at Synology NAS; containter was created using:
`sudo docker run -it -v /volume1/docker/zigbee2mqtt/:/app/data --device=/dev/ttyACM0 --privileged koenkk/zigbee2mqtt:latest`

After creating a clean container everything works fine - zigbee shepard is started and I can start pairing devices.
While pairing metioned Ikea outlet (It happened with CC2531 router as well) everything looks fine untill, but the following error comes:
`TypeError: Cannot read property 'size' of null`
Full log: https://pastebin.com/NVuQAgqe

Next I'm not able to start the containter anymore: https://pastebin.com/dr35YNgB

configuration.yaml https://pastebin.com/UUeEZiwg
database.db https://pastebin.com/kwhArLnb
state.json https://pastebin.com/p2icHWih

The only way to fix is to complete clean image/files and repair again (then situation repeats).

## Debug Info
zigbee2mqtt version: 1.5.1
CC2531 firmware version: CC2531_DEFAULT_20190608.zip

szporwolik, 2019-07-24:
Update: pairing with IKEA router generate similar issue https://pastebin.com/uY1nxNEH

szporwolik, 2019-07-24:
Repairing log (with clean database.db, otherwise it does not start) with DEBUG: https://pastebin.com/PGmjrQK0

Koenkk, 2019-07-26:
Your configuration.yaml defines `whitelist: null`, change it to `whitelist: []` or remote it completely.

szporwolik, 2019-07-26:
Hello , thank you @Koenkk - yes, I can confirmed, this was the configuration issue. Thank you for information.

What is happeing, if someone will put the lines:
'
whitelist:
blacklist:
`
during execution those lines are replaced with "whitelist: null"; with such change zigbee2mqtt is nor starting after restart.

It's not a big issue, but as in example HA is OK with putting empty nodes I would consider this to be a bug/feature request, so nobody will have similar troubles like myself. 

Anyhow, my case is closed - thank you! 👍

Koenkk, 2019-07-27:
Will be set to empty array when null now, thanks!

closed 2019-07-27 by Koenkk