Koenkk/zigbee2mqtt #1776: Xiaomi switch (WXKG02LM) bug or request.

GuGu927, 2019-07-25:
# Bug Report
## What happened

## What did you expect to happen
When version of zigbee2mqtt was 1.4.0,
when I press xiaomi switch(WXKG02LM), sensor show the state(left, right or both..) and turns to blank.(none)
But after upgrade zigbee2mqtt to 1.5.1, sensor state remains last state(left, right or both. not blank)

## How to reproduce it (minimal and precise)
I need state of sensor  "blank"
Because when I use automation(HomeAssitant), it not works.
Thank you

Here's code.
- id: kitchen_light_switch
  hide_entity: true
  trigger:
  - platform: state
    entity_id: sensor.switch_kitchen_click
    to: "left" # ..... "right" etc.
  condition: []
  action:
  - data_template:
      entity_id: >-
        {% if trigger.to_state.state == "left" %}switch.kitchen_light1
        {% elif trigger.to_state.state == "right" %}switch.kitchen_light3
        {% endif %}
    service: switch.toggle

## Debug Info
zigbee2mqtt version:
CC253X firmware version: 20190608

Koenkk, 2019-07-26:
Can you post your log and `configuration.yaml`?

GuGu927, 2019-07-26:
> Can you post your log and `configuration.yaml`?

Here is my log.(Verion 1.5.1)
zigbee2mqtt:info 7/26/2019, 10:39:09 PM MQTT publish: topic 'zigbee2mqtt/switch_kitchen', payload '{"linkquality":55,"battery":100,"voltage":3055,"click":"right"}'
zigbee2mqtt:info 7/26/2019, 10:39:11 PM MQTT publish: topic 'zigbee2mqtt/switch_kitchen', payload '{"linkquality":52,"battery":100,"voltage":3055,"click":"left"}'

This worked in previous versions..(Version 1.4.0)
zigbee2mqtt:info x/x/2019, 10:39:09 PM MQTT publish: topic 'zigbee2mqtt/switch_kitchen', payload '{"linkquality":55,"battery":100,"voltage":3055,"click":"right"}'
zigbee2mqtt:info x/x/2019, 10:39:09 PM MQTT publish: topic 'zigbee2mqtt/switch_kitchen', payload '{"linkquality":55,"battery":100,"voltage":3055,"click":""}'
zigbee2mqtt:info x/x/2019, 10:39:11 PM MQTT publish: topic 'zigbee2mqtt/switch_kitchen', payload '{"linkquality":52,"battery":100,"voltage":3055,"click":"left"}'
zigbee2mqtt:info x/x/2019, 10:39:11 PM MQTT publish: topic 'zigbee2mqtt/switch_kitchen', payload '{"linkquality":52,"battery":100,"voltage":3055,"click":""}'

Koenkk, 2019-07-27:
And your `configuration.yaml`? Are you sure that `homeassistant: true` is set?

GuGu927, 2019-07-29:
@Koenkk Thx. It solved.

closed 2019-07-29 by GuGu927