Koenkk/zigbee2mqtt #1779: Xiaomi DJT11LM Cannot get the Node Descriptor of the Device

colm89, 2019-07-25:
# Bug Report
## What happened
Trying to paid Xiaomi DJT11LM and keep getting the following:

`zigbee2mqtt:error 7/25/2019, 6:08:58 PM Cannot get the Node Descriptor of the Device: 0x00158d0003134356 (Error: Timed out after 10000 ms)`

I have tried both holding for 5 seconds and waiting, and holding for 5 seconds followed by clicking every 1 second until the above shows in the logs. I have tried in excess of 20 times and still get the same response.

## What did you expect to happen
Device to pair in the same way my xiaomi WXKG01LM and MCCGQ01LM have paired

## How to reproduce it (minimal and precise)
Install latest coordinator firmware and latest version of zigbee2mqtt in Hassio, and use both pairing methods (hold for 5 seconds, let go and wait; hold for 5 seconds, press every 1 second)

## Debug Info
zigbee2mqtt version: 1.5.1.1
CC253X firmware version: 20190608

PedroLamas, 2019-07-25:
I'm seeing this issue myself (same versions as @colm89 has). I have another DJT11LM that I had previously paired without any issues (using other versions of software & firmware) but now I can't add a new one...

> zigbee2mqtt:error 7/25/2019, 6:33:47 PM Cannot get the Node Descriptor of the Device: 0x00158d0003130f12 (Error: Timed out after 10000 ms)

GuGu927, 2019-07-25:
similar here.
When I pair xiaomi sensors, it shows undefined.

Koenkk, 2019-07-26:
I think you don't need to press the button every second after restting it, just hold it for +- 10 seconds and wait, before doing this be sure that restart zigbee2mqtt.

PedroLamas, 2019-07-26:
I've restarted zigbee2mqtt, pressed the button for +/-10 sec, saw that the led in the device blinked 3 times, signaling that it found the controller and was registering.

However, it seems to have failed and I can see in the zigbee2mqtt logs the same error message as above

colm89, 2019-07-26:
I have tried the same as PedroLamas and am still getting the same error that it cannot find the node descriptor of the device.

PedroLamas, 2019-07-26:
Funny thing is that the other 2 DJT11LM that I had previously paired are working flawlessly!

andyinleeds, 2019-07-27:
Received a brand new DJT11LM a few days ago and had same issue. 
Changed channel as a test (26 to avoid conflict with any wifi channels) and managed to pair within a few minutes.

PedroLamas, 2019-07-27:
> Changed channel as a test (26 to avoid conflict with any wifi channels) and managed to pair within a few minutes.

Would be fine to do that if it meant that I don't have to re-register all the devices again...

GauntPT, 2019-07-28:
I had the same issue with my Xiaomi door sensors. 

Try the following steps (that helped for me):
- delete the sensor from devices.yaml
- delete the sensor from database.db

Then you should be able to pair the sensor normally.

onkelfunny, 2019-07-28:
Try this:

Press the button for about 5 seconds. The LED lights up 3 times. Then press the button again every 2 seconds (maximum 20 times).

PedroLamas, 2019-07-28:
SUCCESS!! After a couple of tries, I managed to register the sensor using the instructions provided by @onkelfunny! 😄

Koenkk, 2019-07-29:
I've updated the pairing instructions under https://www.zigbee2mqtt.io/devices/DJT11LM.html with the ones from @onkelfunny. Assuming that the problem is solved now. 

Note that after the refactoring of zigbee-herdsman has been finished, we can probably achieve the same pairing experience as on the Xiaomi hub :smile:

closed 2019-07-29 by Koenkk