Koenkk/zigbee2mqtt #1781: Xiaomi Sensor Pairing Bug

GuGu927, 2019-07-26:
# Bug Report
## What happened
I'm using latest version of zigbee2mqtt.(Hass.io add-on)
There is a problem with pairing the device(Xiaomi WSDCGQ11LM).

Here's my config : https://pastebin.com/MGzV300N
Here's my db : https://pastebin.com/xUsLPsSM
Here's my log :  https://pastebin.com/63Dt01G3
  zigbee2mqtt:info 7/26/2019, 9:16:45 PM air_livingroom (0x00158d00023711f2): unkown - undefined unknown (EndDevice)
  zigbee2mqtt:info 7/26/2019, 9:16:45 PM air_room1 (0x00158d00022728f7): unkown - undefined unknown (EndDevice)

When sensor updates status : https://pastebin.com/UhLUdbsJ

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.5.1
CC253X firmware version:  CC2531_SOURCE_ROUTING_20190619

GuGu927, 2019-07-26:
I tested it in various ways.
I found the cause.

"zigbee_shepherd_devices"
This option creates the problem.

When    "zigbee_shepherd_devices": false
Sensor works great, pairing good.
When    "zigbee_shepherd_devices": true
Sensor not works, pairing issue.

Koenkk, 2019-07-26:
You should only use that option when you are using a custom zigbee-shepherd-converters (for e.g. adding support for unsupported devices). Assuming this can be closed now.

closed 2019-07-26 by Koenkk

GuGu927, 2019-07-26:
@Koenkk  I'm using custom zigbee-shepherd-converters.
I'm using unsupported devices... :(

GuGu927, 2019-07-26:
> You should only use that option when you are using a custom zigbee-shepherd-converters (for e.g. adding support for unsupported devices). Assuming this can be closed now.
@Koenkk I'm using custom zigbee-shepherd-converters.
I'm using unsupported devices... :(

GuGu927, 2019-07-26:
It was available  zigbee2mqtt  version 1.4.0