Koenkk/zigbee2mqtt #1784: Error while starting zigbee-shepherd - (Error: Write for 'BUFFER' not available)

jonwest, 2019-07-27:
# Bug Report
## What happened
Fresh install of Debian Stretch with Docker-CE, running the `latest-dev` Docker branch, Zigbee2MQTT fails to start with an error: `error: Error while starting zigbee-shepherd! (Error: Write for 'BUFFER' not available)` (full debug log [here](https://gist.github.com/jonwest/bb46d364bc69c66d9b6ff691f293863f)).  I've tried reflashing the firmware, no change.  If I use the `1.5.1` tag rather than `latest-dev` things boot successfully.

## What did you expect to happen
Zigbee2MQTT starts successfully.

## How to reproduce it (minimal and precise)
[Here's my docker-compose.yml](https://gist.github.com/jonwest/91ffad0d99040af890308b71d0875ba1).
- Run `docker-compose up`
- Container fails to start with above error.

## Debug Info
zigbee2mqtt version: 1.5.1 (latest-dev)
CC253X firmware version: [20190619](https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/bin/source_routing/CC2531_SOURCE_ROUTING_20190619.zip) (fresh flash)

Koenkk, 2019-07-27:
This is a mistake from my side in zigbee-herdsman, in the mean time some things have been changed.

Can you try it again with the latest dev docker image and provide me the debug log if it's still not working? (set environment variable `DEBUG=*`).

jonwest, 2019-07-27:
No worries!  I see in the `zigbee-herdsman` repo you'd released a new version since that `dev-latest` image was built.

I've reverted back to 1.5.1 for my "production" setup to keep the wife-approval-factor up, but I'll spin up another instance with a couple of other ZigBee devices to test out the new build.

Koenkk, 2019-07-27:
@jonwest that would be great!

jonwest, 2019-07-27:
Looks to still be occurring.  I specified `DEBUG: "*"` in my docker-compose, but the output logs from Z2M were standard logs.  It _did_ output debug logs to stderr, so I piped that to a logfile, but you'll need to ignore the `<0x02>` type color commands in there.

[Here's a gist of the logfile](https://gist.github.com/jonwest/0ab63cbcc0c0eb94ecc9d8d04c1bf7d2)

Koenkk, 2019-07-28:
The stdout log is missing, I expect that the correct log is shown with `docker logs -f zigbee2mqtt`.

jonwest, 2019-07-28:
Weird!  Not sure why the gist only outputted like 50 lines.  I've updated the gist, should be a few thousand lines there now.

Koenkk, 2019-07-29:
The log doesn't show the `Write fore 'BUFFER' not available` error. It does show the error you mentioned in https://github.com/Koenkk/zigbee2mqtt/issues/1789#issuecomment-515737470.

Are you still having the `Write for 'BUFFER' not available` exception? If not this can be closed.

jonwest, 2019-07-29:
Oh man... I'm sorry!  No idea why/how I pasted the same log from my other issue here.  Looks like with Github Gists, it'll stop pasting after the first `<0x00>` type character, so the log was ending after ~50 lines rather than the entire couple thousand line file.

I've uploaded the _proper_ log file to another pastebin-esque service.  [This should be the proper log file for debugging](https://0bin.net/paste/o+xqdcqkVyW1sJnz#QB2cypbruP0udSGPw1tycZqcXDDaL4QyGJ7TSGFu6T0).  Sorry about that!

Koenkk, 2019-07-29:
Should be fixed in latest dev branch, thanks for reporting (please close if the issue is fixed).

jonwest, 2019-07-30:
Booted up without a hitch, thank you!

closed 2019-07-30 by jonwest

Koenkk, 2019-07-30:
Great, also thanks for the donation, really appreciate it :smile: