Koenkk/zigbee2mqtt #1789: Zigbee network congestion at boot causing group and device dropouts

jonwest, 2019-07-28:
# Bug Report
## What happened
My network consists of around 50-60 devices (>70% are routers), and 11 groups.  When I boot Z2M, often times I'll run into 233 errors.  I _suspect_ it has to do with network congestion as there are a large number of publish messages in quick succession at boot, and the devices have varying levels of routing capabilities.  I do have a CC2530+CC2591 router, which generally helps with link quality, but I'm assuming that when there is a flood of messages like that, it's not going to route for every device, and so some messages can be dropped.

With all of my groups enabled, I get several 205 and 233 errors, however when I remove my groups (which creates another set of errors within zigbee-herdsman) and reboot, all of my devices are available and responsive.

Ideally I would like to be able to use groups as it will reduce the overall amount of messages floating around at any given time, and I generally want to control multiple devices at once.  

One of the solutions I've thought up but have yet to implement would be to either create the groups within software (likely NodeRed) and just issue messages sequentially with a time delay to create the illusion of an `ionice` type solution, though that feels messy and broken.

Another would be to boot Z2M without any groups and script the creation of them in a more "nice" fashion with some delays between message publishing.

I'm totally fine with a slower booting process if it results in a more reliable network overall.

Is there any way to configure a delay between message publishing at initial boot?  If there were, I wouldn't have to rebuild the groups every boot, and would likely have the same smooth boot process I have without groups.

## What did you expect to happen
Network boots without 205/233 errors.

## How to reproduce it (minimal and precise)
With a network of over 40 devices:
- Create 10+ groups with ~2-5 devices per group
- Reboot Z2M

## Debug Info
zigbee2mqtt version: 1.5.1
CC253X firmware version: Latest Z-Stack Home (20190619, I think?)

jonwest, 2019-07-28:
[Here's](https://gist.github.com/jonwest/e30dd8a4934a82c8706ba8fd5b443841) a log from my boot process(es) after removing the 11 groups from my configuration to get to a clean boot without 205/233 errors.  

First I tried removing the `groups:` block from within `configuration.yaml`, but ran into the first `Cannot read property '1' of null` error.  Then I put the groups block back in, booted, then used the `zigbee2mqtt/bridge/config/remove_group` topic to remove each of the groups, thinking that it would be a "cleaner" removal, but the issue persisted when I rebooted again.

If I _kept_ rebooting, eventually it would go through each of the groups and I would be able to boot (as shown in the logs).

Koenkk, 2019-07-29:
Could you try playing with the `queue` settings? https://www.zigbee2mqtt.io/configuration/configuration.html

e.g. changing to

```yaml
queue:
  # Delay in ms between each command
  delay: 250
  # Max no. of simultaneously running commands
  simultaneously: 2
```

jonwest, 2019-07-29:
I'd tried that initially with little success, but a combination of getting rid of groups, and the `queue` settings seems to have smoothed my "production" environment out a bit--cheers!

closed 2019-07-29 by jonwest