Koenkk/zigbee2mqtt #1790: Time out zigbee-sheoerd after Update

drfschmidt, 2019-07-28:
After updating to 1.5.1 zigbee2mqtt is not more starting. Before it works fine.
The Addon is running on raspberry pi 3 b+ with CC2531 ZigBee Stick.
I added the network key and other settings . The addon ist runnung and the green circle is visible.
But I got a time out of zigbee-shepherd starting. The bridge status is offline.
How can I fix this? 
Thanks for suooprt.

***********************************
  zigbee2mqtt:info 7/28/2019, 11:27:01 AM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: request timeout)
  zigbee2mqtt:info 7/28/2019, 11:28:01 AM Starting zigbee-shepherd
  zigbee2mqtt:error 7/28/2019, 11:28:08 AM Error while starting zigbee-shepherd! (Error: request timeout)
  zigbee2mqtt:error 7/28/2019, 11:28:08 AM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 7/28/2019, 11:28:08 AM Failed to start
	{"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/zigbee2mqtt-1.5.1/node_modules/zigbee-herdsman/dist/cc-znp/index.js:208:22)\n    at Object.onceWrapper (events.js:286:20)\n    at CcZnp.emit (events.js:198:13)\n    at Timeout.<anonymous> (/zigbee2mqtt-1.5.1/node_modules/zigbee-herdsman/dist/cc-znp/index.js:192:18)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}
  zigbee2mqtt:error 7/28/2019, 11:28:08 AM Exiting...
npm
 ERR! code
 ELIFECYCLE
npm 
ERR! errno 1
npm
 ERR! zigbee2mqtt@1.5.1 start: `node index.js`
npm
 ERR! Exit status 1
npm ERR!
 
npm ERR!
 Failed at the zigbee2mqtt@1.5.1 start script.

Koenkk, 2019-07-28:
Could you post the debug logging? (can be achieved by setting environment variable `DEBUG` to `*`).

anrada73, 2019-07-28:
same problem.

drfschmidt, 2019-07-29:
I hope, this will help.
./run.sh: line 15: [Info] Configuration backup found in /share/zigbee2mqtt/.configuration.yaml.bk. Skipping config backup.: No such file or directory
2019-07-29T08:43:38: PM2 log: Launching in no daemon mode
2019-07-29T08:43:38: PM2 log: App [npm:0] starting in -fork mode-
2019-07-29T08:43:39: PM2 log: App [npm:0] online
> zigbee2mqtt@1.5.1 start /zigbee2mqtt-1.5.1
> node index.js
  zigbee2mqtt:info 7/29/2019, 8:43:48 AM Logging to directory: '/share/zigbee2mqtt/log/2019-07-29.08-43-44'
  zigbee2mqtt:debug 7/29/2019, 8:43:48 AM Removing old log directory '/share/zigbee2mqtt/log/2019-07-29.08-27-24'
  zigbee2mqtt:debug 7/29/2019, 8:43:48 AM Loaded state from file /share/zigbee2mqtt/state.json
  zigbee2mqtt:debug 7/29/2019, 8:43:48 AM Saving state to file /share/zigbee2mqtt/state.json
  zigbee2mqtt:info 7/29/2019, 8:43:48 AM Starting zigbee2mqtt version 1.5.1 (commit #unknown)
  zigbee2mqtt:info 7/29/2019, 8:43:48 AM Starting zigbee-shepherd
  zigbee2mqtt:debug 7/29/2019, 8:43:48 AM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/share/zigbee2mqtt/database.db","coordBackupPath":"/share/zigbee2mqtt/coordinator_backup.json","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:info 7/29/2019, 8:43:55 AM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: request timeout)
  zigbee2mqtt:info 7/29/2019, 8:44:55 AM Starting zigbee-shepherd
  zigbee2mqtt:error 7/29/2019, 8:45:02 AM Error while starting zigbee-shepherd! (Error: request timeout)
  zigbee2mqtt:error 7/29/2019, 8:45:02 AM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 7/29/2019, 8:45:02 AM Failed to start
	{"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/zigbee2mqtt-1.5.1/node_modules/zigbee-herdsman/dist/cc-znp/index.js:208:22)\n    at Object.onceWrapper (events.js:286:20)\n    at CcZnp.emit (events.js:198:13)\n    at Timeout.<anonymous> (/zigbee2mqtt-1.5.1/node_modules/zigbee-herdsman/dist/cc-znp/index.js:192:18)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}
  zigbee2mqtt:error 7/29/2019, 8:45:02 AM Exiting...
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@1.5.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.5.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-07-29T06_45_02_818Z-debug.log
2019-07-29T08:45:04: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2019-07-29T08:45:04: PM2 log: App [npm:0] starting in -fork mode-
2019-07-29T08:45:04: PM2 log: App [npm:0] online

drfschmidt, 2019-07-29:
I  set  "log_level": "debug" in config.
I don't know how to set debug variable.

drfschmidt, 2019-07-29:
I have Mosquitto broker running.

Koenkk, 2019-07-29:
- Did you try reflashing the stick?
- Are you sure that the port in the configuration is correct?

https://www.zigbee2mqtt.io/information/zigbee2mqtt_fails_to_start.html

drfschmidt, 2019-07-30:
Great!!! I solved the issue,
The port definition was not correct.

Thank you for support.

closed 2019-07-30 by drfschmidt