Koenkk/zigbee2mqtt #1791: Supporting effects: colorloop from Home Assistant Hue

Morphy99, 2019-07-28:
# Bug Report
## What happened
I want to use the effects option which is normally available in HA for Hue bulbs. I've tried it with mine and I get this in the zigbee2mqtt log:
No converter available for 'effect' (colorloop)

Is this not possible to implement or is there some script I can use?
## What did you expect to happen
Bulb colour cycle
## How to reproduce it (minimal and precise)
My automation code:
`light_colour_cycle:
  alias: Hue Light Colour Cycle
  sequence:
    - service: light.turn_on
      data:
        entity_id: light.fireplace_light
        effect: colorloop
        brightness: 255
        transition: 5`

## Debug Info
zigbee2mqtt version: 1.5.1 
CC253X firmware version: 20190223

trekker25, 2019-08-07:
How does this work in the Hue system? Is the color loop inside the lamp firmware? 

Or is the hue bridge sending out lots of Zigbee messages?
I made a color effect directly inside HASS but is flooding the Zigbee network after a while. 
(Making a disco with 8 color gu10 spots in a group. Kids like it. )

kvvoff, 2019-08-08:
I asked Gledopto about the effects, this is what they answered:
[image]

Koenkk, 2019-10-07:
remove stale

LukeHandle, 2019-12-06:
_bump_

burningTyger, 2020-03-21:
I'd very much like to have the loop too. I'm using a Müller tint color strip and using it with the vendor remote control it offers a color loop.

Morphy99, 2020-05-03:
Shame I'm looking at appdaemon for this but I think it's still going to flood mqtt

Koenkk, 2020-10-19:
I've checked how this works with the Hue hub (https://labs.meethue.com/formulas/huelabs/colorloop) but it's doing nothing more then sending the `hue_move` command which is already supported (https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttfriendly_nameset)

Mr-Groch, 2020-10-19:
Yes, please add support for "effect" attrubute with "colorloop" value, as this is one of the standard attributes supported in light.ligt_on home assistant service: https://www.home-assistant.io/integrations/light/. Not only philips hue, but also OSRAM (LEDVANCE) bulbs supports that effect - https://community.home-assistant.io/t/supported-light-effects-for-ledvance-bulbs/195866 (maybe also some other bulbs will support this).

Aerallo, 2020-10-22:
Is there a way to use the hue_move command from within a home assistant script? That would be even better.

Morphy99, 2020-10-22:
> Is there a way to use the hue_move command from within a home assistant script? That would be even better.

Damn phone stopped responding when I opened the page and somehow hit closed! 
I've implemented the hue_move via a button hold press from my IKEA remote, didn't know it existed tbh. It's just a input_boolean switch with mqtt_publish command to turn on. I was looking to control the speed as well but that would involve some kind of data template with the mqtt command which is above my skill level (or lack of!)

gotschi, 2020-10-22:
here's what I use in zha (native HA zigbee) to set the colorloop speed:

`- id: '1553302299487'
  initial_state: true
  alias: Bad Bewegung
  trigger:
  - entity_id: binary_sensor.bad_motion_sensor
    from: 'off'
    platform: state
    to: 'on'
  condition: []
  action:
  - data:
      ieee: '00:17:88:01:00:1c:ef:1e'
      endpoint_id: 11
      cluster_id: 8
      command: 4
      cluster_type: in
      command_type: server
      args:
        - 51
        - 50.0
    service: zha.issue_zigbee_cluster_command
  - data:
      ieee: '00:17:88:01:00:1c:ef:1e'
      endpoint_id: 11
      cluster_id: 6
      command: 1
      cluster_type: in
      command_type: server
    service: zha.issue_zigbee_cluster_command
  - data:
      ieee: '00:17:88:01:00:1c:ef:1e'
      endpoint_id: 11
      cluster_id: 768
      command: 68
      cluster_type: in
      command_type: server
      args:
        - 7
        - 2
        - 1
        - 50.0
        - 0
    service: zha.issue_zigbee_cluster_command`

maybe this is helping in some way

closed 2020-11-29 by github-actions[bot]

townsmcp, 2021-01-23:
> here's what I use in zha (native HA zigbee) to set the **colorloop speed** for a hue bloom,
> maybe this is helping in some way
> 
> ```
>   initial_state: true
>   alias: Bad Bewegung
>   trigger:
>   - entity_id: binary_sensor.bad_motion_sensor
>     from: 'off'
>     platform: state
>     to: 'on'
>   condition: []
>   action:
>   - data:
>       ieee: '00:17:88:01:00:1c:ef:1e'
>       endpoint_id: 11
>       cluster_id: 8
>       command: 4
>       cluster_type: in
>       command_type: server
>       args:
>         - 51
>         - 50.0
>     service: zha.issue_zigbee_cluster_command
>   - data:
>       ieee: '00:17:88:01:00:1c:ef:1e'
>       endpoint_id: 11
>       cluster_id: 6
>       command: 1
>       cluster_type: in
>       command_type: server
>     service: zha.issue_zigbee_cluster_command
>   - data:
>       ieee: '00:17:88:01:00:1c:ef:1e'
>       endpoint_id: 11
>       cluster_id: 768
>       command: 68
>       cluster_type: in
>       command_type: server
>       args:
>         - 7
>         - 2
>         - 1
>         - 50.0
>         - 0
>     service: zha.issue_zigbee_cluster_command```
> ```

@gotschi where do you find out how to do the arguments and what each value means? I can’t seem to find out anything other than number of supported commands but not what each command does

fitim7, 2021-03-14:
Are there news about that? Don't understand, why there are those unnecessary effects like "blink", "breath" or "okay" .. but not a colorloop? Yes, you can start them with "hue_move", but this is just unnecessary complicated. We need a simple implemented colorloop for color devices... deconz has implemented this so long ago...

Krystm, 2021-11-04:
I just moved over and agree this would be great, or random even to start a light with random color unless I am missing that. I saw the effect as others said in both ZHA and deconz but neither could run my network properly. So here I am.

dm82m, 2021-11-14:
Moved from hue access point to ZHA, used “colorloop” effect before and now. Basically it’s working. But the loop was slower and smoother with hue access point compared to ZHA. Any idea here?

townsmcp, 2022-05-08:
Has anyone been able to move this further along? I like the idea of somehow using ‘transition’ to determine the loop speed

deviantintegral, 2022-08-06:
You can use the mqtt.publish service to trigger hue_move. I have the following in an automation for an as-slow-as-possible loop:

```
service: mqtt.publish
data:
  topic: zigbee2mqtt/Desk Backlight/set
  payload: "{\"state\": \"on\", \"hue_move\": 1, \"saturation_move\": 1}"
```

t112013, 2023-01-10:
Does this work with ikea bulbs ?

bplatypus, 2023-10-13:
Hi,
Does it work? Because my standard Hue bulbs are doing nothing when I select 'colorloop' in HA or in z2m web interface (under 'Exposes' tab). 

For me, the following effects are doing something, others not:

- Blink (blink once in white)
- Breathe (works ok!)
- Okay (become green for 2 seconds)
- Channel_change (becomes orange for few seconds)

'Candle', 'Fireplace' and 'colorloop' are doing nothing :-(

kristianrl, 2023-10-16:
@bplatypus and @t112013, take a look at @deviantintegral's comment: https://github.com/Koenkk/zigbee2mqtt/issues/1791#issuecomment-1207299297 - I don't think it'll ever work with IKEA bulbs.

bplatypus, 2023-10-18:
@kristianrl I'm not using ikea bulb. Only Philips Hue bulbs.

Oggymator, 2024-03-10:
I can tell you, that the GLEDOPTO goods use the ZigBee Standard functions.
So yes, color loop is supported for most ZigBee Lights I think.
You can set the start color and also the speed of the color loop.

It would be awesome if we got any function like that into Z2M
@Koenkk

slugzero, 2024-06-09:
As of release 1.37.1, colorloop effect is available for all color lights.

archit3kt, 2025-04-27:
> As of release 1.37.1, colorloop effect is available for all color lights.

Is it still working in recent z2m versions ? I'm on latest 2.2.1 and since 2.0.0 colorloop just do nothing on my Hue white and color ambiance E27 1100lm, firmware 1.76.11

Also launching colorloop on a non Hue bulb only set it to a white / blue light

Oggymator, 2025-04-28:
> Is it still working in recent z2m versions ? I'm on latest 2.2.1 and since 2.0.0 colorloop just do nothing on my Hue white and color ambiance E27 1100lm, firmware 1.76.11
>
> Also launching colorloop [...]

Maybe same problem:

https://github.com/nurikk/zigbee2mqtt-frontend/issues/2354#issuecomment-2829224694