Koenkk/zigbee2mqtt #1792: OSRAM Smart+ plug (AB3257001NJ)    very short range

fred-c1, 2019-07-29:
This is not a zigbee2mqtt issue.
This OSRAM Smart+ plug (AB3257001NJ) issue + solution.

********************* Description of the problem:
I was having the following problem:
When I turn ON (or OFF) the plug, it immediately turns ON (or OFF)
but then sometimes 2 to 5 seconds later, the plug turn back OFF (or ON).

This problem disappears when the plug is very close to the coordinator.

I have observed this behavior with the 2 plugs I have.

I also saw a confirmation of OSRAM Smart+ plug range issue here:
https://fvdm.com/text/hacking/connecting-an-osram-smart-plug-to-philips-hue

More precisely:
"
On deciding the definitive location make sure the distance to a Hue light or the bridge is at most 4 meters. The radio antenna inside the plug is very small.
"

********************* Solution = antenna modification
The plug is relatively easy to open.
You can see some pictures here:
https://forum.smartapfel.de/forum/thread/1843-osram-smart-plug-zerlegt-zigbee-switch/

Yesterday I replaced the PCB antenna by a small wire antenna.
I used a wire of about 30mm
30mm is about 1/4 of wavelength at 2.45GHz

The link quality is now much better.
The plug switches properly every time.
I will try to add a picture of the antenna modification ASAP.

Koenkk, 2019-07-29:
Note that the issue tracker is meant for tracking bugs, for questions please use http://zigbee2mqtt.discourse.group/ or https://discord.gg/dadfWYE.

closed 2019-07-29 by Koenkk

fred-c1, 2019-07-30:
@Koenkk   this is not really a question but some information regarding a "supported device".
I think that people who are looking at buying devices would be interested by this info.

I don't know what is the best technical solution but I think that it would be great to have a central location for information related to supported devices:
pros/cons,  improvements/mods,   repair , etc.

Koenkk, 2019-07-30:
@fred-c1 each device has  a page where info can be put: https://www.zigbee2mqtt.io/devices/AB3257001NJ.html

fred-c1, 2019-07-31:
@Koenkk          I agree, this page should be the central location for device specific information.
but technically I don't know how this should be done.

Do you suggest I edit this document?
https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docgen/device_page_notes.js

I am concerned that if we do this this way it will be difficult to manage given the large amount of information subjects like      pros/cons, issues, improvements/mods, repair , etc.        can generate.

Koenkk, 2019-07-31:
Just edit this file: https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/AB3257001NJ.md

pa0los, 2019-10-10:
@fred-c1 Can you post the photos of your modification?
I would like to do the same to my Smart+ plugs.

fred-c1, 2019-10-10:
@ pa0los
Here is a small guide (even if you only need a small part of this):

> remove the external screw => torx head

> open the box by separating the white plastic from the grey plastic
There are 4 plastic latches. One at the center of each edge.
An old credit card can help.

> remove the 2 internal screws => torx head
You can now get the boards out of the box

> disconnect the PCB antenna by cutting the trace
Be sure to remove at least 1 to 2mm of the trace with this cut.
Looking at my picture the cut is just left of the solder (under the new wire antenna).

> solder the new wire antenna

> the red varnish (nail varnish should work) will make this mod more robust and will prevent any oxidation.

Here is the picture:

[image]

The antenna design of this product is really bad.
The above picture is actually showing a plug which is not so bad (but still very poor range).
I have some plugs for which this mod will be much more difficult because the location of the cut (and solder) is under the shield...   which make the range of the PCB antenna even worse.

gtozzi, 2021-04-18:
Thanks @fred-c1, it worked!

joopdo, 2021-12-17:
Thanks @fred-c1 it worked for me too. Note that at first my wire antenna (10 cm) had less reception, but after I cut it to 4 cm, it had better connection then before. I also added glue besides soldering.

aqwserf, 2023-11-29:
@fred-c1 I'd be very interested by this mod too but I'm not sure what you mean by:

> disconnect the PCB antenna by cutting the trace
> Be sure to remove at least 1 to 2mm of the trace with this cut.

can you provide details, or better some pictures? And what kind of wire antenna do you use? Can speaker cable do the job?

Thanks!

fred-c1, 2023-11-29:
@aqwserf :  If you can provide a picture before the mod, I should be able to provide more info.

fred-c1, 2023-11-29:
@aqwserf : I think that the interesting part is under the "1929" label. Pls remove the label and send a new picture. Also, in some units, the metal shield is very close to the antenna and this make the mod more difficult.

aqwserf, 2023-11-29:
@fred-c1 would it be the small connector on the left of the label?
[image]

aqwserf, 2023-11-29:
@fred-c1 Thanks! I guess the red is to "cut" the built-in antenna from the PCB and instead use other one? I'll probably gonna have to unsolder the cover.
And what about the antenna itself? Speaker wires would do the job? About 3cm?
Thanks!

fred-c1, 2023-11-29:
@aqwserf 
"I guess the red is to "cut" the built-in antenna from the PCB and instead use other one?"  => yes
About the wire:
- The section is not critical. Around AWG24 should be fine
- Length: yes around 30mm

aqwserf, 2023-12-04:
@fred-c1 Finally found some time ti make the mod, but I have a couple of additional questions:

[image]

In red seems to be incoming connection to the antenna and in orange, it seems to be a dead end, so:
1. should I remove the cooper starting in purple and above? 
2. should I also scratch the surface in yellow and solder the wire there? If so, what happens to the remaining antenna from yellow to the blue circle?
3. can't I just scratch and solder in the blue or yellow circle?

Thanks!

fred-c1, 2023-12-04:
@aqwserf 
Try to make sharper photos because the area pointed by your red arrow is not very good.
My guess is :
#1 =yellow area = remove the copper of the PCB track at least 1-2mm  (2 cuts + solder iron heat to detach the copper from the board)
#2 = purple area = solder the wire (/new antenna) here  (same as my photo above)

fred-c1, 2023-12-04:
@aqwserf
Good picture.
I confirm my previous answer.

aqwserf, 2023-12-04:
@fred-c1 Thanks! So based on your information: remove the copper in the yellow area and scratch the surface on the red one then solder the new wire right?

[image]