Koenkk/zigbee2mqtt #1794: Endless serial communication while startup of shepherd

florianluediger, 2019-07-29:
# Bug Report
## What happened

I started with a fresh raspbian install today but can not make zigbee2mqtt start up correctly. When starting via `npm start`, the following log output appears and after that nothing happens anymore:

```
pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.5.1 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 7/29/2019, 1:04:05 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-07-29.13-04-01'
  zigbee2mqtt:info 7/29/2019, 1:04:05 PM Starting zigbee2mqtt version 1.5.1 (commit #ac3b924)
  zigbee2mqtt:info 7/29/2019, 1:04:05 PM Starting zigbee-shepherd
```

When activating `log_level: debug` in the configuration.yaml file, only a little more output appears which does not help me much:

```
pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.5.1 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 7/29/2019, 1:08:12 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-07-29.13-08-09'
  zigbee2mqtt:debug 7/29/2019, 1:08:12 PM Removing old log directory '/opt/zigbee2mqtt/data/log/2019-07-28.22-35-23'
  zigbee2mqtt:debug 7/29/2019, 1:08:12 PM Loaded state from file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:debug 7/29/2019, 1:08:12 PM Saving state to file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:info 7/29/2019, 1:08:13 PM Starting zigbee2mqtt version 1.5.1 (commit #ac3b924)
  zigbee2mqtt:info 7/29/2019, 1:08:13 PM Starting zigbee-shepherd
  zigbee2mqtt:debug 7/29/2019, 1:08:13 PM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/opt/zigbee2mqtt/data/database.db","coordBackupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","sp":{"baudRate":115200,"rtscts":true}}'
```

As a last test, I tried to run with the `DEBUG=*` environment variable set and the console gets completely spammed with serial communication messages. I pressed Ctrl+C as soon as the console started to get spammed but the repeating messages will continue forever:

```
pi@raspberrypi:/opt/zigbee2mqtt $ DEBUG=* npm start

> zigbee2mqtt@1.5.1 start /opt/zigbee2mqtt
> node index.js

  serialport/bindings loading LinuxBinding +0ms
  zigbee2mqtt:info 7/29/2019, 1:10:44 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-07-29.13-10-40'
  zigbee2mqtt:debug 7/29/2019, 1:10:44 PM Removing old log directory '/opt/zigbee2mqtt/data/log/2019-07-29.12-26-23'
  zigbee2mqtt:debug 7/29/2019, 1:10:44 PM Loaded state from file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:debug 7/29/2019, 1:10:44 PM Saving state to file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:info 7/29/2019, 1:10:44 PM Starting zigbee2mqtt version 1.5.1 (commit #ac3b924)
  zigbee2mqtt:info 7/29/2019, 1:10:44 PM Starting zigbee-shepherd
  zigbee2mqtt:debug 7/29/2019, 1:10:44 PM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/opt/zigbee2mqtt/data/database.db","coordBackupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee-shepherd:init zigbee-shepherd booting... +0ms
  cc-znp Initialize, opening serial port +0ms
  serialport/stream opening path: /dev/ttyAMA0 +0ms
  serialport/binding-abstract open +0ms
  serialport/stream _read queueing _read for after open +28ms
  serialport/bindings/poller Creating poller +0ms
  serialport/stream opened path: /dev/ttyAMA0 +37ms
  serialport/stream _write 1 bytes of data +8ms
  serialport/binding-abstract write 1 bytes +67ms
  serialport/stream _read reading +8ms
  serialport/binding-abstract read +8ms
  serialport/bindings/unixWrite Starting write 1 bytes offset 0 bytesToWrite 1 +0ms
  serialport/bindings/unixRead Starting read +0ms
  serialport/bindings/unixWrite write returned null 0 +32ms
  serialport/bindings/unixWrite wrote 0 bytes +2ms
  serialport/bindings/unixWrite Starting write 1 bytes offset 0 bytesToWrite 1 +3ms
  serialport/binding-abstract read +53ms
  serialport/bindings/unixRead Starting read +40ms
  serialport/bindings/unixWrite write returned null 0 +18ms
  serialport/bindings/unixWrite wrote 0 bytes +2ms
  serialport/bindings/unixWrite Starting write 1 bytes offset 0 bytesToWrite 1 +3ms
[... 69 lines cut ...]
  zigbee2mqtt:error 7/29/2019, 1:10:45 PM Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: 'offline
  zigbee-shepherd zigbee-shepherd is stopping. +0ms
  zigbee-shepherd zigbee-shepherd is stopped. +7ms
  zigbee2mqtt:info 7/29/2019, 1:10:45 PM zigbee-shepherd stopped
```

## What did you expect to happen

I was hoping that the application would start normally.

## How to reproduce it (minimal and precise)

Not really sure about that.

## Debug Info
zigbee2mqtt version: Commit ac3b924065c6f2b9dd7433a6f193d1924b34bb7d

CC253X firmware version: CC2531_DEFAULT_20190608 which is the most recent version mentioned in the [flashing guide](https://www.zigbee2mqtt.io/getting_started/flashing_the_cc2531.html)

Mosquitto version: 1.5.7

/opt/zigbee2mqtt/data/configuration.yaml file:

```
```

Koenkk, 2019-07-29:
Can you try `port: /dev/ttyACM0` instead of `port: /dev/ttyAMA0`

florianluediger, 2019-07-29:
That does not work because `/dev/ttyACM0` does not exist on my machine. See the log output below:

```
pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.5.1 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 7/29/2019, 5:00:02 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-07-29.16-59-59'
  zigbee2mqtt:debug 7/29/2019, 5:00:02 PM Removing old log directory '/opt/zigbee2mqtt/data/log/2019-07-29.12-30-35'
  zigbee2mqtt:debug 7/29/2019, 5:00:02 PM Loaded state from file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:debug 7/29/2019, 5:00:02 PM Saving state to file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:info 7/29/2019, 5:00:03 PM Starting zigbee2mqtt version 1.5.1 (commit #ac3b924)
  zigbee2mqtt:info 7/29/2019, 5:00:03 PM Starting zigbee-shepherd
  zigbee2mqtt:debug 7/29/2019, 5:00:03 PM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/opt/zigbee2mqtt/data/database.db","coordBackupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:info 7/29/2019, 5:00:03 PM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: Error: No such file or directory, cannot open /dev/ttyACM0)
  zigbee2mqtt:info 7/29/2019, 5:01:03 PM Starting zigbee-shepherd
  zigbee2mqtt:error 7/29/2019, 5:01:03 PM Error while starting zigbee-shepherd! (Error: Error: No such file or directory, cannot open /dev/ttyACM0)
  zigbee2mqtt:error 7/29/2019, 5:01:03 PM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 7/29/2019, 5:01:03 PM Failed to start
        {"message":"Error: No such file or directory, cannot open /dev/ttyACM0","stack":"Error: Error: No such file or directory, cannot open /dev/ttyACM0"}
  zigbee2mqtt:error 7/29/2019, 5:01:03 PM Exiting...
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.5.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.5.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-07-29T16_01_03_903Z-debug.log

```

However, I am not sure that `/dev/ttyAMA0` is really the CC2531 stick. I have removed the stick from my raspberry pi and rebooted and after that `/dev/ttyAMA0` still exists. `ls -l` results in the following output:

```
pi@raspberrypi:~ $ ls -l /dev/ttyAMA0
crw-rw---- 1 root dialout 204, 64 Jul 29 17:02 /dev/ttyAMA0
```

I have not connected any other device to my raspberry pi so I am really not sure what this is. Maybe WiFi? This is a raspberry pi zero.

FutureCow, 2019-07-29:
What do you see with:
'' sudo dmesg''
After plugging in the Cc2531?

florianluediger, 2019-07-29:
This is the complete output of that command:

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.58+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1245 Fri Jul 12 17:20:08 BST 2019
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero W Rev 1.1
[    0.000000] Memory policy: Data cache writeback
[    0.000000] cma: Reserved 8 MiB at 0x1b400000
[    0.000000] On node 0 totalpages: 114688
[    0.000000]   Normal zone: 1008 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 114688 pages, LIFO batch:31
[    0.000000] random: get_random_bytes called from start_kernel+0x8c/0x49c with crng_init=0
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 113680
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:5A:B1:14 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=6389c98a-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 434444K/458752K available (6954K kernel code, 635K rwdata, 2076K rodata, 464K init, 797K bss, 16116K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xdc800000 - 0xff800000   ( 560 MB)
                   lowmem  : 0xc0000000 - 0xdc000000   ( 448 MB)
                   modules : 0xbf000000 - 0xc0000000   (  16 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (6956 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   ( 464 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 636 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 798 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 25148 entries in 74 pages
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000031] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000093] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000219] bcm2835: system timer (irq = 27)
[    0.000639] Console: colour dummy device 80x30
[    0.001308] console [tty1] enabled
[    0.001390] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060398] pid_max: default: 32768 minimum: 301
[    0.060980] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.061044] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.062509] CPU: Testing write buffer coherency: ok
[    0.063879] Setting up static identity map for 0x8200 - 0x8238
[    0.065345] devtmpfs: initialized
[    0.076991] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.077475] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.077554] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.078961] pinctrl core: initialized pinctrl subsystem
[    0.080705] NET: Registered protocol family 16
[    0.083874] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.091638] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.091713] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.091878] Serial: AMBA PL011 UART driver
[    0.095142] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.141008] bcm2835-dma 20007000.dma: DMA legacy API manager at (ptrval), dmachans=0x1
[    0.143692] SCSI subsystem initialized
[    0.144022] usbcore: registered new interface driver usbfs
[    0.144173] usbcore: registered new interface driver hub
[    0.144438] usbcore: registered new device driver usb
[    0.160998] raspberrypi-firmware soc:firmware: Attached to firmware from 2019-07-09 14:40, variant start
[    0.171030] raspberrypi-firmware soc:firmware: Firmware hash is 6c3fe3f096a93de3b34252ad98cdccadeb534be2
[    0.183464] clocksource: Switched to clocksource timer
[    0.283889] VFS: Disk quotas dquot_6.6.0
[    0.284077] VFS: Dquot-cache hash table entries: 1024 (order 0, 4096 bytes)
[    0.284443] FS-Cache: Loaded
[    0.284844] CacheFiles: Loaded
[    0.304715] NET: Registered protocol family 2
[... 178 lines cut ...]
[   15.487686] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[   15.488809] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Oct 23 2017 03:55:53 version 7.45.98.38 (r674442 CY) FWID 01-e58d219f
[   23.026650] 8021q: 802.1Q VLAN Support v1.8
[   24.273311] random: crng init done
[   24.273338] random: 7 urandom warning(s) missed due to ratelimiting
[   24.579097] uart-pl011 20201000.serial: no DMA platform data
[   25.127276] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[... cut ...]
[   29.381695] Bluetooth: L2CAP socket layer initialized
[   29.381747] Bluetooth: SCO socket layer initialized
[   29.409042] Bluetooth: HCI UART driver ver 2.3
[   29.409059] Bluetooth: HCI UART protocol H4 registered
[   29.409237] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   29.409519] Bluetooth: HCI UART protocol Broadcom registered
[   30.190364] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   30.190378] Bluetooth: BNEP filters: protocol multicast
[   30.190404] Bluetooth: BNEP socket layer initialized

```

FutureCow, 2019-07-29:
What do you see when you unplug and replug the CC2531?

I see something liek this:
`[ 3418.425945] usb 1-6: new full-speed USB device number 11 using xhci_hcd
[ 3418.581668] usb 1-6: New USB device found, idVendor=0451, idProduct=16a8, bcdDevice= 0.09
[ 3418.581674] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3418.581677] usb 1-6: Product: TI CC2531 USB CDC
[ 3418.581680] usb 1-6: Manufacturer: Texas Instruments
[ 3418.581682] usb 1-6: SerialNumber: __0X00124B0018ED382C
[ 3418.613411] cdc_acm 1-6:1.0: ttyACM0: USB ACM device
[ 3418.614022] usbcore: registered new interface driver cdc_acm
[ 3418.614024] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
`

florianluediger, 2019-07-29:
I don't get any new output from dmesg when reconnecting the Stick. I will get another micro USB to USB adapter and try that one, maybe mine is toast.

The power supply I am using for my pi should deliver enough power for both devices but I might try to change that one, too.

Thanks already for the help, I will report back tomorrow.

florianluediger, 2019-07-30:
Okay, I really solved the problem by using a new USB cable. I can't believe I have not figured this out earlier, that would have saved me a lot of fiddeling around.

Thanks a lot for your help! Closing this now as the issue had nothing to do with zigbee2mqtt.

closed 2019-07-30 by florianluediger