Koenkk/zigbee2mqtt #1795: zigbee2mqtt not starting - serialport is open

prankousky, 2019-07-30:
Hi,

I am trying to set up a `CC2531` USB stick. I actually just bought this stick pre-flashed, but my previous stick - which I had flashed myself and which is running fine on another host running `ioBroker` - does also not  work on my main computer.

OS is archlinux, my user has been added to group `uucp`, which means that I have access to `/dev/ttyACM0`; I checked and realized that`/dev/serial/by-id/usb-texas-instruments(etc)` **was** only accessible by group `root`, so I added myself to that as well. Upon reboot, suddenly the device was accessible by `uucp` as well, which I am also part of.

I just downloaded the most recent docker image of `koenkk/zigbee2mqtt` and ran it as described. [This](https://termbin.com/1fad) is the entire log. This `debug('The serialport ' + self._sp.path + ' is opened.');` caught my eye even as somebody who doesn't know much about the subject. Why is the path opened?

zigbee2mqtt version is `zigbee2mqtt version 1.5.1`
The logs will not display the firmware version of this device as it seems like it cannot access it at all.

So even though there might be a chance that **this** stick was not flashed correctly by the manufacturer, my *other* stick is currently working. It is connected to my debian server and successfully transmitting and receiving zigbee via ioBroker zigbee Adapter.

When I insert that stick in my computer and adapt the port, I will get the same error. When I plug it back in my server and restart ioBroker, it will work.

I then tried the new stick with my debian Server. Instead of docker I installed zigbee2mqtt directly to `/opt/zigbee2mqtt`, and when I run it, I get the same error. 

So from my point of view, this should neither be an issue with my computer (as it happens on the server as well), nor a hardware issue (as both the new stick and the one that works fine produce the error).

Can you please tell me how to fix this or what might cause the problem? I am currently attempting to switch from ioBroker and node-red to homeassistant and want to use "plain" zigbee2mqtt instead of the ioBroker adapter (which if I understand correctly is just an adaption of this *for* ioBroker anyway). The reason I bought this additional stick is that I want  to "transfer" my zigbee devices one by one to homeassistant rather than using the old (working) stick with zigbee2mqtt and having to set up all devices, new logics, etc. etc. at one.

Thanks in advance for your support :)

prankousky, 2019-07-30:
This can be deleted. I believe the arch kernel does not support the stick. I have re-flashed current firmware and can now run it on my debian machine - but still not on the arch computer.

prankousky, 2019-08-01:
Um, nevermind. I cannot use it under debian any longer, either. It did work once (when I closed this issue 2 days ago), and now each time I try and start, I get the same `debug('The serialport ' + self._sp.path # ' is opened.')` error. Both when running from `/opt/zigbee2mqtt` and as docker container, both as regular user (with sufficient rights) and as root.

Koenkk, 2019-08-02:
Read: https://www.zigbee2mqtt.io/information/zigbee2mqtt_fails_to_start.html first, if it does not help, please provide the log with `DEBUG=* npm start`.

prankousky, 2019-08-02:
@Koenkk Thank you. This is what I did:

* `➜ ls -l /dev/serial/by-id`
```
lrwxrwxrwx  root  root   -  Fri Aug  2 10:42:03 2019    usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0018E25CE3-if00 ⇒ ../../ttyACM1
lrwxrwxrwx  root  root   -  Wed Jul 31 12:37:33 2019    usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0018ED1B3A-if00 ⇒ ../../ttyACM0

```
* `➜  ~ test -w /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0018E25CE3-if00&& echo success || echo failure`
==> `success`

* `➜  ~ test -w /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0018ED1B3A-if00 && echo success || echo failure`
==> success

> Verify that you have a CC2530 or CC2531 (and not a CC2540)

I have the CC2530. It came pre-flashed, however, I manually flashed it again. After that, it did even work **once**. I was able to connect a new (just delivered, not connected to anything else yet) push button and receive the amount of pushes via `mqtt`. But after stopping the docker container and re-starting it later it did not work any longer.

> ModemManager is installed
Does this matter if I am running a docker container? Either way, I ran `sudo apt-get purge modemmanager` and got an output that roughtly translates to `paket >>modemmanager<< is not installed, so it will not be purged` (my output is in German, but I'm sure the English output would be similar).

When I run `id`, I get
`uid=1000(me) gid=1000(me) groups=1000(me),0(root),20(dialout),24(cdrom),25(floppy),29(audio),30(dip),44(video),46(plugdev),108(netdev),111(bluetooth),999(docker),1002(iobroker)`.

So my `me` user belongs to group `root`, which has access to both `ttyACM` devices. It does not make a difference whether I specify `/dev/ttyACM0/1` or `/dev/serial/by-id/<either of the usb sticks` on my config and when running the container.

This is my `DEBUG=* npm start` log result output (sorry, I had to copy the log output over `tmux` so the line numbers are included)

```
 20 0 info it worked if it ends with ok
 19 1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
 18 2 info using npm@6.10.2
 17 3 info using node@v10.16.0
 16 4 verbose run-script [ 'prestart', 'start', 'poststart' ]
 15 5 info lifecycle zigbee2mqtt@1.5.1~prestart: zigbee2mqtt@1.5.1
 14 6 info lifecycle zigbee2mqtt@1.5.1~start: zigbee2mqtt@1.5.1
 13 7 verbose lifecycle zigbee2mqtt@1.5.1~start: unsafe-perm in lifecycle true
 12 8 verbose lifecycle zigbee2mqtt@1.5.1~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/usr/local/bin:/usr/bin:    /bin:/usr/games
 11 9 verbose lifecycle zigbee2mqtt@1.5.1~start: CWD: /opt/zigbee2mqtt
 10 10 silly lifecycle zigbee2mqtt@1.5.1~start: Args: [ '-c', 'node index.js' ]
  9 11 silly lifecycle zigbee2mqtt@1.5.1~start: Returned: code: 1  signal: null
  8 12 info lifecycle zigbee2mqtt@1.5.1~start: Failed to exec start script
  7 13 verbose stack Error: zigbee2mqtt@1.5.1 start: `node index.js`
  6 13 verbose stack Exit status 1
  5 13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:326:16)
  4 13 verbose stack     at EventEmitter.emit (events.js:198:13)
  3 13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
  2 13 verbose stack     at ChildProcess.emit (events.js:198:13)
  1 13 verbose stack     at maybeClose (internal/child_process.js:982:16)
21  13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
  1 14 verbose pkgid zigbee2mqtt@1.5.1
  2 15 verbose cwd /opt/zigbee2mqtt
  3 16 verbose Linux 4.9.0-9-amd64
  4 17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
  5 18 verbose node v10.16.0
  6 19 verbose npm  v6.10.2
  7 20 error code ELIFECYCLE
  8 21 error errno 1
  9 22 error zigbee2mqtt@1.5.1 start: `node index.js`
 10 22 error Exit status 1
 11 23 error Failed at the zigbee2mqtt@1.5.1 start script.
 12 23 error This is probably not a problem with npm. There is likely additional logging output above.
 13 24 verbose exit [ 1, true ]
```
and this is the console output when I run `DEBUG=* npm start`

```
> zigbee2mqtt@1.5.1 start /opt/zigbee2mqtt
> node index.js

[... 38 lines cut ...]
                    debug('The serialport ' + self._sp.path + ' is opened.');
                                                       ^

TypeError: Cannot read property 'path' of null
    at Timeout.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/cc-znp/index.js:45:56)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.5.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.5.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/ben/.npm/_logs/2019-08-02T09_09_30_372Z-debug.log

```

What really bugs me is that it worked once and now won't any longer. I'd even get that the store I ordered from flashed the device incorrectly... But I already **re-flashed** it myself and then it did work once. This should mean (imho) that both it was flashed correctly (otherwise it would not work) as well as that my configuration was correct.

Yesterday when testing I also stopped the container and ran `docker pull koenkk/zigbee2mqtt:latest` to make sure I was running the correct version. However, this output is all from running directly on my machine. The docker container will always attempt to start zigbee shepherd, then stop, or not even get that far.

Koenkk, 2019-08-03:
I don't think this issue comes from the stick but is rather a configuration issue.

Can you post your `configuration.yaml`?

Rafik89, 2019-08-03:
Hello, I have a similar problem, earlier it worked without problems, now I don't want to start properly:

My specifications are RPi3+, Home Assistant 0.96.5, System HassOS 2.12, zigbee2mqtt-armhf with version 1.5.1.1.

CONFIG:

CONFIG:
```

{
  "data_path": "/share/zigbee2mqtt",
  "devices": "devices.yaml",
  "groups": "groups.yaml",
  "homeassistant": true,
  "permit_join": false,
  "mqtt": {
    "base_topic": "zigbee2mqtt",
    "server": "mqtt://127.0.0.1",
    "user": "mqqt",
    "password": "PASS"
  },
  "serial": {
    "port": "/dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B001936DD4F-if00"
  },
  "advanced": {
    "pan_id": 6724,
    "channel": 11,
    "network_key": [
      176,
      148,
      195,
      196,
      118,
      210,
      210,
      127,
      21,
      37,
      61,
      100,
      106,
      29,
      200,
      153
    ],
    "availability_blacklist": []
  },
  "ban": [],
  "whitelist": [],
  "queue": {}
}
```

```
[Info] Configuration backup found in /share/zigbee2mqtt/.configuration.yaml.bk. Skipping config backup.: No such file or directory
2019-08-04T01:33:33: PM2 log: Launching in no daemon mode
2019-08-04T01:33:33: PM2 log: App [npm:0] starting in -fork mode-
2019-08-04T01:33:33: PM2 log: App [npm:0] online
> zigbee2mqtt@1.5.1 start /zigbee2mqtt-1.5.1
> node index.js
  zigbee2mqtt:info 8/4/2019, 1:33:38 AM Logging to directory: '/share/zigbee2mqtt/log/2019-08-04.01-33-36'
  zigbee2mqtt:info 8/4/2019, 1:33:38 AM Starting zigbee2mqtt version 1.5.1 (commit #unknown)
  zigbee2mqtt:info 8/4/2019, 1:33:38 AM Starting zigbee-shepherd
  zigbee2mqtt:info 8/4/2019, 1:33:50 AM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: request timeout)
  zigbee2mqtt:info 8/4/2019, 1:34:50 AM Starting zigbee-shepherd
  zigbee2mqtt:error 8/4/2019, 1:35:02 AM Error while starting zigbee-shepherd! (Error: request timeout)
  zigbee2mqtt:error 8/4/2019, 1:35:02 AM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 8/4/2019, 1:35:02 AM Failed to start
	{"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/zigbee2mqtt-1.5.1/node_modules/zigbee-herdsman/dist/cc-znp/index.js:208:22)\n    at Object.onceWrapper (events.js:286:20)\n    at CcZnp.emit (events.js:198:13)\n    at Timeout.<anonymous> (/zigbee2mqtt-1.5.1/node_modules/zigbee-herdsman/dist/cc-znp/index.js:192:18)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}
  zigbee2mqtt:error 8/4/2019, 1:35:02 AM Exiting...
npm
 
ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.5.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.5.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-08-03T23_35_37_938Z-debug.log
2019-08-04T01:35:39: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2019-08-04T01:35:39: PM2 log: App [npm:0] starting in -fork mode-
2019-08-04T01:35:39: PM2 log: App [npm:0] online
> zigbee2mqtt@1.5.1 start /zigbee2mqtt-1.5.1
> node index.js
```

```
ls -l /dev/serial/by-id
lrwxrwxrwx    1 root     root            13 Aug  3 23:20 usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B001936DD4F-if00 -> ../../ttyACM0
```
```

test -w /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B001936DD4F-if00 && ech
o success || echo failure
success
# test -w /dev/ttyACM0 && echo success || echo failure
success
```

I also pressed the reset button on the USB stick, the green LED lights up for about a minute and then goes out.

Koenkk, 2019-08-04:
@rafik89 what firmware version are you running, does flashing https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing fix the problem?

Rafik89, 2019-08-04:
You're right! I flash new firmware for USB Stick and it moved on :)

prankousky, 2019-08-04:
This is my `configuration.yaml`

```
 15 homeassistant: true
 14 permit_join: true
 13 mqtt:
 (...)
  8 serial:
  7   port: /dev/ttyACM1
  6 devices:
  5   '0x00158d00026edd8e':
  4     friendly_name: unten/arbeitszimmer/open/fenster/open
  3     retain: true
  2   '0x00158d0002f97973':
  1     friendly_name: 'misc/button01'
16      retain: true
```

I don't know what to say... after deleting `.storage/ ` and then restarting, it worked fine. I have already paired some devices and get the expected payloads via mqtt. I am kinda afraid to stop the docker container though because last time it worked and I did that, it did not work any longer. But I assume that something was messed up in `.storage/` and that deleting fixed the issue.

My config was like this before too, though (when I got those error messages from above). So I don't think there is anything wrong with it..?

prankousky, 2019-08-05:
So it still works today. I had to restart the docker container multiple times because I changed `friend_name` of multiple newly added devices. I still assume that storage had something to do with this, so I'll close the issue. Thanks for this great project!

closed 2019-08-05 by prankousky