Koenkk/zigbee2mqtt #1796: No success adding HEIMAN CO sensor HS1CA-E

iz3man, 2019-07-30:
I purchased a HEIMAN CO Sensor, even though it's not in the supported device list. I was hoping to be able to add it myself (didn't do it before). Guess what, i didn't have success.

This is what happened after I long pressed the pairing button:

```
New device 'undefined' with address 0x000d6f000fadaff2 connected!
MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x000d6f000fadaff2","meta":{}}'
Device with modelID 'undefined' is not supported.
Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Device with modelID 'COSensor-EM' is not supported.
Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
```

So I added this to fromZigbee.js

```
heiman_co: {
    cid: 'genBasic',
    type: 'devChange',
    convert: (model, msg, publish, options) => {
        console.log(msg.data)
        },
},
```

After restart of zigbee2mqtt this showed up when trying to pair

```
PM Device with modelID 'COSensor-EM' is not supported.
PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
```

So i added this to devices.js

```
    {
        zigbeeModel: ['COSensor-EM'], 
        model: 'HS1CA-E', 
        vendor: 'HEIMAN', 
        description: 'Heiman Samrt Carbon Monoxide Sensor', 
        supports: 'co level', 
        fromZigbee: [fz.heiman_co],
        toZigbee: [],
    },
```

Now i get this, and am stuck. 

```
PM Message without device!
PM Device with modelID 'undefined' is not supported.
PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
PM Device incoming...
PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
PM Succesfully reenabled joining
```

Can someone please help me? I guess i did everything correctly, but i want to learn to do it, as i got other devices as well that are not supported and which I want to add to the database,

Thanks

Koenkk, 2019-07-31:
This device is already supported: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L3571 (switch to zigbee2mqt dev branch).

closed 2019-07-31 by Koenkk

iz3man, 2019-07-31:
LOL :) Great. Thanks. Still I'd liked to add it myself. Is someone willing to lead me through the process? There might be other products in the future where i may need to add them by myself.

Koenkk, 2019-07-31:
Process differs per devices, but main guidelines can be found here: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html