Koenkk/zigbee2mqtt #1798: Error: Cluster with name 'manuSpecificCluster' does not exist

yozik04, 2019-07-30:
# Bug Report
I am on the `latest-dev` branch in docker container. https://github.com/Koenkk/zigbee2mqtt/commit/93e5636f0c51f5fd403ea1250db9456bfe7b5fcc.
On latest firmware from https://github.com/koenkk/z-stack-firmware.
## What happened
I have pulled `latest-dev` from docker hub and restarted zigbee2mqtt with same config using docker-compose.
zigbee2mqtt fails to start.
```
  zigbee2mqtt:info 7/30/2019, 2:54:27 PM Starting zigbee2mqtt version 1.5.1 (commit #unknown)
  zigbee2mqtt:info 7/30/2019, 2:54:27 PM Starting zigbee-shepherd
  zigbee2mqtt:info 7/30/2019, 2:54:29 PM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds) (Error: Cluster with name 'manuSpecificCluster' does not exist)
  zigbee2mqtt:info 7/30/2019, 2:55:29 PM Starting zigbee-shepherd
  zigbee2mqtt:error 7/30/2019, 2:55:31 PM Error while starting zigbee-shepherd! (Error: dev exists, unregister it first.)
  zigbee2mqtt:error 7/30/2019, 2:55:31 PM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 7/30/2019, 2:55:31 PM Failed to start
        {"message":"dev exists, unregister it first.","stack":"Error: dev exists, unregister it first.\n    at /app/node_modules/zigbee-herdsman/dist/lib/shepherd.js:449:19\n    at Promise.apply (/app/node_modules/zigbee-herdsman/node_modules/q/q.js:1185:26)\n    at Promise.promise.promiseDispatch (/app/node_modules/zigbee-herdsman/node_modules/q/q.js:808:41)\n    at /app/node_modules/zigbee-herdsman/node_modules/q/q.js:1411:14\n    at runSingle (/app/node_modules/zigbee-herdsman/node_modules/q/q.js:137:13)\n    at flush (/app/node_modules/zigbee-herdsman/node_modules/q/q.js:125:13)\n    at process._tickCallback (internal/process/next_tick.js:61:11)"}
  zigbee2mqtt:error 7/30/2019, 2:55:31 PM Exiting...
```

I have logged into container and recursively grepped `manuSpecificCluster`. It only exists in the database.db file. For WXKG01LM - Xiaomi MiJia wireless switch.

Interesting that if I run `latest` everything starts fine.
## What did you expect to happen
Normal start
## How to reproduce it (minimal and precise)
Just run latest dev docker container.
## Debug Info
zigbee2mqtt version: 1.5.1
CC253X firmware version: 20190619

didiht, 2019-07-31:
Exactly the same happens in my environment with Aqara sensors WSDCGQ11LM.
I cannot change to dev branch at the moment.

Koenkk, 2019-07-31:
Thanks, fixed in latest dev

closed 2019-07-31 by Koenkk

prein2, 2019-07-31:
yes, working!

prein2, 2019-07-31:
that comment was a little too quick:

zigbee2mqtt is working and stays working, but this message occurs in the log:

`7/31/2019, 8:37:12 PM - error:  Error: Cluster with ID '64636' does not exist
    at getClusterByID (/app/node_modules/zigbee-herdsman/dist/zcl/index.js:30:15)
    at Object.getClusterLegacy (/app/node_modules/zigbee-herdsman/dist/zcl/index.js:40:19)
    at /app/node_modules/zigbee-herdsman/dist/lib/components/af.js:342:29
    at Object._.forEach (/app/node_modules/zigbee-herdsman/node_modules/busyman/index.js:411:27)
    at Object.af.zclClustersReq (/app/node_modules/zigbee-herdsman/dist/lib/components/af.js:341:7)
    at /app/node_modules/zigbee-herdsman/dist/lib/components/event_handlers.js:252:32
    at _fulfilled (/app/node_modules/zigbee-herdsman/node_modules/q/q.js:854:54)
    at /app/node_modules/zigbee-herdsman/node_modules/q/q.js:883:30
    at Promise.promise.promiseDispatch (/app/node_modules/zigbee-herdsman/node_modules/q/q.js:816:13)
    at /app/node_modules/zigbee-herdsman/node_modules/q/q.js:877:14`

yozik04, 2019-07-31:
Runs fine after last commit. Thank you.

Koenkk, 2019-07-31:
@prein2 should also be fixed (wait until new docker image is up).