Koenkk/zigbee2mqtt #1800: Issue with timestamps using docker container

tavalin, 2019-07-31:
# Bug Report
## What happened

There seems to be a few issues related to date time stamps using the docker image.

I'm running the zigbee2mqtt arm32-dev docker image on a Raspberry Pi with last_seen enabled in the configuration.yaml.  

### Issue 1
The std out logging message (using `docker logs`) have incorrect timestamps e.g.  14:40PM BST is currently shown as `7/31/2019, 1:40:02 PM`

### Issue 2
ISO_8601_local format are not giving the correct results.

I have mapped the `/etc/timezone` volumes from my host linux environment to the container.

```
cat /etc/timezone
Europe/London
```

The timezone output on the MQTT topic is incorrect. 

## What did you expect to happen

**ISO_8601_local gives**:
A last_seen timestamp for 14:25 BST (currently UTC+1) is shown as 
`"last_seen":"2019-07-31T13:25:34+00:00"`

I would expect:

`"last_seen":"2019-07-31T14:25:34+01:00"`

**ISO_8601 gives:**

A last_seen timestamp for 14:32 BST (currently UTC+1) is shown as
`"last_seen":"2019-07-31T13:32:38.859Z"`
 which looks correct (I think) but does not appear as easy for me to convert to a local time.

## How to reproduce it (minimal and precise)

1) Run docker container mounting /etc/localtime volume from host to container
2) Enable `last_seen` configuration specifying` ISO_8601_local`
3) Check output message in MQTT

## Debug Info
zigbee2mqtt version: v1.5.1 (arm32-dev tag)

Koenkk, 2019-07-31:
Can you try mounting `/etc/localtime:/etc/localtime:ro`

prein2, 2019-07-31:
why don't you just pass enviroment value to your docker:

`
docker run --name=Zigbee2mqtt-1 \
   -e TZ=Europe/Amsterdam \
   -it \
   -v /szigbeedatalocation :/app/data \
   --device=/dev/ttyACM0 \
   koenkk/zigbee2mqtt:latest
`

tavalin, 2019-07-31:
@Koenkk mounting `/etc/localtime:/etc/localtime:ro` results in:
 `"last_seen":"2019-07-31T17:46:32-01:00"` when infact that should be 19:46.  Looks like some weird timezone calculation is being done.

@prein2 setting the environment variable seems to have done the trick:

`"last_seen":"2019-07-31T19:57:54+01:00"` so I will go with that. It's probably worth adding this to the docs though if setting the environment variable is the recommended approach to setting the timezone of the container.

Koenkk, 2019-08-01:
Added to the docs, thanks @prein2 !

closed 2019-08-01 by Koenkk