Koenkk/zigbee2mqtt #1801: RTCGQ01LM / WSDCGQ11LM - No battery info

zibous, 2019-07-31:
**xiaomi human body movement sensor - RTCGQ01LM (2019.01)**

```
zigbee2mqtt version: zigbee2mqtt 1.5.1 (latest dev)
npm v6.10.2
node v10.14.2
CC253X firmware version: latest

## Hardware
CC2531 USB stick with external antenna as coordinator 
CC2531 USB stick as router
```
## configuration.yaml
```yaml
homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://*******:1883'
  user: *******
  password: ********
  reject_unauthorized: false
  client_id: rb3zigbee2mqtt
  include_device_information: true
serial:
  port: /dev/ttyACM0
  disable_led: true
advanced:
  channel: 11
  cache_state: false
  baudrate: 115200
  rtscts: true
  last_seen: ISO_8601
  elapsed: true
  log_level: debug
  report: true
  homeassistant_discovery_topic: homeassistant
queue:
  delay: 250
  simultaneously: 5
devices:
  '0x00158d000358e8cd':
    friendly_name: 'bewegungsmelder1'
    retain: false
```
Why can not I get any information about the battery status, voltage from this device. 
- "hwVersion":"unknown" ??
-  "swBuildId":"unknown" ??
-  "dateCode":"unknown" ??
-   battery info ??

Do I do something wrong with the configuration?

## Debug Info
```json
zigbee2mqtt:info 7/31/2019, 3:14:17 PM MQTT publish: 
  topic 'zigbee2mqtt/bewegungsmelder1', 
  payload '{
     "occupancy":true,
     "linkquality":120,
     "last_seen":"2019-07-31T13:14:17.628Z",
     "elapsed":5473,"device":{       
        "ieeeAddr":"0x00158d000358e8cd",
        "friendlyName":"bewegungsmelder1",
        "type":"EndDevice",
        "nwkAddr":36520,
        "manufId":4151,
        "manufName":"LUMI",
        "powerSource":"Battery",
        "modelId":"lumi.sensor_motion",
        "hwVersion":"unknown",
        "swBuildId":"unknown",
        "dateCode":"unknown",
        "status":"online"
     }
  }'
```

**Same problem I have with the Xiaomi WSDCGQ11LM - temperature & humidity sensor** 
Even with this device I miss the information about the battery status

```json
{
  "pressure" : 969,
  "linkquality" : 134,
  "last_seen" : "2019-07-31T14:02:33.227Z",
  "elapsed" : 23,
  "device" : {
    "ieeeAddr" : "0x00158d00036b56d1",
    "friendlyName" : "temperatursensor1",
    "type" : "EndDevice",
    "nwkAddr" : 44696,
    "manufId" : 4151,
    "manufName" : "LUMI",
    "powerSource" : "Battery",
    "modelId" : "lumi.weather",
    "hwVersion" : 30,
    "swBuildId" : "3000-0001",
    "dateCode" : "20161129",
    "status" : "online"
  }
}
```

Koenkk, 2019-08-01:
This is because during pairing the device did not answer all requests.  Note that the battery percentage itself can take some time to pop-up (max 24 hours).

zibous, 2019-08-01:
@Koenkk 
Thank you for the information and explanation.

Yes it is. Now I get (after 20h) data, but if I also see that there are always interruptions of data delivery. It looks like data is not always delivered. 

Is that the case with the Zigbee devices or something is wrong with my configuration ?

**MiJia temperature & humidity sensor Xiaomi WSDCGQ11LM - RTCGQ01LM**
[image]

**MiJia human body movement sensor - WSDCGQ11LM**
[image]

Koenkk, 2019-08-01:
You will see this when sensor become unavailable, did you restart zigbee2mqtt or the pc?

zibous, 2019-08-01:
@Koenkk 

>> did you restart zigbee2mqtt or the pc?
No not at 23:00, not working on this time...

zigbee2mqtt, USB Stick  CC2531 coordinator is installed on a Raspberry 3+ and runs 24/7.

One possibility would be that the Raspberry has a timeout problem with the port: / dev / ttyACM0

But it could also be on the router, I do not know why, but this sometimes goes offline. Whenever I restart the coordinator, then I have to manually reconnect the router, because this does not connect automatically.

```
zigbee2mqtt 1.5.1
npm v6.10.2
node v10.14.2
firmware CC2531 coordinator: 20190608
firmware CC2531 router: 2018-09-26
```

But when restarting the coordinator I see in the log that all data is supplied by the devices:

```json
{
  "occupancy" : false,
  "linkquality" : 0,
  "last_seen" : "2019-08-01T10:02:11.882Z",
  "battery" : 100,
  "voltage" : 3075,
  "device" : {
    "ieeeAddr" : "0x00158d000358e8cd",
    "friendlyName" : "bewegungsmelder1",
    "type" : "EndDevice",
    "nwkAddr" : 36520,
    "manufId" : 4151,
    "manufName" : "LUMI",
    "powerSource" : "Battery",
    "modelId" : "lumi.sensor_motion",
    "hwVersion" : "unknown",
    "swBuildId" : "unknown",
    "dateCode" : "unknown",
    "status" : "online"
  }
}

{
  "temperature" : 24.36,
  "linkquality" : 39,
  "last_seen" : "2019-08-01T11:09:24.588Z",
  "humidity" : 57.35,
  "pressure" : 968,
  "battery" : 100,
  "voltage" : 3025,
  "device" : {
    "ieeeAddr" : "0x00158d00036b7ae0",
    "friendlyName" : "temperatursensor2",
    "type" : "EndDevice",
    "nwkAddr" : 58446,
    "manufId" : 4151,
    "manufName" : "LUMI",
    "powerSource" : "Battery",
    "modelId" : "lumi.weather",
    "hwVersion" : 30,
    "swBuildId" : "3000-0001",
    "dateCode" : "20161129",
    "status" : "online"
  }
}

{
  "battery" : 100,
  "voltage" : 3045,
  "linkquality" : 42,
  "last_seen" : "2019-08-01T11:46:09.563Z",
  "device" : {
    "ieeeAddr" : "0x00158d00028079c1",
    "friendlyName" : "wassersensor1",
    "type" : "EndDevice",
    "nwkAddr" : 24962,
    "manufId" : 4151,
    "manufName" : "LUMI",
    "powerSource" : "Battery",
    "modelId" : "lumi.sensor_wleak.aq1",
    "hwVersion" : 2,
    "swBuildId" : "3000-0001",
    "dateCode" : "20170721",
    "status" : "online"
  }
}
``

Koenkk, 2019-08-02:
The values you see here are cached values from `data/state.json`. Can you provide the zigbee2mqtt log of around 7:00 (regarding the image below)?

[image]

zibous, 2019-08-02:
@Koenkk 

Unfortunately I do not have the log file anymore. I have re-set the environment, but I can not figure out where the mistake is. I have replaced the router with 2 Osram Smart + Plugs and all devices in one room (distance up to 2m).

1. New installation on a new Raspberry Pi 3 Model B
   ```     
      Raspbian Linux 10 (Debian 10 buster)
      node v12.5.0
      npm 6.10.1
      CC2531 USB stick with external antenna as coordinator 
      zigbee2mqtt version 1.5.1 (commit #ac3b924)
      Coordinator firmware version: '20190608'
   ```
2. Starting zigbee2mqtt version 1.5.1 (commit # ac3b924)
3. Pairing 2 OSRAM Smart+ Plugs
4. ***Both OSRAM Smart+ Plugs are online***
5. Restart Coordinator
6. ***Both OSRAM Smart+ Plugs are offline and never go online***
    But switch can be switched (ON / OFF) with mqtt command
7. Power off OSRAM Smart + Plug and Power On
8. ***Both OSRAM Smart+ Plugs now online again.***
9. With the other devices (temperature, motion and water sensor) it happens again and again that information is missing or only supplied from the cache.

Why do not the OSRAM Smart + Plugs (routers) reconnect when the coordinator is restarted? Same problem I had with the CC2531 router.

Log see: https://pastebin.com/kupv3BeT
Database see: https://pastebin.com/pfRxmByz

***configuration***
```yaml
homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://mqttserver:1883'
  user: mqttUserName
  password: thisIsNotPublic
  reject_unauthorized: false
  client_id: rb3zigbee2mqtt
  include_device_information: true
serial:
  port: /dev/ttyACM0
  disable_led: true
advanced:
  channel: 14
  cache_state: false
  baudrate: 115200
  rtscts: true
  last_seen: ISO_8601
  elapsed: true
  log_directory: /var/log/zigbee2mqtt/
  log_level: debug
  report: true
  homeassistant_discovery_topic: homeassistant
devices:
  '0x00158d00028208ae':
    friendly_name: '0x00158d00028208ae'
    retain: false
  '0x00158d00028079c1':
    friendly_name: 'wassersensor1'
    retain: false
  '0x00158d00028079c0':
    friendly_name: 'wassersensor0'
    retain: false
  '0x00158d00036b7ae0':
    friendly_name: 'wassersensor2'
    retain: false
  '0x00158d00036b56d1':
    friendly_name: 'temperatursensor1'
    retain: false
  '0x00158d000358e8cd':
    friendly_name: 'bewegungsmelder1'
    retain: false
  '0x7cb03eaa0a00ebba':
    friendly_name: 'smartplug1'
    retain: false
  '0x7cb03eaa0a0145d9':
    friendly_name: 'smartplug2'
    retain: false
```

Log see: https://pastebin.com/kupv3BeT
Database see: https://pastebin.com/pfRxmByz

CC2531 ZigBee USB-Sick: https://www.ebay.at/itm/CC2531-ZigBee-USB-Sick-zigbee2mqtt-ioBroker-FHEM-SMA-8dbi-Antenne-Firmware/113384283158?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

zibous, 2019-08-03:
@Koenkk

I'm not sure, but I think I found the explanation. Whenever the "linkquality" of a sensor drops, not all data is delivered at state_topic:

[image]

```yaml
- platform: "mqtt"
  name: "Motion Bad Battery Status"
  state_topic: "zigbee2mqtt / motion1"
  availability_topic: "zigbee2mqtt / bridge / state"
  unit_of_measurement: "%"
  device_class: "battery"
  value_template: "{{value_json.battery}}"

- platform: "mqtt"
  name: "Motion Bad Signal Status"
  state_topic: "zigbee2mqtt / motion1"
  availability_topic: "zigbee2mqtt / bridge / state"
  unit_of_measurement: "dbm"
  value_template: "{{value_json.linkquality}}"
```
Therefore, the data can not be read out for all attributes in the graphics for this time. But if the link quality increases, then all the data is delivered. This affects only the devices "***RTCGQ01LM, WSDCGQ11LM, SJCGQ11LM***", but this effect does not occur with the ***OSRAM AB3257001NJ***.

```
8/3/2019, 07:36:01 - info:  
  MQTT publish: topic 'zigbee2mqtt/bewegungsmelder1', payload 
  '{
     "occupancy":false,
     "linkquality":139,  <<-- down
     "last_seen":"2019-08-03T05:36:01.730Z",
    "elapsed":90000,
     "device":{"ieeeAddr":"0x00158d000358e8cd",
          "friendlyName":"bewegungsmelder1",
       .....}
   }'

8/3/2019, 08:07:53 - info: 
  MQTT publish: topic 'zigbee2mqtt/bewegungsmelder1', payload 
  '{ 
     "battery":100,
     "voltage":3085,
     "linkquality":147,  <<--- up
     "last_seen":"2019-08-03T06:07:53.645Z",
     "elapsed":1911915,
     "device":{"ieeeAddr":"0x00158d000358e8cd",
        "friendlyName":"bewegungsmelder1",
        .....}
  }'
```

**Is there a solution for this ?**

Log see: https://pastebin.com/t9RcCPtg

Koenkk, 2019-08-03:
Can you post your configuration.yaml?

zibous, 2019-08-03:
@Koenkk 

Yes, this is the what i use:

```yaml
homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://IPADDESS:1883'
  user: mqttUserName
  password: thisIsNotPublic
  reject_unauthorized: false
  client_id: rb3zigbee2mqtt
  include_device_information: true
serial:
  port: /dev/ttyACM0
  disable_led: true
advanced:
  channel: 14
  cache_state: false
  baudrate: 115200
  rtscts: true
  last_seen: ISO_8601_local
  elapsed: true
  log_directory: /var/log/zigbee2mqtt/
  log_level: debug
  report: true
  homeassistant_discovery_topic: homeassistant
devices:
  '0x00158d00028208ae':
    friendly_name: 'wassersensor2'
    retain: false
  '0x00158d00028079c1':
    friendly_name: 'wassersensor1'
    retain: false
  '0x00158d00028079c0':
    friendly_name: 'wassersensor0'
    retain: false
  '0x00158d00036b7ae0':
    friendly_name: 'temperatursensor2'
    retain: false
  '0x00158d00036b56d1':
    friendly_name: 'temperatursensor1'
    retain: false
  '0x00158d000358e8cd':
    friendly_name: 'bewegungsmelder1'
    retain: false
  '0x7cb03eaa0a00ebba':
    friendly_name: 'smartplug1'
    retain: false
  '0x7cb03eaa0a0145d9':
    friendly_name: 'smartplug2'
    retain: false
```

Koenkk, 2019-08-03:
You use `cache_state: false` that means that states will not be cached, e.g.

- Sensor sends occupancy -> occupancy: true message reported
- Sensor sends battery -> battery: 100 reported

With `cache_state: true` (default), this will be:
- Sensor sends occupancy -> occupancy: true message reported
- Sensor sends battery -> battery: 100 and occupancy: true reported

When Home Assistant doesn't receive complete payloads it's is expected you will see these gaps. You should also see a warning when starting zigbee2mqtt that the home assistant integration won't work properly.

Closing as this is expected behaviour.

closed 2019-08-03 by Koenkk

zibous, 2019-08-03:
@Koenkk 

Thanks, so i will set  "cache_state: true".

Is there a solution for the Osram Smart?
They always go offline when I restart the Coordinator.

see:
`Why do not the OSRAM Smart + Plugs (routers) reconnect when the coordinator is restarted? Same problem I had with the CC2531 router. `

zibous, 2019-08-03:
cache_state: true, Coordinator restarted.

[image]

OSRAM Smart + Plugs (routers) are OFFLINE, but MQTT command "ON/OFF" works.

Koenkk, 2019-08-03:
That offline/online status is not reliable (will be fixed after zigbee-herdsman refactor: https://github.com/Koenkk/zigbee-herdsman/issues/13).

zibous, 2019-08-03:
Thanks for the support, great job 👍