Koenkk/zigbee2mqtt #1802: Three connections on graph

abmantis, 2019-07-31:
# Bug Report
## What happened
There are three routes between some devices:
[image]

## What did you expect to happen
Only two connections.

## How to reproduce it (minimal and precise)
Not sure, it just happens.

## Debug Info
zigbee2mqtt version: dev
CC253X firmware version: 20190223

Koenkk, 2019-08-01:
@clockbrain could you help with this?

clockbrain, 2019-08-01:
Only a guess but its possible some lqi or route requests from one network scan are still outstanding when later network map scan starts so it collects those results from different scans all together in the one map.

@abmantis how many devices do you have? have you changed the default queue settings? how frequently are you requesting a network map?

abmantis, 2019-08-01:
@clockbrain  I have 10 devices. I have not changed the queue settings. I don't have it updating automatically ATM. I just update it manually when I want to check it.

Woodcat64, 2019-08-02:
This happened to me as well
zigbee2mqtt version: 1.5.1.1
CC2531_DEFAULT_20190608

[image]

clockbrain, 2019-08-02:
@abmantis & @Woodcat64 are either of you using Admin Panel for Node-Red because that by default triggers a network map request every 30 seconds?

If that regular Node-Red request happens in the background and overlaps with a manually triggered map request then I can see how the results might get mangled.

abmantis, 2019-08-02:
@clockbrain no, no automatic updates here (and I can be sure of that because of the timestamp on the graph).

Woodcat64, 2019-08-03:
@clockbrain, no automatic  updates here as well. After z2m restart, all is back to normal.

renedis, 2019-08-06:
Same story here. HA auto refreshes the map every 10 minutes with an automation.

Currently 38 devices.
zigbee2mqtt version: 1.5.1.1
CC2530_CC2591_SOURCE_ROUTING_20190619

[image]

clockbrain, 2019-08-06:
Can anyone provide a log file with debug so I can figure out what is causing this. At the moment I can only make a guess as I don't see this on my network map.

Is it repeatable or only a very rare occurrence?

closed 2019-10-12 by stale[bot]