Koenkk/zigbee2mqtt #1805: MiJia temperature & humidity sensor (WSDCGQ01LM) restarts the zigbee2mqtt when button press

kvvoff, 2019-08-01:
I am using version zigbee2mqtt 1.5.1.1 Hassio addon on HassOS and MiJia temperature & humidity sensor (WSDCGQ01LM)

Everything worked as it should, but once the readings stopped. I pressed a button on the sensor and this led to a restart of the zigbee2mqtt . And so now every time I press the button zigbee2mqtt  restarts. This is what the log shows at this moment:
[image]
[image]

`/zigbee2mqtt-1.5.1/lib/extension/deviceReceive.js:109
                if (c.cid === cid) {
                      ^
TypeError: Cannot read property 'cid' of undefined
    at mappedDevice.fromZigbee.filter (/zigbee2mqtt-1.5.1/lib/extension/deviceReceive.js:109:23)
    at Array.filter (<anonymous>)
    at DeviceReceive.onZigbeeMessage (/zigbee2mqtt-1.5.1/lib/extension/deviceReceive.js:105:52)
    at extensions.filter.forEach (/zigbee2mqtt-1.5.1/lib/controller.js:143:31)
    at Array.forEach (<anonymous>)
    at Controller.onZigbeeMessage (/zigbee2mqtt-1.5.1/lib/controller.js:143:14)
    at Zigbee.onMessage (/zigbee2mqtt-1.5.1/lib/zigbee.js:275:18)
    at ZShepherd.emit (events.js:198:13)
    at ZShepherd.<anonymous> (/zigbee2mqtt-1.5.1/node_modules/zigbee-herdsman/dist/lib/shepherd.js:125:14)
    at ZShepherd.emit (events.js:198:13)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@1.5.1 start: `node index.js`
npm ERR!
 Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.5.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm
 ERR!     /root/.npm/_logs/2019-08-01T17_22_11_062Z-debug.log
2019-08-01T20:22:15: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2019-08-01T20:22:15: PM2 log: App [npm:0] starting in -fork mode-
2019-08-01T20:22:15: PM2 log: App [npm:0] online
> zigbee2mqtt@1.5.1 start /zigbee2mqtt-1.5.1
> node index.js`

The same thing happened before the update (on version 1.4)

Koenkk, 2019-08-02:
Did you map a custom zigbee-shepherd-converters? If so, remove it.

kvvoff, 2019-08-02:
@Koenkk ok, thanks. I will do it and write the result

kvvoff, 2019-08-04:
@Koenkk Indeed, I turned off zigbee-shepherd-converters and my temperature sensor works as it should. But why did this happen?  How do I use zigbee-shepherd-converters?

Koenkk, 2019-08-05:
Good! I'm not sure why you added the mapping in the first place, this should only be done when wanting to add support for new devices.

Closing as this is solved.

closed 2019-08-05 by Koenkk

kvvoff, 2019-08-05:
@Koenkk[ I added a new light bulb](https://github.com/Koenkk/zigbee-shepherd-converters/issues/523) and forgot to remove zigbee-shepherd-converters :) Thanks!