Koenkk/zigbee2mqtt #1806: Cannot publish message to device because XXXXX is not known by zigbee-shepherd

gericho, 2019-08-02:
Device IKEA Tradfri
https://www.zigbee2mqtt.io/devices/LED1650R5.html
```

Aug 02 01:35:15 HomeAssistant npm[19673]:   zigbee2mqtt:info 8/2/2019, 1:35:15 AM Connecting to MQTT server at mqtt://192.168.0.8
Aug 02 01:35:16 HomeAssistant npm[19673]:   zigbee2mqtt:error 8/2/2019, 1:35:16 AM Zigbee cannot determine endpoint for '0x000b57fffe991092'
Aug 02 01:35:16 HomeAssistant npm[19673]:   zigbee2mqtt:error 8/2/2019, 1:35:16 AM Cannot publish message to device because '0x000b57fffe991092' is not known by zigbee-shepherd
Aug 02 01:35:16 HomeAssistant npm[19673]:   zigbee2mqtt:error 8/2/2019, 1:35:16 AM Zigbee cannot determine endpoint for '0x000b57fffea714ed'
Aug 02 01:35:16 HomeAssistant npm[19673]:   zigbee2mqtt:error 8/2/2019, 1:35:16 AM Cannot publish message to device because '0x000b57fffea714ed' is not known by zigbee-shepherd
Aug 02 01:35:16 HomeAssistant npm[19673]:   zigbee2mqtt:error 8/2/2019, 1:35:16 AM Zigbee cannot determine endpoint for '0x000b57fffebd9502'
Aug 02 01:35:16 HomeAssistant npm[19673]:   zigbee2mqtt:error 8/2/2019, 1:35:16 AM Cannot publish message to device because '0x000b57fffebd9502' is not known by zigbee-shepherd
Aug 02 01:35:17 HomeAssistant npm[19673]:   zigbee2mqtt:info 8/2/2019, 1:35:17 AM zigbee-shepherd ready
Aug 02 01:35:18 HomeAssistant npm[19673]:   zigbee2mqtt:info 8/2/2019, 1:35:18 AM Connected to MQTT server
```
I get this `is not known by zigbee-shepherd` code, but everything works, but Hass keeps old removed lamps in enitty list also after integration removal.

Items was removed by publishing a packet from HA:
```
zigbee2mqtt/bridge/config/remove
'0x000b57fffebd9502'
```
and the log says
```

Aug 02 01:59:09 HomeAssistant npm[19673]:   zigbee2mqtt:info 8/2/2019, 1:59:09 AM Successfully removed '0x000b57fffebd9502'
Aug 02 01:59:09 HomeAssistant npm[19673]:   zigbee2mqtt:info 8/2/2019, 1:59:09 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"'0x000b57fffebd9502'"}'
```
then 

`sudo systemctl restart zigbee2mqtt`
the messages `is not known by zigbee-shepherd` comes up again
`

The Coordinator firmware version is the '20190425'`

gericho, 2019-08-02:
confirmed also on Coordinator firmware version `CC2531_DEFAULT_20190608`

Koenkk, 2019-08-03:
It seems that after removing you try to execute commands on the device?

gericho, 2019-08-03:
its seems yes, but I stopped HA completely and restart zigbee2mqtt, same messages 
` Zigbee cannot determine endpoint for '0x000b57fffebd9502'`

Koenkk, 2019-08-04:
Do you see any occurrences of `0x000b57fffebd9502` in `data/database.db`?

gericho, 2019-08-04:
> Do you see any occurrences of `0x000b57fffebd9502` in `data/database.db`?

unfortunately not but the ghost entity remains, also in HA!

**Solved by** reinstalling zigbee2mqtt, restoring /data (not sure it is necessary), deleting /var/lib/mosquitto/mosquitto.db 

**step by step explained in the last comment here**
[https://github.com/Koenkk/zigbee2mqtt/issues/835](url)

Koenkk, 2019-08-05:
Thanks for the explanation!

closed 2019-08-05 by Koenkk