Koenkk/zigbee2mqtt #1808: Cannot Reconnect Device

Hoanthieu, 2019-08-02:
# Bug Report
Cannot Connect Device again after get modelID
## What happened
When I first connected my Lock to my termux, i can get that log and and the information was automatically written to data/database.db and data/configuration.
After that, I write a block of code to support that lock, but cannot control or get state.
I delete all saved information, and reconnect, but no information is visible in log 
## What did you expect to happen
show:
**zigbee2mqtt:warn 8/2/2019, 7:59:07 AM Device with modelID 'TY0A01' is not supported.
zigbee2mqtt:warn 8/2/2019, 7:59:07 AM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html**
when I try to connect Lock again
## How to reproduce it (minimal and precise)
connect, delete all information and reconnect
Lock model: TY0A01
Vendor: Pine World
## Debug Info
zigbee2mqtt version: 1.5.1
CC253X firmware version:

Koenkk, 2019-08-03:
I don't understand the problem completely, but note that you will only see the above messages when the device sends a message.

You can use: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L4034 as a start.

closed 2019-10-09 by stale[bot]

yoobeedoobee, 2020-04-17:
If this is still relevant, try to bring your device closer to the zigbee antenna when pairing