Koenkk/zigbee2mqtt #1809: Support to Develco Smoke Alarm SMSZB-120

jjposio, 2019-08-02:
# Bug Report

Support for [Develco Smoke Alarm SMSZB-120.](https://www.develcoproducts.com/products/sensors-and-alarms/smoke-alarm/)
This specific smoke alarm seems to have IAS WD cluster and should possibly work as siren as well.

## What happened

This smoke alarm looks a lot like Heiman counterpart, but has also IAS WD cluster available, that is missing from some of the supported smoke alarms. IAS WD cluster is mentioned in [technical manual](https://www.smarthome-store.de/media/documents/smoke-alarm-technical-manual.pdf).

The initial setup was done and smoke alarm reports battery, linkquality, smoke and temperature values, 
...
zigbee2mqtt:info 8/2/2019, 2:13:46 PM MQTT publish: topic 'zigbee2mqtt/0x0015bc0031002651', payload '{"battery_low":false,"linkquality":23,"temperature":23.93,"smoke":false}'
...

But no luck with triggering the siren off.

## What did you expect to happen

Sended following via developer / mqtt in order to get siren on.
zigbee2mqtt/[Device]/set
  {"warning": ""}

Log indicates this:
  zigbee2mqtt:info 8/2/2019, 2:14:42 PM Zigbee publish to device '0x0015bc0031002651', ssIasWd - startWarning - {"startwarninginfo":53,"warningduration":10} - {"manufSpec":0,"disDefaultRsp":0} - 1

But there is no sound coming from the siren what so ever. Any idea what is wrong?

## How to reproduce it (minimal and precise)

Initial setup to devices.js

     // develco
     {
        zigbeeModel: ['SMSZB-120'],
        model: 'SMSZB-123',
        vendor: 'Develco Products A/S',
        description: 'Smoke alarm and siren',
        fromZigbee:[fz.heiman_smoke,fz.generic_temperature,fz.battery_200,fz.ignore_iaszone_change],
        toZigbee:[tz.warning],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
                const ep2 = shepherd.find(ieeeAddr, 35);
                const ep3 = shepherd.find(ieeeAddr, 38);
                const actions = [
                        (cb) => ep2.bind('genPowerCfg',coordinator,cb),
                        (cb) => ep3.bind('msTemperatureMeasurement', coordinator, cb),
                        (cb) => ep2.write('ssIasZone','iasCieAddr',coordinator.device.getIeeeAddr(), cb),
                        (cb) => ep2.functional('ssIasZone','enrollRsp',{enrollrspcode: 0, zoneid: 23}, cb),
                        (cb) => ep2.report('genPowerCfg','batteryVoltage',0,65534,0,cb),
                        (cb) => ep3.report('msTemperatureMeasurement','measuredValue',30, 600, 1, cb),
                ];
                execute(ep3, actions, callback, 1000);
        },
    },

Send following via developer / mqtt
zigbee2mqtt/[Device]/set
{"warning": ""}

## Debug Info
zigbee2mqtt version: 1.5.1.1
CC253X firmware version:

database.db entry for the device in attached file.
[Develco_SmokeAlarm_dbEnty.txt](https://github.com/Koenkk/zigbee2mqtt/files/3461051/Develco_SmokeAlarm_dbEnty.txt)

And state.json after initial setup:
    "0x0015bc0031002651": {
        "battery_low": false,
        "linkquality": 23,
        "temperature": 25.06,
        "smoke": false
    },

Koenkk, 2019-08-03:
Your warning payload is not correct, see https://www.zigbee2mqtt.io/devices/HS2WD-E.html#triggering-the-alarm for the correct payload.

KennethLavrsen, 2019-08-05:
I am also trying to setup this device and added the reporters setup as described for devices.js

I do not get any payload with "smoke" in it. @jjposio are you sure this is the correct devices.js? Or are you running some dev version?

jjposio, 2019-08-05:
> I am also trying to setup this device and added the reporters setup as described for devices.js
> 
> I do not get any payload with "smoke" in it. @jjposio are you sure this is the correct devices.js? Or are you running some dev version?

Yes, that devices.js above is the one I am using and no dev branches.
Though that "smoke" key-value pair might not visible right away after you initially start zigbee2mqtt up after adding the device.. It took some time for me before it  was visible in logs.

E.g. first I got these
zigbee2mqtt:info 8/5/2019, 10:25:29 PM MQTT publish: topic 'zigbee2mqtt/0x0015bc0031002651', payload '{"battery_low":false,"linkquality":15,"temperature":24.37}'
...
and after a while, also smoke showed up.
zigbee2mqtt:info 8/5/2019, 10:26:17 PM MQTT publish: topic 'zigbee2mqtt/0x0015bc0031002651', payload '{"battery_low":false,"linkquality":18,"temperature":24.31,"smoke":false}'

jjposio, 2019-08-05:
> Your warning payload is not correct, see https://www.zigbee2mqtt.io/devices/HS2WD-E.html#triggering-the-alarm for the correct payload.

Thanks. I will test with this.

KennethLavrsen, 2019-08-05:
Even after several hours I get this in the log

```
8/5/2019, 10:40:45 PM - info: MQTT publish: topic 'zigbee2mqtt/develco_smoke_1', payload '{"battery_low":false,"linkquality":39,"temperature":26.5,"device":{"ieeeAddr":"0x0015bc00310022a9","friendlyName":"develco_smoke_1","type":"EndDevice","nwkAddr":16111,"manufId":4117,"manufName":"Develco Products A/S","powerSource":"Battery","modelId":"SMSZB-120","hwVersion":"unknown","swBuildId":"unknown","dateCode":"2018-03-15 03:10","status":"offline"}}'
8/5/2019, 10:40:45 PM - warn: No converter available for 'SMSZB-123' with cid 'msTemperatureMeasurement', type 'devChange' and data '{"cid":"msTemperatureMeasurement","data":{"measuredValue":2650}}'
8/5/2019, 10:40:45 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
```
No trace of any smoke info

I am using a git checkout from master branch and it was updated today.

KennethLavrsen, 2019-08-05:
Tried to set fire to a stick and blow the smoke in the thing setting off the alarm. No smole in the logs or the mqtt payload. It does not work. There is something you have that I don't

jjposio, 2019-08-06:
> Your warning payload is not correct, see https://www.zigbee2mqtt.io/devices/HS2WD-E.html#triggering-the-alarm for the correct payload.

Tested this out with following error on log:
zigbee2mqtt:error 8/6/2019, 2:12:03 PM No converter available for 'state' ({"warning": {"duration": 10, "mode": "emergency", "strobe": false}}.)

jjposio, 2019-08-06:
> Tried to set fire to a stick and blow the smoke in the thing setting off the alarm. No smole in the logs or the mqtt payload. It does not work. There is something you have that I don't. I notice that you recorded a version 1.5.1.1. Mine says 1.5.1. Was that a typo?
> 
> Update. I have also tried to wipe everything, reflash the CC2531 with the latest release. No smoke ever. Does not work. How can I help?

[image]
I took the version number from here.

And I also did the setup again for the device, just to be sure that I didn't miss out anything.
I deleted everything related to smoke alarm from configuration.yaml and database.db. I restarted zigbee2mqtt, paired the device and checked logs that it says device not supported.

After that I connected Hassio via SSH. Entered hassio container and copy-pasted code from here to node_modules/zigbee-shepherd-converters/devices.js. exited and restarted container.
And after a while smoke alarm started reporting again, including smoke key-value.

KennethLavrsen, 2019-08-06:
@jjposio then we are not using the same code base. You run on the Home Assistant hass.io addon which is built from a different Github repo
I run from the Koenkk with a git checkout from master. I will try to compare the two later.
By the way I think Koen meant us to use the zigbee-shepherd-converters repo for support of new devices.

Koenkk, 2019-08-06:
> Tested this out with following error on log:
> zigbee2mqtt:error 8/6/2019, 2:12:03 PM No converter available for 'state' ({"warning": {"duration": 10, "mode": "emergency", "strobe": false}}.)

The payload is probably not correct, can you provide the zigbee2mqtt debug log when executing this?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

jjposio, 2019-08-07:
> The payload is probably not correct, can you provide the zigbee2mqtt debug log when executing this?
>
> To enable debug logging set in `configuration.yaml`:
>
> ```yaml
> advanced:
>   log_level: debug
> ```

That adds another line to log,..

 zigbee2mqtt:debug 8/7/2019, 1:32:22 PM Received MQTT message on 'zigbee2mqtt/0x0015bc0031002651/set' with data '{"warning": {"duration": 10, "mode": "emergency", "strobe": false}}.'
  zigbee2mqtt:error 8/7/2019, 1:32:22 PM No converter available for 'state' ({"warning": {"duration": 10, "mode": "emergency", "strobe": false}}.)

Koenkk, 2019-08-07:
Try removing the trailing dot {"duration": 10, "mode": "emergency", "strobe": false}}. <--

jjposio, 2019-08-07:
> Try removing the trailing dot {"duration": 10, "mode": "emergency", "strobe": false}}. <--

Ah, yes. Now there seems to be a bit more information. Noticed also that there is warn about msTemperatureMeasurement.

zigbee2mqtt:error 8/7/2019, 2:26:17 PM Zigbee publish to device '0x0015bc0031002651', ssIasWd - startWarning - {"startwarninginfo":49,"warningduration":10} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: Timed out after 30000 ms

zigbee2mqtt:info 8/7/2019, 2:26:17 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: Timed out after 30000 ms","meta":{"entity":{"ID":"0x0015bc0031002651","type":"device","friendlyName":"0x0015bc0031002651"},"message":"{\"warning\": {\"duration\": 10, \"mode\": \"emergency\", \"strobe\": false}}"}}'

zigbee2mqtt:debug 8/7/2019, 2:26:29 PM Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":48}' of device 'SMSZB-120' (0x0015bc0031002651) of endpoint 35

zigbee2mqtt:info 8/7/2019, 2:26:29 PM MQTT publish: topic 'zigbee2mqtt/0x0015bc0031002651', payload '{"battery_low":false,"linkquality":13,"temperature":25.75,"smoke":false}'

zigbee2mqtt:debug 8/7/2019, 2:27:00 PM Received zigbee message of type 'attReport' with data '{"cid":"msTemperatureMeasurement","data":{"measuredValue":2581}}' of device 'SMSZB-120' (0x0015bc0031002651) of endpoint 38

zigbee2mqtt:info 8/7/2019, 2:27:00 PM MQTT publish: topic 'zigbee2mqtt/0x0015bc0031002651', payload '{"battery_low":false,"linkquality":15,"temperature":25.81,"smoke":false}'

zigbee2mqtt:debug 8/7/2019, 2:27:00 PM Received zigbee message of type 'devChange' with data '{"cid":"msTemperatureMeasurement","data":{"measuredValue":2581}}' of device 'SMSZB-120' (0x0015bc0031002651) of endpoint 38

zigbee2mqtt:warn 8/7/2019, 2:27:00 PM No converter available for 'SMSZB-123' with cid 'msTemperatureMeasurement', type 'devChange' and data '{"cid":"msTemperatureMeasurement","data":{"measuredValue":2581}}'

zigbee2mqtt:warn 8/7/2019, 2:27:00 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

KennethLavrsen, 2019-08-08:
Hi again. I I not trying to hijack this but this is a bug report that contains the code to add a new device and it does not work. I use a Zigbee2Mqtt which is straight out of git and I only have 2 devices setup at the moment. I have had this running for days and the sensor reports temperature but no smoke state which is the whole point of a smoke sensor. So even if the OP gets his MQTT siren to work please do not enter the suggested code because something is missing that makes it work for him

jjposio, 2019-08-09:
> Hi again. I I not trying to hijack this but this is a bug report that contains the code to add a new device and it does not work. I use a Zigbee2Mqtt which is straight out of git and I only have 2 devices setup at the moment. I have had this running for days and the sensor reports temperature but no smoke state which is the whole point of a smoke sensor. So even if the OP gets his MQTT siren to work please do not enter the suggested code because something is missing that makes it work for him
> 
> I am not sure what is happening in this .js file but I can see that most other sensors are defined with a line like
> 
> ```
>             const device = shepherd.find(ieeeAddr, 1);
> ```
> 
> And later
> 
> ```
>             execute(device, actions, callback, 250);
> ```
> 
> Here we have
> 
> ```
>             const ep2 = shepherd.find(ieeeAddr, 35);
>             const ep3 = shepherd.find(ieeeAddr, 38);
> ```
> 
> and then
> 
> ```
>             execute(ep3, actions, callback, 1000);
> ```
> 
> What happened with the defined ep2?
> 
> I am not good at javascript (I mostly code C and Perl) but it seems something is missing

Well, I can assure you that there is nothing left out from that code above. That is just what I have initially wrote to devices.js quick and dirty style to get some information out. 
And what would possibly be the point of not telling something, why it works for me as it does?

And what comes to this ep2 - ep3 case. I took that way of handling multiple ieeeAddresses from Osram Smart+ switch mini setup from devices.js. And my initial thought was also that could it ever work that way, but it did.
Though I have made it a bit sleekier later on. This works equally to code above and causes less confusion:

     // develco
     {
        zigbeeModel: ['SMSZB-120'],
        model: 'SMSZB-123',
        vendor: 'Develco Products A/S',
        description: 'Smoke alarm and siren',
        fromZigbee:[fz.heiman_smoke,fz.generic_temperature,fz.battery_200,fz.ignore_iaszone_change],
        toZigbee:[tz.warning],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
                const ep1 = shepherd.find(ieeeAddr, 35);
                execute(ep1, [
                                (cb) => ep1.bind('genPowerCfg',coordinator,cb),
                                (cb) => ep1.write('ssIasZone','iasCieAddr',coordinator.device.getIeeeAddr(), cb),
                                (cb) => ep1.functional('ssIasZone','enrollRsp',{enrollrspcode: 0, zoneid: 23}, cb),
                                (cb) => ep1.report('genPowerCfg','batteryVoltage',0,65534,0,cb),
                                ], () => {
                                        const ep2 = shepherd.find(ieeeAddr, 38);
                                        execute(ep2, [
                                                (cb) => ep2.bind('msTemperatureMeasurement', coordinator, cb),
                                                (cb) => ep2.report('msTemperatureMeasurement','measuredValue',30, 600, 1, cb)], callback);
                });
        },
     },

jjposio, 2019-08-26:
Hi, @KennethLavrsen, did you manage to get this to work any better than it did before? Or event triggered the alarm?
I finally got some spare time to check this again and do some experiments with code, but still, no way of triggering the alarm. @Koenkk any pointers what to look out for?

KennethLavrsen, 2019-08-26:
No - no luck.
But I have been in travelling on business half the time since last posting. But I can say that I never managed to get the smoke reported.

Can you send me a copy of your devices.js file so I can compare and/or try with yours? There is still that small difference that you get yours as a HA addon and I pull it from git. Even though the addon should also pull from same git when it builds

Koenkk, 2019-08-27:
Maybe the enrollment is not correct, #1888 may fix this.

jjposio, 2019-08-31:
@KennethLavrsen Here is copy from devices.js I use. And looking forward to get this 1888 for testing.

[devices.zip](https://github.com/Koenkk/zigbee2mqtt/files/3562727/devices.zip)

KennethLavrsen, 2019-09-03:
Thanks for the file. There were indeed large changes from your initial post.
I simply replaced the entire file and repaired the sensor.

Result. Still no smoke. So it must be the firmware inside the sensor that is different. I think I will move on and buy a sensor which is known to be supported by either Zigbee2MQTT or Deconz. Or maybe I make my own ESP8266 based solution. I wish the manufacturers would be more active getting their hardware supported in the popular software

KennethLavrsen, 2019-09-05:
After I upgraded my deconz it now supports the smoke alarm. So I moved it over to deconz. I only use zigbee2mqtt when deconz does not support something. If you need to gain info about it - looking at the deconz code is a source of info. You can also activate the siren in deconz

SwoopX, 2019-09-26:
Hi there,

let me chime in here as I might be able to aid in better support this sensor. I also contributed in the integration in deCONZ and studied the technical documentation quite extensively ;)

But first: @KennethLavrsen: Does the sensor work flawlessly in deconz? With my dev-gateway, I'm having issues especially with the attribute reporting (temperature and battery). However, the siren works.

If I'd have seen this post earlier, it would have spared my some time to write my own device definition (to better investigate the sensor's behaviour and functionality), however, I'm more than happy th share what I have. Reports are set up according to Develco documentation.

**Device definition:**
```
{
    zigbeeModel: ['SMSZB-120'],
    model: 'SMSZB-120',
    vendor: 'Develco Products A/S',
    description: 'Smoke detector with siren',
    supports: 'smoke, warning, temperature',
    fromZigbee: [
        fz.generic_temperature,
        fz.ignore_temperature_change,
        fz.cr123_battery_att_report,
        fz.develco_smoke,
        fz.develco_smoke_status_change
    ],
    toZigbee: [tz.warning],
    configure: (ieeeAddr, shepherd, coordinator, callback) => {
        const ep35 = shepherd.find(ieeeAddr, 35);
        
        execute(ep35, [
            //(cb) => ep35.bind('ssIasZone', coordinator, cb),
            (cb) => ep35.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
            (cb) => ep35.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb),
            (cb) => ep35.bind('genPowerCfg', coordinator, cb),
            (cb) => ep35.report('genPowerCfg', 'batteryVoltage', 600, 43200, cb),
        ], () => {
        const ep38 = shepherd.find(ieeeAddr, 38);
        
        execute(ep38, [
            (cb) => ep38.bind('msTemperatureMeasurement', coordinator, cb),
            (cb) => ep38.report('msTemperatureMeasurement', 'measuredValue', 60, 600, 10, cb)
        ], callback);
       });
    },    
},
```
In the "fromZigbee" passage, I'm afraid the battery converter from @jjposio is not fully adequate, as it does not correctly determine the battery alarm state. Therefore, I've created "cr123_battery_att_report".

**Converters:**
```
cr123_battery_att_report: {
	cid: 'genPowerCfg',
	type: 'attReport',
	convert: (model, msg, publish, options) => {
		const result = {};
		if (typeof msg.data.data['batteryVoltage'] == 'number') {
			const battery = {max: 3000, min: 2500};
			const voltage = msg.data.data['batteryVoltage'] * 100;
			result.battery = toPercentage(voltage, battery.min, battery.max);
			result.voltage = voltage;
		}
		if (typeof msg.data.data['batteryAlarmState'] == 'number') {
			result.battery_alarm_state = msg.data.data['batteryAlarmState'];
		}
		return result;
	},
},
```

```
develco_smoke: {
    cid: 'ssIasZone',
    type: ['attReport', 'readRsp', 'devChange'],
    convert: (model, msg, publish, options) => {
        const zoneState = msg.data.data.zoneState;
        const zoneType = msg.data.data.zoneType;
        const zoneStatus = msg.data.zoneStatus;
        const iasCieAddr = msg.data.iasCieAddr;
        const zoneId = msg.data.data.zoneId;
        
        return {
            sensor_type: zoneType,
            enrolled: zoneState,
            smoke: (zoneStatus & 1) > 0, // Bit 1 = Alarm: Smoke
            IASCIEAddress: iasCieAddr,
            zoneID: zoneId,
            battery_low: (zoneStatus & 1<<3) > 0, // Bit 4 = Battery LOW indicator
        };
    },
},
```
This converter has been extended to also provide all available and probably relevant attributes. The type part might also be tweaked later on.

```
develco_smoke_status_change: {
    cid: 'ssIasZone',
    type: ['statusChange'],
    convert: (model, msg, publish, options) => {
        const zoneStatus = msg.data.zoneStatus;
        
        return {
            smoke: (zoneStatus & 1) > 0, // Bit 1 = Alarm: Smoke
            battery_low: (zoneStatus & 1<<3) > 0, // Bit 4 = Battery LOW indicator
        };
    },
},
```
This one might be omitted, since it does not really be relevant. Could also be rewritten to ignore those messages.

I'm not too sure if the fire alarm would go off since the IasCie write does not seem to work correctly as I see a success on the first write but afterwards, it returns to undefined default. Also, the enroll request is not working at all. Maybe it has something to do with the auto-enroll the sensor does... Still testing this one.

What's also not bee tested yet is the siren for warnings. I know the device behaves a little different than one would expect but it definitively must fire on the byte value 00000001 or hex 01. on the first glimpse, it may be that the warning converter is not 100% correct. Haven't checked it yet.

Hope that's useful,
cheers!

KennethLavrsen, 2019-09-27:
I am not sure it works flawless in Deconz. I see a battery level and temperature.
But most importantly it reports smoke which is the key feature. That is what I could not get to work in Zigbee2Mqtt.

I note that I can turn on the siren but I have trouble turning it off. I have to press the device button

SwoopX, 2019-09-27:
Thanks for sharing your experience! In my case, it is exactly the other way around, and I cannot figure out why which drives me nuts

> I note that I can turn on the siren but I have trouble turning it off. I have to press the device button
Is that in deconz? If yes, you might want to check out this threat:
[https://github.com/dresden-elektronik/deconz-rest-plugin/pull/1593](https://github.com/dresden-elektronik/deconz-rest-plugin/pull/1593)

If you issue a "select" via api for the siren, it has a duration for 3 sec., "lselect" has 200 sec ;)

closed 2019-12-03 by stale[bot]

BlizzWave, 2020-02-21:
Has anyone figured this out yet?

SwoopX, 2020-02-21:
What exactly? Managed to get it fully supported in deconz.

BlizzWave, 2020-02-21:
Talking about zigbee2mqtt here. I know it works in deconz

SwoopX, 2020-02-21:
Ah, haven't tried it ever since.