Koenkk/zigbee2mqtt #1810: Brightness, color and transition with ikea light

irgusite, 2019-08-03:
# Bug Report
## What happened

Related to : #341 

I'm using circadian lighting and zigbee2mqtt with ikea lights. 
When circadian is updating the light brightness and color, the bulb doesn't apply the new brightness. 
What I saw, is that when updating the brightness and color, along with a transition time, only the color updates but not the brightness, and the z2m plugin reports the new brightness (but not applied)

After more observation, as long as the color transition is not over, the brightness is reset to it's precedent state but not communicated to the bulb.

## What did you expect to happen
To have the brightness and color changed with a transition time

## How to reproduce it (minimal and precise)
- Have an ikea bulb or pannel (tested on both)
- Turn off circadian and such. 
- Change the brightness and color of the bulb with a transition time
- see only the color change

## Debug Info
zigbee2mqtt version: 59a5c6b
CC253X firmware version: from july 2019

Koenkk, 2019-08-03:
Can you provide the debug log when the MQTT commands are send?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

irgusite, 2019-08-05:
I will do it this evening when I get home.

irgusite, 2019-08-06:
Hey! I found time to do the debug. Find here the 3 log extracts with color+brightness+transition, color+transition and brightness+transition: 

https://pastebin.com/7pbihrRC

Koenkk, 2019-08-08:
- Which cases don't give expected behaviour?
- On line 17 I see `brightness":200`, on line 11 `"brightness":20`, what happend in between? Do you see any pointers why the brightness did increase?

irgusite, 2019-08-08:
The case with brightness color and transition. The color and transition are
applied, but the brightness isn't, but it shows as if it were. (the actual
lamp doesn't change brightness but  it is changed on the dashboard)

irgusite, 2019-08-16:
Oh, I didn't answer your second question. For the brightness change I have no idea. I don't remember changing it in-between and I had the other plug-ins off...

Koenkk, 2019-08-18:
Could you try to investigate that further?

irgusite, 2019-08-22:
Hey, sorry for the delay, I hasn't got any time. So for the brightness change, it was me that changed it. 
So, if I send

`{
  "entity_id": "light.ampoule_salon_mqtt",
"color_temp":440,
"brightness":10,
"transition":2
}`
to my bulb, the color temps is applied with the transition delay. But the brighness isn't changed (in real, you see the light dim a little bit, and come back to it's original brightness, as if the brightness change is interupted by the color change)
The log lines for this event are these: https://pastebin.com/N5Pxf98x

As I see in the log, the two commands are send one after another, and this could be the cause of that, the first "moveToLevelWithOnOff " is received, but just after, ignored by the bulb because it receives the "moveToColorTemp " which is applied.

Koenkk, 2019-08-26:
> As I see in the log, the two commands are send one after another, and this could be the cause of that, the first "moveToLevelWithOnOff " is received, but just after, ignored by the bulb because it receives the "moveToColorTemp " which is applied.

Can you also reproduce this manually?

irgusite, 2019-08-26:
> Can you also reproduce this manually?

What do you mean by manually?

Koenkk, 2019-08-27:
By sending the mqtt commands via a client like mqtt.fx. It seems that the bulb discards the command, could you try to find a sequence when it doenst? Maybe e.g. first color temperature and then brightness?

I will also investigate this after #1888

irgusite, 2019-08-28:
Sooooo, after multiple tries (don't have the logs). When you first set brightness and then color (during the transition interval) the brightness gets reset to original brightness and color is applied with transition.

When doing it in reverse (color and THEN brightness) the brightness gets ignored. But, the next time the color is updated with the SAME value, the brightness is applied.

TL;DR; The brightness is always ignored by the bulb if a color command is send in the transition interval. (before of after)

irgusite, 2019-08-28:
Furthermore, if color_temp is send, before brightness, but the color is the same as alread set, the brightness is updated

Koenkk, 2019-08-28:
Thanks for your investigation, so it seems that a current transition is discarded when a new transition is started.

irgusite, 2019-08-28:
only the brightness transition is discarded, the color isn't....

abmantis, 2019-08-29:
@Koenkk I think the problem is the one described in https://github.com/Koenkk/zigbee2mqtt/issues/341

I've reproduced it:

```
zigbee2mqtt:debug 8/29/2019, 10:43:54 PM Received MQTT message on 'zigbee2mqtt/Office/set' with data '{"state": "ON", "transition": 1, "brightness": 255, "color_temp": 166}'
zigbee2mqtt:info 8/29/2019, 10:43:54 PM Zigbee publish to device 'Office', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":10} - {"manufSpec":0,"disDefaultRsp":0} - 1
zigbee2mqtt:info 8/29/2019, 10:43:54 PM MQTT publish: topic 'zigbee2mqtt/Office', payload '{"state":"ON","linkquality":102,"brightness":255,"color_temp":166,"color_mode":2,"color":{"x":0.4599,"y":0.4106}}'
zigbee2mqtt:info 8/29/2019, 10:43:54 PM Zigbee publish to device 'Office', lightingColorCtrl - moveToColorTemp - {"colortemp":166,"transtime":10} - {"manufSpec":0,"disDefaultRsp":0} - 1
zigbee2mqtt:info 8/29/2019, 10:43:55 PM MQTT publish: topic 'zigbee2mqtt/Office', payload '{"state":"ON","linkquality":102,"brightness":255,"color_temp":166,"color_mode":2,"color":{"x":0.4599,"y":0.4106}}'
```

Even tough the MQTT message contains both the brightness and color temp, there are two different zigbee messages sent to the light. It seems that the second one (color temp) cancels the transition that is happening for the first one.

Koenkk, 2019-08-30:
So it seems this is just a software limitation of the bulb itself, not sure how we can work around it.

irgusite, 2019-08-30:
Would it be possible to add a delay between the 2 orders (via config param or something similar)?

abmantis, 2019-08-30:
I don't think a delay would work. Let's say you do a change to brightness and ct that should take 5 seconds. For it to work with a delay, it would have to send the ct change command 5 seconds after the brightness command, which would make the whole change take 10 seconds.

irgusite, 2019-08-30:
Yeah, but at the moment, it just cancels the other command. 
Personally I don't care if the 2 transitions take each 5 seconds to complete, with the bulb you can't have both at the same time. But at the moment you just have 1 of the transitions and not the other

That why I'm proposing a config, so you can activate or not the behavior. But at the moment, it just doesn't work at all with half of the integrations from HA.

Making the correction at the level of z2m makes it easier for every integration.

abmantis, 2019-08-30:
Yeah, you're right. I too prefer that it takes longer but gets there. If there's no other way, that is better than the current behavior.
By the way, have you checked if there is any firmware update for the bulb? I may do that later with deconz.

irgusite, 2019-09-17:
Any news on a possible implementation/correction?

Koenkk, 2019-09-24:
I was just thinking about this, what if we only pusblish the transition for the last published value?

E.g. when sending `{brightness: 100, color_temp: 200, transition: 5}` :

Currently:
- brightness 100 with transition 5 <--- fails to apply
- color temp 200 with transition 5

Proposed fix:
- brightness 100 without transition
- color temp 200 with transition 5

irgusite, 2019-09-24:
Didn't thought of that, seems a good idea. :)

abmantis, 2019-09-24:
Good idea. I just suggest changing the order. I think brightness changing abruptly is worse than color temp.

Koenkk, 2019-09-25:
- I've pushed a fix for this in the dev branch (https://github.com/Koenkk/zigbee2mqtt/issues/1810#issuecomment-534511680).
- I've only enabled this for TRADFRI bulbs as this works fine already for e.g. Philips Hue bulbs.
- @abmantis this is a bit tricky as in zigbee the brightness command can also control the state while the color cannot. Therefore currently the brightness command is always executed before the color one.

irgusite, 2019-09-26:
Nope it doesn't work.

I updated like this:
sudo systemctl stop zigbee2mqtt
cd /opt/zigbee2mqtt

# Backup configuration
cp -R data data-backup

# Update
git pull origin dev
git checkout dev
rm -rf node_modules
npm install

# Restore configuration
cp -R data-backup/* data

# Start zigbee2mqtt
sudo systemctl start zigbee2mqtt

Koenkk, 2019-09-27:
Can you share the contents of `node_modules/zigbee-herdsman-converters/converters/fromZigbee.js`?

atxbyea, 2019-10-03:
I installed the latest-dev docker yesterday, but it appears to still be having issues with adjusting brightness and temperature of my IKEA bulbs.

Koenkk, 2019-10-30:
@atxbyea do you control them via groups?

irgusite, 2019-10-30:
Just tested, and it seems to work fine :) Thanks for your work 👍

Koenkk, 2019-10-30:
@irgusite thanks for confirming.

I've also pushed a fix for TRADFRI devices in a group. Assuming this is fixed now.

closed 2019-10-30 by Koenkk

dgarciadonado, 2020-05-01:
Hello @Koenkk , I am experimenting the same issue with Ikea Tradfri bulbs. 
When I turn on the light defining both brightness and color, and specifiying a transition, the bulb applies the correct brightness and color but doesn't show the transition. Is it solved in the Dev branch or this is now applied on the Master branch?
If a new issue is needed, just tell me and I will open it.
Thank you very much.

Koenkk, 2020-05-02:
@dgarciadonado in that case the transition will be only applied over the color, the brightness is set immediately (tradfri bulbs don't support transitioning on both properties).

dgarciadonado, 2020-05-02:
> @dgarciadonado in that case the transition will be only applied over the color, the brightness is set immediately (tradfri bulbs don't support transitioning on both properties).

Thank you @Koenkk , and is there any way to force brightness transition instead of temperature transition (if using both at the same time)

Koenkk, 2020-05-02:
@dgarciadonado no at the moment not (and not very easy to change).

w3host, 2020-05-22:
This workaround can help in HA: 
https://www.wouterbulten.nl/blog/tech/ikea-tradfri-temp-and-brightness-with-home-assistant/

So firstly set brightness and transition and after 1s delay set color_temp...
 
 action:
  - data:
      brightness_pct: 100
      transition: 1
    entity_id: light.0x14b457fffe4e0ecf_light
    service: light.turn_on
  - delay: 00:00:01
  - data:
      color_temp: 300
    entity_id: light.0x14b457fffe4e0ecf_light
    service: light.turn_on

dgarciadonado, 2020-05-22:
Thank you!! I will give a try!

dgarciadonado, 2020-05-22:
Hello @w3host, now is working! Thank you!
I am using many scripts like this (changing brightness and temperature):

```
      - service: light.turn_on
        data_template:
          entity_id: light.salon_1_zigbee_group
          brightness: 127 
          transition: 2
      - delay: '00:00:02'
      - service: light.turn_on
        data_template:
          entity_id: light.salon_1_zigbee_group           
          color_temp: 266 
          transition: 3 
```
Now I have another problem: if the script is executing the final transition and I call another script which affects this light, the lights changes to be crazy and doesn`t reflect the real Brightness and Color.
Maybe you have experiencing this behaviour?

Thank you again!