Koenkk/zigbee2mqtt #1811: IKEA E1603/E1702 plug is always offline

PedroLamas, 2019-08-03:
# Bug Report
## What happened

I have 2 IKEA Tradfri smart plugs that always show as offline (thus making them not appear automatically in Home Assistant)

[image]

## What did you expect to happen

I've manually configured these in Home Assistant and I can turn them on/off, so I know they work fine, yet on the network map they always show as offline and if I leave Home Assistant to auto-discover the devices, it never finds these plugs.

## How to reproduce it (minimal and precise)

I'm running both Zigbee2mqtt and Home Assistant via docker, the whole configuration is [here](https://github.com/PedroLamas/nanopineo2-docker-config).

## Debug Info
zigbee2mqtt version: 1.5.1
CC253X firmware version: 20190608

Debug log available [here](https://pastebin.com/vMydzB8z)

Koenkk, 2019-08-03:
It is known that the offline/online state is not reliable: https://github.com/Koenkk/zigbee-herdsman/issues/13

In zigbee2mqtt 1.5.1 this device does not expose any sensors to homeassistant (so what you see is expected behaviour).

However in zigbee2mqtt 1.6 it will expose a linkquality sensor (https://github.com/Koenkk/zigbee2mqtt/issues/1417). For now you can test that in the dev branch.

closed 2019-08-03 by Koenkk

PedroLamas, 2019-08-03:
Thank you for the quick reply! In that case, I will be trying the `dev` branch and see how it behaves! 🙂