Koenkk/zigbee2mqtt #1813: on hass.io zigbee2mqtt update not working.

zambujal, 2019-08-05:
PROBLEM DETECTED
Please create an issue here with the contents of below: https://github.com/koenkk/zigbee2mqtt/issues
===== Stack =====
Error: Write for 'UTC' not available
    at Function.write (/app/node_modules/zigbee-herdsman/dist/buffalo/buffalo.js:240:19)
    at Function.write (/app/node_modules/zigbee-herdsman/dist/zcl/buffaloZcl.js:247:26)
    at Function.writeUseDataType (/app/node_modules/zigbee-herdsman/dist/zcl/buffaloZcl.js:48:21)
    at Function.write (/app/node_modules/zigbee-herdsman/dist/zcl/buffaloZcl.js:243:25)
    at ZclFrame.writePayloadGlobal (/app/node_modules/zigbee-herdsman/dist/zcl/zclFrame.js:95:54)
    at ZclFrame.toBuffer (/app/node_modules/zigbee-herdsman/dist/zcl/zclFrame.js:53:29)
    at Object.frame (/app/node_modules/zigbee-herdsman/dist/lib/components/zcl.js:30:31)
    at Object.af.zclFoundation (/app/node_modules/zigbee-herdsman/dist/lib/components/af.js:198:25)
    at ZShepherd._foundation (/app/node_modules/zigbee-herdsman/dist/lib/shepherd.js:597:20)
    at Endpoint.ep.foundation (/app/node_modules/zigbee-herdsman/dist/lib/shepherd.js:504:25)
===== Arguments =====
{ frameType: 0, manufSpec: 0, direction: 1, disDefaultRsp: 1 } 0 74 'readRsp' [ { attrId: 0, status: 0, attrData: 618288106, dataType: 226 } ] undefined
===== zcl-packet =====
{"type":"Buffer","data":[24,74,1,0,0,0,226,234,83,218,36]}
===== zcl =====
null
===============
/app/node_modules/zigbee-herdsman/dist/lib/components/af.js:37
        throw new TypeError('Af rawPayload should be a buffer.');
        ^
TypeError: Af rawPayload should be a buffer.
    at Object.af.send (/app/node_modules/zigbee-herdsman/dist/lib/components/af.js:37:15)
    at Object.af.zclFoundation (/app/node_modules/zigbee-herdsman/dist/lib/components/af.js:219:8)
    at ZShepherd._foundation (/app/node_modules/zigbee-herdsman/dist/lib/shepherd.js:597:20)
    at Endpoint.ep.foundation (/app/node_modules/zigbee-herdsman/dist/lib/shepherd.js:504:25)
    at Object.queue.push [as func] (/app/lib/zigbee.js:520:24)
    at ZigbeeQueue.executeNext (/app/lib/util/zigbeeQueue.js:76:18)
    at Timeout.timer.setInterval [as _onTimeout] (/app/lib/util/zigbeeQueue.js:43:18)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.5.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.5.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

Koenkk, 2019-08-05:
Due to this I found a critical bug, thanks!!

Fixed in the latest version (wait 2 hours before updating to give the new image some time to be generated).

zambujal, 2019-08-08:
No problem Koen, many thanks for this work!

[https://ipmcdn.avast.com/images/icons/icon-envelope-tick-round-orange-animated-no-repeat-v1.gif]<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>  Sem vírus. www.avast.com<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>

closed 2019-08-09 by Koenkk